Azure/Industrial-IoT #1685: OPC Publisher container stopping

developedbymayur, 2022-05-04:
[image]

Hi all , we have found that opc container stops timely throwing following error. Your suggestions will be helpful.

cristipogacean, 2022-05-04:
this looks like a connectivity issue between the opc publisher and the OPC UA server. the OPC publisher manages to restore the connection though. Is it possible that the server was restarted in this case?

developedbymayur, 2022-05-05:
> this looks like a connectivity issue between the opc publisher and the OPC UA server. the OPC publisher manages to restore the connection though. Is it possible that the server was restarted in this case?

No the server was not restarted , what kind of information do you need? can you please let me know

developedbymayur, 2022-05-05:
`Opc.Ua.ServiceResultException: BadOutOfMemory
   at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
   at Opc.Ua.SessionClient.EndPublish(IAsyncResult result, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at Opc.Ua.Client.Session.OnPublishComplete(IAsyncResult result)
[06:20:34 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) PUBLISH #164 - Unhandled error during Publish.
 BadOutOfMemory 'BadOutOfMemory'`

developedbymayur, 2022-05-05:
`[08:12:13 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session '25e31c4c8a33b7f2046d513d854736e6732c4371' set to refresh due to Keepalive with reported status BadNoCommunication 'Server not responding to keep alive requests.' 
[08:12:23 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session '25e31c4c8a33b7f2046d513d854736e6732c4371' set to refresh due to Keepalive with reported status BadNoCommunication 'Server not responding to keep alive requests.' 
[08:12:33 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session '25e31c4c8a33b7f2046d513d854736e6732c4371' set to refresh due to Keepalive with reported status BadNoCommunication 'Server not responding to keep alive requests.' 
[08:12:38 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to refresh session '25e31c4c8a33b7f2046d513d854736e6732c4371' 
Opc.Ua.ServiceResultException: BadSessionIdInvalid
   at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
   at Opc.Ua.SessionClient.Read(RequestHeader requestHeader, Double maxAge, TimestampsToReturn timestampsToReturn, ReadValueIdCollection nodesToRead, DataValueCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at Opc.Ua.Client.Session.FetchNamespaceTables()
   at Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager.HandleRefreshAsync(ConnectionIdentifier id, SessionWrapper wrapper, CancellationToken ct) in D:\a\1\s\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Protocol\src\Services\DefaultSessionManager.cs:line 484
[08:12:43 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session '25e31c4c8a33b7f2046d513d854736e6732c4371' set to refresh due to Keepalive with reported status BadNoCommunication 'Server not responding to keep alive requests.' 
[08:12:43 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session '25e31c4c8a33b7f2046d513d854736e6732c4371' missed 1 Keepalive(s) due to BadNoCommunication, waiting to reconnect... 
[08:12:43 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to reconnect session '25e31c4c8a33b7f2046d513d854736e6732c4371' due to BadSessionIdInvalid,  disposing and trying to create new 
[08:12:44 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Creating session '25e31c4c8a33b7f2046d513d854736e6732c4371' for endpoint 'opc.tcp:/xx.xx.xx.xx:49320'... 
[08:12:45 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session '25e31c4c8a33b7f2046d513d854736e6732c4371' created, loading complex type system ...  
[08:12:45 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session '25e31c4c8a33b7f2046d513d854736e6732c4371' complex type system loaded 
[08:12:45 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Azure IIoT 25e31c4c8a33b7f2046d513d854736e6732c4371 took 00:00:00.7427280. 
[08:12:45 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Connected to 'opc.tcp:/xx.xx.xx.xx:49320' 
[08:12:45 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Added 308 monitored items to subscription opc.tcp:/xx.xx.xx.xx:49320_2d22a0bccf1eabcda6f1a8b434a54d938f9b94c2 
[08:12:45 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Now monitoring 308 nodes in subscription opc.tcp:/xx.xx.xx.xx:49320_2d22a0bccf1eabcda6f1a8b434a54d938f9b94c2 
[08:12:45 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Added 406 monitored items to subscription opc.tcp:/xx.xx.xx.xx:49320_d72ea3446a4992868c66dad161f579d5bd17053c 
[08:12:45 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Now monitoring 406 nodes in subscription opc.tcp:/xx.xx.xx.xx:49320_d72ea3446a4992868c66dad161f579d5bd17053c 
[08:12:45 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Set Monitoring to Reporting for 308 nodes in subscription opc.tcp:/xx.xx.xx.xx:49320_2d22a0bccf1eabcda6f1a8b434a54d938f9b94c2 
[08:12:45 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Set Monitoring to Reporting for 406 nodes in subscription opc.tcp:/xx.xx.xx.xx:49320_d72ea3446a4992868c66dad161f579d5bd17053c ```

cristipogacean, 2022-05-05:
> this looks like a connectivity issue between the opc publisher and the OPC UA server. the OPC publisher manages to restore the connection though. Is it possible that the server was restarted in this [...]

We will need to understand what exactly is the problem here. From the traces we can see that there is a connectivity issue between OPC publisher and the OPC UA Server that is mitigated and the connection is reestablished. this is perfectly valid functionality so we don't see a problem here. 

The really strange thing is the `BadOutOfMemory` status code we receive on the publish response. I looked into the opc ua client stack code and we're not actively setting the `BadOutOfMemory` status code anywhere in the code. So, I suspect this comes from the server as the response on the publish service. Further on I could imagine the server crashes and is being restarted, scenario which is kind-of what we see in the logs reporting successful reconnection of the publisher. Could you have a look on the server's logs for some weird things? It might be worthy to contact the server's vendor and ask for their opinion.

developedbymayur, 2022-05-05:
Hii Sir , It shows connected back but data is not forwarded into output , when the container is restarted it forwards data again.

developedbymayur, 2022-05-05:
Regarding Bad out of Memory : Does RAM of the VM can affect this error?

hansgschossmann, 2022-05-09:
@developedbymayur as @cristipogacean pointed already out our current analysis says that the message contains a status code ("BadOutOfMemory")  which we got back from the server (from the port number this looks like a KepWare server). If we run out of memory in the container. this would surface differently.

Have you been able to check the logs and status of your OPC UA server you are connecting to?

hansgschossmann, 2022-06-21:
@developedbymayur there was no response for a while. i assume the issue has been resolved. closing for now. feel free to reactivate if there is still an issue.

closed 2022-06-21 by hansgschossmann