Azure/Industrial-IoT #1708: Can't run OPC UA Publisher standalone module inside IoT Edge Simulator environment (iotedgehubdev)

whakey, 2022-05-17:
```OpcUaPublisher                | [11:29:12 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running outside iotedge context. 
OpcUaPublisher                | [11:29:12 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module VilleEdgeDevSim_OpcUaPublisher disconnected due to Communication_Error - now Disconnected... 
OpcUaPublisher                | [11:29:12 ERR Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host failed to start. 
OpcUaPublisher                | [11:29:12 ERR Root] Error during module execution - restarting! 
OpcUaPublisher                | System.AggregateException: One or more errors occurred. (TLS authentication error.)
OpcUaPublisher                |  ---> System.Security.Authentication.AuthenticationException: TLS authentication error.
OpcUaPublisher                |  ---> System.AggregateException: One or more errors occurred. (The remote certificate is invalid according to the validation procedure.)
OpcUaPublisher                |  ---> System.Security.Authentication.AuthenticationException: The remote certificate is invalid according to the validation procedure.
OpcUaPublisher                |    at System.Net.Security.SslStream.StartSendAuthResetSignal(ProtocolToken message, AsyncProtocolRequest asyncRequest, ExceptionDispatchInfo exception)
OpcUaPublisher                |    at System.Net.Security.SslStream.CheckCompletionBeforeNextReceive(ProtocolToken message, AsyncProtocolRequest asyncRequest)
OpcUaPublisher                |    at System.Net.Security.SslStream.StartSendBlob(Byte[] incoming, Int32 count, AsyncProtocolRequest asyncRequest)
OpcUaPublisher                |    at System.Net.Security.SslStream.ProcessReceivedBlob(Byte[] buffer, Int32 count, AsyncProtocolRequest asyncRequest)
OpcUaPublisher                |    at System.Net.Security.SslStream.PartialFrameCallback(AsyncProtocolRequest asyncRequest)
OpcUaPublisher                | --- End of stack trace from previous location where exception was thrown ---
OpcUaPublisher                |    at System.Net.Security.SslStream.EndProcessAuthentication(IAsyncResult result)
OpcUaPublisher                |    at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)
OpcUaPublisher                |    --- End of inner exception stack trace ---
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenInternalAsync(CancellationToken cancellationToken)
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsync(CancellationToken cancellationToken)
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass27_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
OpcUaPublisher                | --- End of stack trace from previous location where exception was thrown ---
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttIotHubAdapter.WriteMessageAsync(IChannelHandlerContext context, Object message, Func`3 exceptionHandler)
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttIotHubAdapter.ConnectAsync(IChannelHandlerContext context)
OpcUaPublisher                |    --- End of inner exception stack trace ---
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass38_0.<<OpenInternalAsync>b__0>d.MoveNext()
OpcUaPublisher                | --- End of stack trace from previous location where exception was thrown ---
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)
OpcUaPublisher                |    at Microsoft.Azure.Devices.Client.InternalClient.OpenAsync()
OpcUaPublisher                |    at Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory.ModuleClientAdapter.CreateAsync(String product, IotHubConnectionStringBuilder cs, String deviceId, String moduleId, ITransportSettings transportSetting, TimeSpan timeout, IRetryPolicy retry, Action onConnectionLost, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Default\IoTSdkFactory.cs:line 281
OpcUaPublisher                |    at Microsoft.Azure.IIoT.Utils.Try.Options[T](Func`1[] options)
OpcUaPublisher                |    --- End of inner exception stack trace ---
OpcUaPublisher                |    at Microsoft.Azure.IIoT.Utils.Try.Options[T](Func`1[] options) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Core\src\Utils\Try.cs:line 106
OpcUaPublisher                |    at Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory.CreateAsync(String product, IProcessControl ctrl) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Default\IoTSdkFactory.cs:line 199
OpcUaPublisher                |    at Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost.StartAsync(String type, String siteId, String productInfo, String version, IProcessControl reset) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Framework\src\Hosting\ModuleHost.cs:line 176
OpcUaPublisher                |    at Microsoft.Azure.IIoT.Modules.OpcUa.Publisher.ModuleProcess.RunAsync() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\ModuleProcess.cs:line 126
```

No matter what I try I can't seem to get publisher module to work inside IoT Edge Dev. I created a new iot edge solution with VS Code IoT extension, and that runs no problem. Then I added OPC UA Publisher module to the template with:

```
"OpcUaPublisher": {
            "settings": {
              "image": "mcr.microsoft.com/iotedge/opc-publisher:2.8.2",
              "createOptions": {
                "Hostname": "opcpublisher",
                "Cmd": [
                  "PkiRootPath=/mount/pki",
                  "--pf=/mount/published_nodes.json",
                  "--lf=/mount/publisher.log",
                  "--mm=PubSub",
                  "--me=Json",
                  "--fm=true",
                  "--fd=false",
                  "--bs=100",
                  "--di=20",
                  "--sc=1",
                  "--aa"
                ],
                "HostConfig": {
                  "Mounts": [
                    {
                      "Type": "bind",
                      "Source": "/opcua",
                      "Target": "/mount"
                    }
                  ],
                  "CapDrop": [
                    "CHOWN",
                    "SETUID"
                  ]
                }
              }
            },
            "type": "docker",
            "version": "1.0",
            "status": "running",
            "restartPolicy": "always"
          }
```          
Which works with my actual test device, but gives me above error with simulator.

I see the issue probably has something to do with "Running outside iotedge context.". I tried with adding the module connection string with environment variables but that didn't change anything.

Has anyone gotten this working with iotedgehubdev version 0.14.14?

marcschier, 2022-05-18:
Duplicate of #1211

dacolgit, 2022-05-18:
Please try the following steps:

tested with opc publisher version 2.8.2, iotedgehubdev version 0.14.14.

Create and edge device in IoTHub

Setup iotedgehubdev:
`iotedgehubdev setup -c "<edge-device-connection-string>"`
Start and debug a single module
`iotedgehubdev start`
Output module credential
`iotedgehubdev modulecred`
Create an EdgeHubConnectionString env variable with the value returned from previous step

Start publisher module

Publisher starts publishing using iotedgehubdev.
In the log publisher displays 'Running outside iotedge context.', but this is just because in simulation environment
Edge Hub connection string is passed to module via environment variable vs. IoT edge where the iotedged daemon does it securely.
If iotedgehubdev is stopped the publisher returns the following error:

[image]

doc references:
[Azure/iotedgehubdev: IoT Edge Hub Dev Tool (github.com)](https://github.com/Azure/iotedgehubdev)
[Develop and debug modules for Azure IoT Edge | Microsoft Docs](https://docs.microsoft.com/en-gb/azure/iot-edge/how-to-vs-code-develop-module?view=iotedge-2020-11)

marcschier, 2023-06-22:
Fixed in preview 4.

closed 2023-06-22 by marcschier