Azure/Industrial-IoT #1709: Full Precision Float/Double Serializer

kiranpradeep, 2022-05-18:
**Is your feature request related to a problem? Please describe.**
If opc server has float value of higher precision e.g. 47.446998596191406f, the current opc publisher writes 47.447 as json result. I assume this is for JamesNK/Newtonsoft.Json#2540. This makes it appear like this opc publisher is losing precision.

**Describe the solution you'd like**
In [pn.json](https://github.com/Azure/Industrial-IoT/blob/main/docs/modules/publisher-migrationpath.md#configuration-file-pnjson), add a new boolean parameter as below, which add full precision functionality to existing serializer used in this project. [Example](https://dotnetfiddle.net/LVV1we). 

```json
[
  {
    "EndpointUrl": "string",
    "fullprecisionfloatdouble": true
  }
]
 
```

**Describe alternatives you've considered**

Fix at source which is NewtonSoft.Json. But that will break compatability of existing json clients. So even though ideal and possible, I will not follow it, as that will break backward compatibility.

**Additional context**
This ofcourse has the consequence that 1.9 in opc server will be serialised as 1.8999999, but that is exactly the need of the user providing _fullprecisionfloatdouble_ parameter.

kiran-pradeep-palakkappilil, 2022-06-30:
@cristipogacean is the pr #1711 abandoned ?

cristipogacean, 2022-06-30:
@kiran-pradeep-palakkappilil, yes, we decided to put this on-hold for the moment. The attempt I started here, to fix the encoding of floats and doubles throughout all the IIoT platform, is very likely to cause too much trouble through breaking changes so we need to take a step back and look for an alternative.

marcschier, 2023-06-26:
Thanks to the simplification in 2.9 of the entire codebase we will now use G9 and G17 to roundtrip double/float from OPC UA (JsonEncoderEx).  Please note that in JSON we use the official 64bit double and 32 bit float string encoding then.  This does not mean you get mathematical string representation, but that if you parse the double/float in the JSON you get the original value back.  If your server has values with precision beyond IEEE754 double then the server should expose the value as string or "decimal" variant (the latter is something that almost no OPC UA clients support though).

PR #2007 (2023-06-26): Fix roundtrip double encoding in JsonEncoder
Fix #1709. We will now use G9 and G17 to roundtrip double/float from OPC UA (JsonEncoderEx).
closed 2023-06-26 by marcschier