Azure/Industrial-IoT #1723: Deploying Azure IIoT Solution on Azure Devops

Vkas47, 2022-05-26:
**Describe the bug**
Hi Team,

When we are trying to deploying the code in Azure Devops we are getting error in e2e-testing files. When we go through the same file there is no x86 mention on the file. We are not aware of what modification need to be done and where we have to recommended things. Below is the error output : -
--------------------------------------------------------------Error-----------------------------------------------------------------------------
##[error]D:\a\1\s\e2e-tests\IIoTPlatform-E2E-Tests\IIoTPlatform-E2E-Tests.sln.metaproj(0,0): Error MSB4126: The specified solution configuration "Release|x86" is invalid. Please specify a valid solution configuration using the Configuration and Platform properties (e.g. MSBuild.exe Solution.sln /p:Configuration=Debug /p:Platform="Any CPU") or leave those properties blank to use the default solution configuration
--------------------------------------------------------------Error----------------------------------------------------------------------------
**To Reproduce**
Steps to reproduce the behavior:
1. Go to  Azure Pipeline
2. Click on  Starter Pipeline it takes azure-pipeline.yml file from the branch where i clone Azure-IIoT source code start running the service
3. Scroll down to Failed Job output and look for the error
4. See error

**Expected behavior**
The Azure IIoT code should deploy end to end Azure IIoT Solution on Azure Cloud Platform

**Screenshots**
[image]

Vkas47, 2022-05-31:
Hi @Karok2m  we are trying to deploy this code with master branch. We are using starter pipeline in Azure Devops where default file of azure.pipeline.yml is picked by Azure devops by default and when we trying to run it it says error the following above error.

karok2m, 2022-05-31:
@Vkas47, can you please use `main` branch ? `master` branch is deprecated and not maintained any more. The development now happens in `main` branch. Also, please make sure that the correct pipeline file is picked up, as ours is called `azure-pipelines.yml` with `-` instead of `.`.

karok2m, 2022-09-30:
@Vkas47, were you able to change the default branch in Azure DevOps pipeline ? I am wondering if we can close this issue or there is anything else you need to get going.

karok2m, 2022-10-05:
Closing due to inactivity. @Vkas47, please reopen the issue if you need further assistance.