Azure/Industrial-IoT #1726: OPC Twin Module in IoT Edge StandAlone

albandrod, 2022-05-30:
Hi, I'm just developed an IoT Solution where I have deployed the IoT Edge with OPC Publisher module in standalone mode. Now, what I need is to write some information back to the OPC Server, I now I can do this with the OPC Twin module, I already deployed it in the Edge:

[image]

But now, i do not know how I can write information to the OPC node, I tried with Direct Methods, I have been reviewing the information from POSTMAN, but I do not know which information do I need to provide

https://github.com/Azure/Industrial-IoT/blob/main/docs/services/twin.md
https://github.com/Azure/Industrial-IoT/blob/main/docs/api/twin/readme.md

Can someone assist me? or do i need to deploy the industrial IoT solution to this purpose, because the rest of the system has already been deployed and working, if it's possible i do not want to undone the work that has already been done.

Thanks!

hansgschossmann, 2022-05-31:
@albandrod as our documentation states: https://docs.microsoft.com/en-us/azure/industrial-iot/overview-what-is-industrial-iot the OPC Twin module running standalone is an unsupported scenario.

marcschier, 2023-07-03:
OPC Publisher 2.9.0-preview4 contains OPC Twin and supports writing back to the OPC UA server using direct method API. This is documented under ./docs/opc-publisher/readme.md. Please let us know if this worked for you.

closed 2023-07-03 by marcschier