Azure/Industrial-IoT #1759: Got BadNodeIdUnknown errors in OPCPublisher

LanceLiMS, 2022-07-20:
**Describe the bug**
Hi MS team, our user got an issue and failed to send message to IoT Hub while using OPCPublisher module  (version 2.7.x)
Error message:

2022-07-18 08:11:48.021 +00:00 [WRN] Failed to add new monitored item 'ns=3;b="zC7Ve2pVnEefL/g8/0V2fxIAAAAAAAAA"' due to 'Cannot parse node id text: 'b="zC7Ve2pVnEefL/g8/0V2fxIAAAAAAAAA"''
2022-07-18 08:11:48.022 +00:00 [INF] Added 0 monitored items to subscription opc.tcp://WIN-8AKBBS5LISO:4852_c3c57879910cda18658d7aaf8ff227badca8f5e2
 
[image]

Published_nodes.json from Cx side: 

[
    {
        "EndpointUrl": "opc.tcp://WIN-8AKBBS5LISO:4852",
        "UseSecurity": true,
        "OpcAuthenticationMode": "UsernamePassword",
        "OpcAuthenticationUsername": "",
        "OpcAuthenticationPassword": "",
        "OpcNodes": [
            {
                "Id": "ns=3;b=gRWBX6BEbEWuznrr1phvwc0AAAAAAAAA",
                "DisplayName": "Total Counters",
                "OpcSamplingIntervalTimespan": 2000,
                "OpcPublishingIntervalTimespan": 5000,
                "HeartbeatIntervalTimespan": 3600,
                "QueueSize": 10
            }
        ]
    }
]
 
Previously we though it was caused by json formatting issue, we tried using "Id": "ns=3;b=\"gRWBX6BEbEWuznrr1phvwc0AAAAAAAAA\"", but still failed.

[image]

The funny thing is that if we connect to the endpoint of Prosys OPC UA Simulation Server with similar published_nodes.json file, everything works as expected (We performed a test).
The OPC UA server from Cx side is:

[image]

 
Please let me know if you have any thoughts on this issue. Thanks in advance!

cristipogacean, 2022-07-20:
The nodeId here is an opaque type one. This means that the identifier itself is a byte array. For the OPC Publisher to properly parse opaque NodeIds defined in the configuration file, the id (byte array that follows the `b=` ) must be the string format of the 64 encoded value applied on the source byte array. No quote characters are expected, just the plain 64bit encoded string.
E.g.: if the opaque identifier is `[0x61, 0x62, 0x63]`, the node Id in the config file shall be: `b=YWJj`.

Any generic OPC UA Client like UA Expert in the example below will display the nodeId in the right format required by the publisher. 

[image]

The error you showed here is because the quote character is not a valid character for a 64 bit encoded string.

cristipogacean, 2022-07-26:
Eventually the issue here was the fact that the node configured to be published was only supporting HistoryRead and no CurrentRead access levels. Because of this, the node cannot be monitored respectively subscribed on.

closed 2022-07-26 by cristipogacean