Azure/Industrial-IoT #1765: Data delay issues

BGbakker, 2022-08-05:
We are running into some data delay issues, and were looking for the place where the delay is taking place.

Setup:
We have a single hub with 2 edge devices connected. 
The first edge had 5 endpoints connected to an OPC server. The second has 10 endpoints connected to 2 OPC servers (5 each). 
We are experiencing a data delay on the second edge that we can’t explain. All data coming over 3 of the 10 endpoints was delayed up to 8 hours.
Both edge devices are running version 2.8.3  and running with sufficient resources. 

History:
Before we have seen that if some devices connected to the OPC server had connection issues, that this produced so much data that it caused the delay of all data of the devices connected to that (and only that) endpoint. Fixing the connection issue (or disconnecting the device) fixed the data overload.

Issue:
The first edge had a device with connection issues that caused a delay in data as we had seen before. 
The second edge also displayed a delay of about the same time (both around 8 hours) but no excess data was being produced by the devices. This delay was seen on all data from devices connected to 3 of the endpoints. 

We are looking for the point of delay but are so far unsuccessful. Does anyone have some suggestions on where the delay can be found?

cristipogacean, 2022-08-05:
can you please share some details on:
- where you see the delay? is it at edgeHub's level or at IoTHub?
- is it possible that the IoT hub runs out of message quota during the day?
- are the messages pile up in the edgeHub because of upstream limitations?
- how is the configuration of the publishers: e.g standalone or orchestrated, is batching enabled 

I think the issue is throughput throttling somewhere on the upstream.

BGbakker, 2022-08-08:
Hi, 
First an update: The delay issue resolved itself over all endpoints when we disconnected the device producing excessive data. At this moment we are looking for options to check historic data or points to check when the issue reappears.

We notice the delay when using the data. For use we save the data in a database and record both the time of the data and the time of inserting the data into the database. Normally the time difference here is a few seconds. The time of alle the devices are the same.
Whether this delay is at the edgeHub, IoTHub, or data processing, that is what we are trying to find.

No we did not run into the message quota.

There is no indication in the logging of the edgeHub that there was a massage pileup. Also the diagnostic information of the publishers showed no message in the outgress buffer.

The publishers are orchestrated. Batching is enabled by default and the settings are unchanged.

If you know of other methods to check a message pileup on the edgeHub or any other point please let us know. As mentioned before, we are looking for where the data is delayed.

cristipogacean, 2022-08-09:
you can get more information from edgeHub module if you enable the `Debug` log level by adding the `RuntimeLogLevel = Debug` environment variable to the edgeHub's deployment manifest. You ca do this in portal on the edge device's `Set Modules` configuration like in the following print-screen:
[image]

hansgschossmann, 2022-08-30:
@BGbakker on your second edge device with the delayed data: was all data delayed or only data from certain endpoints? I understand from your description above that it is only certain endpoints data.

You say "disconnected the device producing excessive data." What does excessive data mean? Does device mean OPC UA Server here? If yes: how many endpoints of the 10 endpoints are from this server? Is one of the 3 endpoints with delayed data from this server?

BGbakker, 2022-08-30:
Hi,
No not all data was delayed, only data from certain endpoints. 

"disconnected the device producing excessive data." does not mean disconnecting the OPC UA server. The OPC Server connects to multiple PLC's. We disconnect the PLC's from the OPC server and remove the corresponding  tags from the endpoint configuration.

If you have any more questions let me know.

monikavar, 2022-09-27:
Hope the above suggestion helped. Closing the issue due to inactivity. 
Please feel free to re-open if the issue exists.

closed 2022-09-27 by monikavar