Azure/Industrial-IoT #1766: Interval based send suggestion needed

b-enigma-con, 2022-08-11:
Hello all,

We're looking for some suggestions. We want the OPC Publisher to periodically send OPC values to the cloud, and not on a datachange. We have used the heartbeat functionality which seems to be doing more or less what we need, but data changes are sent as well. If we have a heartbeat every 10 seconds and the value changes in between two heartbeats, we receive an extra message for the datachange in between heartbeats, which we don't want.

So the heartbeat only works partly for us, we don't want that message on a data change. I've looked at the possibility to set the DataChangeFilter / DataChangeTrigger for this purpose , but it does not seem to be possible (yet?). 

I don't feel much for creating a customized version of the OPC Publisher at this point. So basically the question is: how can we let the OPC Publisher only sent values periodically, without the data changes themselves? As always, thank you in advance!

cristipogacean, 2022-08-11:
Hello @b-enigma-con,

What if you set the PublishingInterval = SamplingInterval = heartbeat interval? e.g. 10 seconds like in your example.
The publisher will then only generate data-changes in the same cycle as heartbeats.

b-enigma-con, 2022-08-30:
Sounds like a good suggestion, I will try this and let you know. Thank you.

b-enigma-con, 2022-09-07:
Not yet (other priorities came up), but will let you know the outcome. I expect to pick this up in the coming 2 weeks.

monikavar, 2022-09-27:
@b-enigma-con Closing this issue due to inactivity. Please re-open if needed.

b-enigma-con, 2022-09-30:
I finally managed to find some time to test this suggestion. I can confirm that it works exactly as I wanted.

marcschier, 2023-07-03:
In 2.9.0-preview4 we support cyclic reads now. "UseCyclicRead": true on a node id will enable this cacapbility.  Let me know if this works for you.

closed 2023-07-03 by marcschier