Azure/Industrial-IoT #1767: BadTimeout exception during publish due to sequenceNumber mismach

dimtix, 2022-08-12:
**Describe the bug**
We are using publisher (2.8.2) in standalone mode. After updating the pn.json (same data items, same server, new DataSetWriterId - new test in our E2E test pipeline) we sometimes observe that publishing stops or is no longer possible because a timeout exception is constantly thrown. Restarting of the publisher fixes the problem.

**To Reproduce**
Steps to reproduce the behavior:
1. replace pn.json (same content different DataSetWriterIds)
2. publishing stops

**Expected behavior**
Publishing is possible without publisher restart - no Timeout exceptions.

Unfortunately, the error is not reproducible. Happens only from time to time.

Could you explain the behavior? What can be done about it? Is it a bug in publisher or in the OPC UA stack?

Corresponding logs:
'2022-08-10T20:59:28.093566790Z [2022-08-10 20:59:28.093 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Refreshing done for session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7' 
2022-08-10T20:59:28.106508260Z [2022-08-10 20:59:28.106 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7': server current state: Running 
2022-08-10T20:59:29.318226166Z [2022-08-10 20:59:29.317 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7': server current state: Running 
2022-08-10T20:59:29.318324167Z [2022-08-10 20:59:29.317 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7', subscription 'SingleWriterW3ee5539f421f4972b9a446430f79a0b7_0' - sequence# 6-08/10/2022 20:59:29 
2022-08-10T20:59:29.318330967Z [2022-08-10 20:59:29.317 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] DataChange for subscription 'SingleWriterW3ee5539f421f4972b9a446430f79a0b7_0'/'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7' has unexpected sequenceNumber 6 vs expected 1, isKeepAlive True, publishTime 08/10/2022 20:59:29 
2022-08-10T20:59:29.318385268Z [2022-08-10 20:59:29.318 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) PUBLISH #11 - Unhandled error during Publish.
2022-08-10T20:59:29.318389968Z  BadTimeout 'BadTimeout'
2022-08-10T20:59:29.318393368Z  
2022-08-10T20:59:29.318396868Z Opc.Ua.ServiceResultException: BadTimeout
2022-08-10T20:59:29.318400668Z    at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
2022-08-10T20:59:29.318405868Z    at Opc.Ua.SessionClient.EndPublish(IAsyncResult result, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
2022-08-10T20:59:29.318409968Z    at Opc.Ua.Client.Session.OnPublishComplete(IAsyncResult result)
2022-08-10T20:59:31.368463479Z [2022-08-10 20:59:31.368 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7': server current state: Running 
2022-08-10T20:59:31.368683082Z [2022-08-10 20:59:31.368 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7', subscription 'SingleWriterW3ee5539f421f4972b9a446430f79a0b7_1000' - sequence# 3-08/10/2022 20:59:31 
2022-08-10T20:59:31.368886985Z [2022-08-10 20:59:31.368 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] DataChange for subscription 'SingleWriterW3ee5539f421f4972b9a446430f79a0b7_1000'/'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7' with sequenceNumber 3, isKeepAlive True, publishTime 08/10/2022 20:59:31 
2022-08-10T20:59:31.369025086Z [2022-08-10 20:59:31.368 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Found 1 notifications with null value or not good status code for 'SingleWriterW3ee5539f421f4972b9a446430f79a0b7_1000'/'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7' subscription. 
2022-08-10T20:59:31.369122188Z [2022-08-10 20:59:31.369 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7 received message with sequenceNumber 6 
2022-08-10T20:59:32.357676861Z [2022-08-10 20:59:32.357 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Sink block in engine opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7 triggered with 1 messages 
2022-08-10T20:59:35.309186709Z [2022-08-10 20:59:35.308 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW64a4168558ac472baf88d9113c93fe6c': server current state: Running 
2022-08-10T20:59:36.371601376Z [2022-08-10 20:59:36.371 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7': server current state: Running 
2022-08-10T20:59:36.371922681Z [2022-08-10 20:59:36.371 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7', subscription 'SingleWriterW3ee5539f421f4972b9a446430f79a0b7_1000' - sequence# 3-08/10/2022 20:59:36 
2022-08-10T20:59:36.372202084Z [2022-08-10 20:59:36.371 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] DataChange for subscription 'SingleWriterW3ee5539f421f4972b9a446430f79a0b7_1000'/'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7' has unexpected sequenceNumber 3 vs expected 4, isKeepAlive False, publishTime 08/10/2022 20:59:36 
2022-08-10T20:59:36.372344186Z [2022-08-10 20:59:36.372 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Found 1 notifications with null value or not good status code for 'SingleWriterW3ee5539f421f4972b9a446430f79a0b7_1000'/'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7' subscription. 
2022-08-10T20:59:36.372387287Z [2022-08-10 20:59:36.372 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7 received message with sequenceNumber 7 
2022-08-10T20:59:37.363153326Z [2022-08-10 20:59:37.362 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Sink block in engine opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7 triggered with 1 messages 
2022-08-10T20:59:38.089600939Z [2022-08-10 20:59:38.089 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Runner Keepalive reset due to checkAlive  
2022-08-10T20:59:38.089914843Z [2022-08-10 20:59:38.089 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Refreshing session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW64a4168558ac472baf88d9113c93fe6c' 
2022-08-10T20:59:38.089927244Z [2022-08-10 20:59:38.089 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Refreshing session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7' 
2022-08-10T20:59:38.090469751Z [2022-08-10 20:59:38.090 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Refreshing done for session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7' 
2022-08-10T20:59:38.090482351Z [2022-08-10 20:59:38.090 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Refreshing done for session 'opc.tcp://localhost:21878/_8E33560C_SingleWriterW64a4168558ac472baf88d9113c93fe6c' 
[... cut ...]
2022-08-10T20:59:40.325115709Z [2022-08-10 20:59:40.324 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Event for subscription: SingleWriterW3ee5539f421f4972b9a446430f79a0b7_0, sequence#: 6 isKeepAlive: False, publishTime: 08/10/2022 20:59:40 
2022-08-10T20:59:40.325123109Z [2022-08-10 20:59:40.324 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7 received message with sequenceNumber 4 '

cristipogacean, 2022-08-12:
@dimtix, would it be possible for you to try to replicate the same issue with 2.8.3? in case you still see this, please share some more details on the configuration you're using. 
BTW, I find a bit strange that the server url points to `localhost`. Isn't this causing connectivity issues?

dimtix, 2022-08-12:
@cristipogacean the server and publisher are running in the same k8s pod - the server is accessible via localhost.
Are the changes from 2.8.2 to 2.8.3 that substantial? Unfortunately it is not so easy to update our test setup. But I would do it of cause if there is any chance that it fixes the issue

cristipogacean, 2022-08-12:
@dimtix, in 2.8.3 the opc ua stack was updated because to some serious security issues. With the new version, we expose now stack logs in publisher's log. This could help us debug further. I have no idea though if updating will help at all for your issue, but it's highly recommended due to the security fixes.

We just had a different issue with publish timeouts and here the cause was apparently the publisher host's and the opc ua server clocks were completely out of sync. Resyncing them solved the issue in that case. It might not be the case here, but maybe it triggers some ideas on your side. 

regarding the sequence number expected vs actual, especially this log is strange: 
`2022-08-10T20:59:29.318330967Z [2022-08-10 20:59:29.317 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] DataChange for subscription 'SingleWriterW3ee5539f421f4972b9a446430f79a0b7_0'/'opc.tcp://localhost:21878/_8E33560C_SingleWriterW3ee5539f421f4972b9a446430f79a0b7' has unexpected sequenceNumber 6 vs expected 1, isKeepAlive True, publishTime 08/10/2022 20:59:29`
 - it seems the publisher expects a new subscription was created, but the-server still sends data changes to the old one.

Another thing I noticed: it seems you're using publishing interval 0 here: `SingleWriterW3ee5539f421f4972b9a446430f79a0b7_0` is this intentional? even though 0 is a legit publishing interval, it may trigger eventually such publish timeouts.

dimtix, 2022-08-12:
How can you see that the publishing interval is 0?

Indeed with the publishing interval set 0:
`[
    {
        "EndpointUrl": "opc.tcp://localhost:21878/",
        "DataSetWriterGroup": "SingleWriterWbb2122d3b89d470cbb318f6cb3f537bc",
        "DataSetWriterId": "SingleWriterWbb2122d3b89d470cbb318f6cb3f537bc",
        "OpcNodes": [
            {
                "Id": "nsu=http://test.com;s=33",
                "OpcSamplingInterval": 0,
                "OpcPublishingInterval": 0,
                "DisplayName": "test_s_33"
            }
        ]
    }
]`

I **always** had this problem with the sequenceNumber mismatch and thrown BadTimeOut exception. But it was more my fault (no intention) and I removed the setting. In that case I think the default setting of 1000ms is taken. You can see that in the logs:
`2022-08-10T20:59:08.355014084Z [2022-08-10 20:59:08.354 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Subscription 'ModelSingleWriterW3ee5539f421f4972b9a446430f79a0b7_0'/'opc.tcp://localhost:21878/_8E33560C_ModelSingleWriterW3ee5539f421f4972b9a446430f79a0b7' state actual(revised)/desired: PublishingEnabled True/True, PublishingInterval 1000/1000, KeepAliveCount 10/10, LifetimeCount 10000/10000 `

Unfortunately, the removal of  "OpcPublishingInterval": 0 has only partially helped and the error reoccurs from time to time - I would say in 30% of runs.

hansgschossmann, 2022-08-30:
@dimtix @cristipogacean is on vacation right now.

i think the publishing interval value is the last part of the subscription reported in the log: SingleWriterW3ee5539f421f4972b9a446430f79a0b7_0, but the log after your update gives a different indication.

your log also says "OpcSamplingInterval: 0". be aware that this might kill some OPC UA servers and makes no sense from my point of view, since it is not deterministic at all.

above you says "the server and publisher are running in the same k8s pod": pls note that this is not a supported configuration.

monikavar, 2022-09-27:
Closing due to inactivity. Please reactivate if the issue exists.

Mat-pa, 2022-11-04:
We are also experiencing a strange behaviour with unexpected sequenceNumber 

`[2022-11-04 17:44:46.863 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] DataChange for subscription '1000_ns=2;s=[redacted identifier]'/'opc.tcp:/[IP:Port, redacted]_1AB1XZ1' has unexpected sequenceNumber 52 vs expected 53, isKeepAlive False, publishTime 11/04/2022 17:45:21
`

We are on version 2.8.3 and this also occurs after docker restart OPCPublisher

Mat-pa, 2022-11-04:
> Closing due to inactivity. Please reactivate if the issue exists.

@monikavar please reactivate

marcschier, 2023-01-16:
Logging sequence number errors has been addressed in 2.8.4.  It happens when we receive keep alive messages which contain the last sequence number, and not an incremented one.  This is per spec, and we updated the logs to not show this condition.

closed 2023-01-16 by marcschier