Azure/Industrial-IoT #1768: publisher: Behavior of "OpcSamplingInterval" in pn.json

codeminic, 2022-08-26:
**Describe the bug**
Not sure if this is a bug or desired behavior. When setting the `OpcSamplingInterval` to a value lower than the `OpcPublishingInterval`, the module only emits a single value, even if there is more than one change of the value within the time specified in `OpcPublishingInterval`. 
The structure of the output allows multiple values for the same `NodeId`, that's why i assumed that it will output multiple values when setup as discribed above. 

**To Reproduce**
Steps to reproduce the behavior:
1. Setup a pn.json with one node, set the `OpcSamplingInterval` to a smaller value than the `OpcPublishingInterval`
2. Run the publisher module
3. Check the output

**Expected behavior**
When setting the `OpcSamplingInterval` to a value lower than the `OpcPublishingInterval` and there are multiple changes for the configured node within the time of `OpcPublishingInterval`, the publisher module should send more than one value for the configured node to the ouput.

**Configuration**

Content of the `pn.json`:
```json
[
  {
    "EndpointUrl": "opc.tcp://uademo.prosysopc.com:53530/OPCUA/SimulationServer",
    "UseSecurity": false,
    "OpcNodes": [
      {
        "Id": "ns=3;i=1001",
        "OpcSamplingInterval": 2000,
        "OpcPublishingInterval": 10000,
        "DisplayName": "Counter",
        "HeartbeatInterval": 20
      }
    ]
  }
]
```

Content of the `debug.template.json`:
```json
{
  "$schema-template": "2.0.0",
  "modulesContent": {
    "$edgeAgent": {
      "properties.desired": {
        "schemaVersion": "1.1",
        "runtime": {
          "type": "docker",
          "settings": {
            "minDockerVersion": "v1.25",
            "loggingOptions": "",
            "registryCredentials": {
              "crstdpeiot": {
                "username": "xxx",
                "password": "xxx",
                "address": "xxx"
              }
            }
          }
        },
        "systemModules": {
          "edgeAgent": {
            "type": "docker",
            "settings": {
              "image": "mcr.microsoft.com/azureiotedge-agent:1.3",
              "createOptions": {}
            }
          },
          "edgeHub": {
            "type": "docker",
            "status": "running",
            "restartPolicy": "always",
            "settings": {
              "image": "mcr.microsoft.com/azureiotedge-hub:1.3",
              "createOptions": {
                "HostConfig": {
                  "PortBindings": {
                    "5671/tcp": [
                      {
                        "HostPort": "5671"
                      }
                    ],
                    "8883/tcp": [
                      {
                        "HostPort": "8883"
                      }
                    ],
                    "443/tcp": [
                      {
                        "HostPort": "443"
                      }
                    ]
                  }
                }
              }
            }
          }
        },
        "modules": {
          "publisher": {
            "settings": {
              "image": "mcr.microsoft.com/iotedge/opc-publisher:2.8",
              "createOptions": {
                "Hostname": "publisher",
                "Cmd": [
                  "--pf=/appdata/pn.json",
                  "--di=60",
                  "--me=Json",
                  "--mm=PubSub",
                  "--tm",
                  "--aa",
                  "--si=2",
                  "--ms=262144",
                  "--BypassCertVerification=true",
                  "--ll=debug"
                ],
                "HostConfig": {
                  "Binds": [
                    "C:\\temp\\publisher:/appdata"
                  ]
                }
              }
            },
            "type": "docker",
            "version": "1.0",
            "status": "running",
            "restartPolicy": "always"
          }
        }
      }
    },
    "$edgeHub": {
      "properties.desired": {
        "schemaVersion": "1.1",
        "routes": {
          "PublisherToLogger": "FROM /messages/modules/publisher/* INTO $upstream"
        },
        "storeAndForwardConfiguration": {
          "timeToLiveSecs": 7200
        }
      }
    }
  }
}
```

**Desktop (please complete the following information):**
 - OS: Windows 11, Ubuntu 20.04
 - Browser -
 - Version: 2.8

Thx for your feedback!

codeminic, 2022-08-26:
@JoshouaDPE FYI

hansgschossmann, 2022-08-30:
@codeminic @JoshouaDPE 

What you describe is as expected using your configuration.

The behavior is described in more detail in the [OPC UA specification](https://reference.opcfoundation.org/Core/Part4/5.12.1/).

To achieve what you are looking for you need to set the QueueSize parameter in the node configuration. I think this is missing in the documentation of the OPC Publisher configuration on docs.microsoft.com and on github, but i need to validate when the team is back from vacation. 

See [this file](https://github.com/Azure/Industrial-IoT/blob/main/components/opc-ua/src/Microsoft.Azure.IIoT.OpcUa/src/Publisher/Models/OpcNodeModel.cs) for the configuration for a node via the configuration file. This would do exactly what you are looking for. Can you validate if this is working as you expect and let me know here?

The command line parameter to change the default setting of the QueueSize (default is 1) is `DefaultQueueSize=VALUE` documented [here](https://docs.microsoft.com/en-us/azure/industrial-iot/reference-command-line-arguments?view=iotedge-1.4).

codeminic, 2022-09-05:
@hansgschossmann explicitly setting the the `QueueSize` to a higher value worked, thanks!

hansgschossmann, 2022-09-06:
good to hear. you are welcome. keeping it open to fix the doc.

closed 2023-01-16 by marcschier