Azure/Industrial-IoT #1769: OPCPublisher unable to push data to IoT Hub

spinshin, 2022-08-30:
I am trying to follow [this guide](https://docs.microsoft.com/en-us/azure/digital-twins/how-to-ingest-opcua-data) to push data to IoT Hub from Prosys Simulation Server. However, the publisher module is not pushing the data into IoT Hub.

<details>
  <summary>iotedge list</summary>

```
NAME             STATUS           DESCRIPTION      Config
OPCPublisher     running          Up an hour       mcr.microsoft.com/iotedge/opc-publisher:latest
edgeAgent        running          Up an hour       mcr.microsoft.com/azureiotedge-agent:1.1
edgeHub          running          Up an hour       mcr.microsoft.com/azureiotedge-hub:1.1
```

</details>

<details>
  <summary>iotedge check --verbose</summary>
  
```
  Configuration checks (aziot-identity-service)
  ---------------------------------------------
  √ keyd configuration is well-formed - OK
  √ certd configuration is well-formed - OK
  √ tpmd configuration is well-formed - OK
  √ identityd configuration is well-formed - OK
  √ daemon configurations up-to-date with config.toml - OK
  √ identityd config toml file specifies a valid hostname - OK
  √ aziot-identity-service package is up-to-date - OK
  √ host time is close to reference time - OK
  √ preloaded certificates are valid - OK
  √ keyd is running - OK
  √ certd is running - OK
  √ identityd is running - OK
  √ read all preloaded certificates from the Certificates Service - OK
  √ read all preloaded key pairs from the Keys Service - OK
  √ check all EST server URLs utilize HTTPS - OK
  √ ensure all preloaded certificates match preloaded private keys with the same ID - OK
  
  Connectivity checks (aziot-identity-service)
  --------------------------------------------
  √ host can connect to and perform TLS handshake with iothub AMQP port - OK
  √ host can connect to and perform TLS handshake with iothub HTTPS / WebSockets port - OK
  √ host can connect to and perform TLS handshake with iothub MQTT port - OK
  
  Configuration checks
  --------------------
  √ aziot-edged configuration is well-formed - OK
  √ configuration up-to-date with config.toml - OK
  √ container engine is installed and functional - OK
  √ configuration has correct URIs for daemon mgmt endpoint - OK
  √ aziot-edge package is up-to-date - OK
  √ container time is close to host time - OK
  √ DNS server - OK
  √ production readiness: logs policy - OK
  √ production readiness: Edge Agent's storage directory is persisted on the host filesystem - OK
  √ production readiness: Edge Hub's storage directory is persisted on the host filesystem - OK
  √ Agent image is valid and can be pulled from upstream - OK
  √ proxy settings are consistent in aziot-edged, aziot-identityd, moby daemon and config.toml - OK
  
  Connectivity checks
  -------------------
  √ container on the default network can connect to upstream AMQP port - OK
  √ container on the default network can connect to upstream HTTPS / WebSockets port - OK
  √ container on the default network can connect to upstream MQTT port - OK
      skipping because of not required in this configuration
  √ container on the IoT Edge module network can connect to upstream AMQP port - OK
  √ container on the IoT Edge module network can connect to upstream HTTPS / WebSockets port - OK
  √ container on the IoT Edge module network can connect to upstream MQTT port - OK
      skipping because of not required in this configuration
  35 check(s) succeeded.
  2 check(s) were skipped due to errors from other checks.
[... 31 lines cut ...]
[2022-08-30 06:56:26.418 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Stopping worker supervisor
[2022-08-30 06:56:26.418 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Worker supervisor successfully stopped
[2022-08-30 06:56:26.419 INF Microsoft.Azure.IIoT.Http.HealthChecks.HealthCheckManager] Health checks stopped.
[2022-08-30 06:56:26.421 INF Root] Stopped prometheus metric server
[2022-08-30 06:56:26.421 INF Root] Module stopped.
[2022-08-30 06:56:26.421 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Stopping worker supervisor
[2022-08-30 06:56:26.421 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Worker supervisor successfully stopped
[2022-08-30 06:56:26.424 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to opcpublisher using MqttOverTcp.
[2022-08-30 06:56:26.430 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.StandaloneJobOrchestrator] File /appdata/publishednodes.json has changed, last known hash , new hash D486B9AE83A2689672F63279E456D6D27DD0DB8FFCF7E58EC2EC96B629B5B782, reloading...
[2022-08-30 06:56:26.430 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.StandaloneJobOrchestrator] Validation schema file  does not exist or is disabled, ignoring validation of /appdata/publishednodes.json file.
[2022-08-30 06:56:26.436 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Read 1 entry models from published nodes file in 00:00:00.0053532
[2022-08-30 06:56:26.436 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Total count of OpcNodes after job conversion: 6
[2022-08-30 06:56:26.437 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Converted published nodes entry models to jobs in 00:00:00.0007390
[2022-08-30 06:56:26.437 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.StandaloneJobOrchestrator] Job opc.tcp://opcserver.example.com:53530/OPCUA/SimulationServer_33F44328 loaded with dataSetGroup null with dataSetWriters 1000
[2022-08-30 06:56:26.465 INF Root] Starting module OpcPublisher version 2.8.3.1.
[2022-08-30 06:56:26.465 INF Root] Initiating prometheus at port 9702/metrics
[2022-08-30 06:56:26.469 INF Root] Prometheus metric server started.
[2022-08-30 06:56:26.470 INF Microsoft.Azure.IIoT.Http.HealthChecks.HealthCheckManager] Health checks started.
[2022-08-30 06:56:26.470 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client.
[2022-08-30 06:56:26.470 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context.
[2022-08-30 06:56:26.487 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module edgedevice_OPCPublisher disconnected due to Communication_Error - now Disconnected_Retrying...
[2022-08-30 07:01:31.365 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module edgedevice_OPCPublisher disconnected due to Retry_Expired - now Disconnected...
```

</details>

hansgschossmann, 2022-08-30:
@spinshin looks like OPC Publisher can not connect using MQTT, which is weird looking at the check results.

What is also a bit strange is that the check is referring to IoT Edge 1.2 terms (config.toml, *-service) but the edgeAgent, edgeHub tags are 1.1.

Can you provide a support-bundle?

spinshin, 2022-08-31:
> @spinshin looks like OPC Publisher can not connect using MQTT, which is weird looking at the check results.
> 
> What is also a bit strange is that the check is referring to IoT Edge 1.2 terms (config.toml, *-service) but the edgeAgent, edgeHub tags are 1.1.
> 
> Can you provide a support-bundle?

Here it is: 
[support_bundle_2022_08_31_03_58_03_UTC.zip](https://github.com/Azure/Industrial-IoT/files/9458384/support_bundle_2022_08_31_03_58_03_UTC.zip)

spinshin, 2022-08-31:
In case you want to see the config file format.
```
auto_reprovisioning_mode = "OnErrorOnly"

[provisioning]
source = "manual"
connection_string = "<connection-string>"

[aziot_keys]

[preloaded_keys]

[cert_issuance]

[preloaded_certs]

[tpm]

[agent]
name = "edgeAgent"
type = "docker"
imagePullPolicy = "on-create"

[agent.config]
image = "mcr.microsoft.com/azureiotedge-agent:1.4"

[agent.config.createOptions]

[agent.env]

[connect]
workload_uri = "unix:///var/run/iotedge/workload.sock"
management_uri = "unix:///var/run/iotedge/mgmt.sock"

[listen]
workload_uri = "fd://aziot-edged.workload.socket"
management_uri = "fd://aziot-edged.mgmt.socket"
min_tls_version = "tls1.0"

[watchdog]
max_retries = "infinite"

[moby_runtime]
uri = "unix:///var/run/docker.sock"
network = "azure-iot-edge"
```

hansgschossmann, 2022-08-31:
how does your deploment manifest looks like. in your config.toml your are specifying an edgeAgent image of 1.4: "mcr.microsoft.com/azureiotedge-agent:1.4" and your deployment manifest seems to specify "mcr.microsoft.com/azureiotedge-agent:1.1"

To be consistent pls do install the IoT Edge version 1.1 and retry with the deployment manifest you have.

spinshin, 2022-09-05:
Can you have a look at this as well? I made sure that everything has the version 1.4. Data flow messages can be seen in the OPCPublisher logs as well. However, I can't see any data being received by the IoT Hub in the monitor.

<details>
  <summary>sudo iotedge list</summary>

```
NAME             STATUS           DESCRIPTION      Config
OPCPublisher     running          Up 2 days        mcr.microsoft.com/iotedge/opc-publisher:latest
edgeAgent        running          Up 2 days        mcr.microsoft.com/azureiotedge-agent:1.4
edgeHub          running          Up 2 days        mcr.microsoft.com/azureiotedge-hub:1.4
```

</details>

<details>
  <summary>sudo iotedge check --verbose</summary>

```
√ host time is close to reference time - OK
√ preloaded certificates are valid - OK
√ keyd is running - OK
√ certd is running - OK
√ identityd is running - OK
√ read all preloaded certificates from the Certificates Service - OK
√ read all preloaded key pairs from the Keys Service - OK
√ check all EST server URLs utilize HTTPS - OK
√ ensure all preloaded certificates match preloaded private keys with the same ID - OK

Connectivity checks (aziot-identity-service)
--------------------------------------------
√ host can connect to and perform TLS handshake with iothub AMQP port - OK
√ host can connect to and perform TLS handshake with iothub HTTPS / WebSockets port - OK
√ host can connect to and perform TLS handshake with iothub MQTT port - OK

Configuration checks
--------------------
√ aziot-edged configuration is well-formed - OK
√ configuration up-to-date with config.toml - OK
√ container engine is installed and functional - OK
√ configuration has correct URIs for daemon mgmt endpoint - OK
√ aziot-edge package is up-to-date - OK
√ container time is close to host time - OK
√ DNS server - OK
√ production readiness: logs policy - OK
√ production readiness: Edge Agent's storage directory is persisted on the host filesystem - OK
√ production readiness: Edge Hub's storage directory is persisted on the host filesystem - OK
√ Agent image is valid and can be pulled from upstream - OK
√ proxy settings are consistent in aziot-edged, aziot-identityd, moby daemon and config.toml - OK

Connectivity checks
-------------------
√ container on the default network can connect to upstream AMQP port - OK
√ container on the default network can connect to upstream HTTPS / WebSockets port - OK
√ container on the default network can connect to upstream MQTT port - OK
    skipping because of not required in this configuration
√ container on the IoT Edge module network can connect to upstream AMQP port - OK
√ container on the IoT Edge module network can connect to upstream HTTPS / WebSockets port - OK
√ container on the IoT Edge module network can connect to upstream MQTT port - OK
    skipping because of not required in this configuration
35 check(s) succeeded.
2 check(s) were skipped due to errors from other checks.
```

</details>

<details>
  <summary>/etc/aziot/config.toml</summary>

```
auto_reprovisioning_mode = "OnErrorOnly"

[provisioning]
source = "manual"
connection_string = "HostName=cyberhub.azure-devices.net;DeviceId=iotedge;SharedAccessKey=X3N6S74CHdWtbJonTz+yo5dDnYvUwJ/bgjBNpUwFoX8="

[aziot_keys]

[preloaded_keys]

[cert_issuance]

[preloaded_certs]

[tpm]

[agent]
name = "edgeAgent"
type = "docker"
imagePullPolicy = "on-create"

[agent.config]
image = "mcr.microsoft.com/azureiotedge-agent:1.4"

[agent.config.createOptions]

[agent.env]

[connect]
workload_uri = "unix:///var/run/iotedge/workload.sock"
management_uri = "unix:///var/run/iotedge/mgmt.sock"

[listen]
workload_uri = "fd://aziot-edged.workload.socket"
management_uri = "fd://aziot-edged.mgmt.socket"
min_tls_version = "tls1.0"

[watchdog]
max_retries = "infinite"

[moby_runtime]
uri = "unix:///var/run/docker.sock"
network = "azure-iot-edge"
```

</details>

[support_bundle_2022_09_05_04_50_18_UTC.zip](https://github.com/Azure/Industrial-IoT/files/9486769/support_bundle_2022_09_05_04_50_18_UTC.zip)

hansgschossmann, 2022-09-06:
@spinshin  the support_bundle looks good. With monitor you mean the "Messages used today" count in the Overview of the IoT Hub portal page?

Can you restart edgeHub and provide the first few hunderd lines of log output?

spinshin, 2022-09-06:
> With monitor you mean the "Messages used today" count in the Overview of the IoT Hub portal page?

By monitor, I mean that I can't see any messages being received when I monitor the hub using the az cli:
 `az iot hub monitor-events -n <iot-hub-instance> -t 0`

> edgeHub logs

```
<6> 2022-09-06 19:25:06.358 +00:00 [INF] - Termination requested, initiating shutdown.
<6> 2022-09-06 19:25:06.360 +00:00 [INF] - Stopping the protocol heads...
<6> 2022-09-06 19:25:06.361 +00:00 [INF] - Closing protocol heads - (MQTT, AMQP, HTTP)
<6> 2022-09-06 19:25:06.362 +00:00 [INF] - Stopping MQTT protocol head
<6> 2022-09-06 19:25:06.382 +00:00 [INF] - Closing HTTP head
<6> 2022-09-06 19:25:06.398 +00:00 [INF] - Waiting for cleanup to finish
<6> 2022-09-06 19:25:06.400 +00:00 [INF] - Closed HTTP head
2022-09-06 19:25:16  Starting Edge Hub
2022-09-06 19:25:16  Starting Edge Hub
2022-09-06 19:25:16  Changing ownership of storage folder: /iotedge/edgeHub/edgeHub to 13623
2022-09-06 19:25:16  Changing ownership of backup folder: /tmp/edgeHub_backup to 13623
2022-09-06 19:25:17.027 +00:00 Edge Hub Main()
<6> 2022-09-06 19:25:17.577 +00:00 [INF] - Installing certificates [CN=aziot-edge CA iotedge:12/01/2022 06:20:00] to Root
<6> 2022-09-06 19:25:17.628 +00:00 [INF] - Installing certificates [CN=aziot-edge CA iotedge:12/01/2022 06:20:00] to Root
<6> 2022-09-06 19:25:17.635 +00:00 [INF] - Enabling SSL protocols: Tls12
<6> 2022-09-06 19:25:17.748 +00:00 [INF] - Experimental features configuration: {"Enabled":false,"DisableCloudSubscriptions":false,"DisableConnectivityCheck":false}
<6> 2022-09-06 19:25:18.134 +00:00 [INF] - Created persistent store at /iotedge/edgeHub/edgeHub
<6> 2022-09-06 19:25:18.171 +00:00 [INF] - Initializing Edge Hub
<6> 2022-09-06 19:25:18.171 +00:00 [INF] -
        █████╗ ███████╗██╗   ██╗██████╗ ███████╗
       ██╔══██╗╚══███╔╝██║   ██║██╔══██╗██╔════╝
       ███████║  ███╔╝ ██║   ██║██████╔╝█████╗
       ██╔══██║ ███╔╝  ██║   ██║██╔══██╗██╔══╝
       ██║  ██║███████╗╚██████╔╝██║  ██║███████╗
       ╚═╝  ╚═╝╚══════╝ ╚═════╝ ╚═╝  ╚═╝╚══════╝

 ██╗ ██████╗ ████████╗    ███████╗██████╗  ██████╗ ███████╗
 ██║██╔═══██╗╚══██╔══╝    ██╔════╝██╔══██╗██╔════╝ ██╔════╝
 ██║██║   ██║   ██║       █████╗  ██║  ██║██║  ███╗█████╗
 ██║██║   ██║   ██║       ██╔══╝  ██║  ██║██║   ██║██╔══╝
 ██║╚██████╔╝   ██║       ███████╗██████╔╝╚██████╔╝███████╗
 ╚═╝ ╚═════╝    ╚═╝       ╚══════╝╚═════╝  ╚═════╝ ╚══════╝

<6> 2022-09-06 19:25:18.172 +00:00 [INF] - Version - 1.4.0.59620487 (157650c3b1c5fb58b17f5975e6f1417a5a5f4f89)
<6> 2022-09-06 19:25:18.173 +00:00 [INF] - OptimizeForPerformance=True
<6> 2022-09-06 19:25:18.173 +00:00 [INF] - MessageAckTimeoutSecs=30
<6> 2022-09-06 19:25:18.173 +00:00 [INF] - Loaded server certificate with expiration date of "2022-10-06T19:25:17.0000000+00:00"
<6> 2022-09-06 19:25:18.179 +00:00 [INF] - Using Asp Net server for metrics
<6> 2022-09-06 19:25:18.220 +00:00 [INF] - Created new message store
<6> 2022-09-06 19:25:18.270 +00:00 [INF] - Created DeviceConnectivityManager with connected check frequency 00:05:00 and disconnected check frequency 00:02:00
<6> 2022-09-06 19:25:18.287 +00:00 [INF] - Started task to cleanup processed and stale messages
<6> 2022-09-06 19:25:18.461 +00:00 [INF] - Add node: iotedge
<6> 2022-09-06 19:25:18.462 +00:00 [INF] - Add node: iotedge/$edgeAgent
<6> 2022-09-06 19:25:18.463 +00:00 [INF] - Add node: iotedge/$edgeHub
<6> 2022-09-06 19:25:18.463 +00:00 [INF] - Add node: iotedge/DefenderIotMicroAgent
<6> 2022-09-06 19:25:18.463 +00:00 [INF] - Add node: iotedge/OPCPublisher
<6> 2022-09-06 19:25:18.464 +00:00 [INF] - Created device scope identities cache
<6> 2022-09-06 19:25:18.475 +00:00 [INF] - Initialized storing twin manager
<6> 2022-09-06 19:25:18.487 +00:00 [INF] - Updated node: iotedge/$edgeHub
<6> 2022-09-06 19:25:18.487 +00:00 [INF] - Initializing configuration
<6> 2022-09-06 19:25:18.498 +00:00 [INF] - New device connection for device iotedge/$edgeHub
<6> 2022-09-06 19:25:18.505 +00:00 [INF] - Client iotedge/$edgeHub connected to edgeHub, processing existing subscriptions.
<6> 2022-09-06 19:25:18.590 +00:00 [INF] - Starting refresh of device scope identities cache
<6> 2022-09-06 19:25:18.663 +00:00 [INF] - Attempting to connect to IoT Hub for client iotedge/$edgeHub via AMQP...
<6> 2022-09-06 19:25:18.687 +00:00 [INF] - Processing pending subscriptions for iotedge/$edgeHub
<6> 2022-09-06 19:25:18.891 +00:00 [INF] - Experimental features configuration: {"Enabled":false,"DisableCloudSubscriptions":false,"DisableConnectivityCheck":false}
<6> 2022-09-06 19:25:18.904 +00:00 [INF] - Starting timer to authenticate connections with a period of 300 seconds
<6> 2022-09-06 19:25:18.944 +00:00 [INF] - Scheduling server certificate renewal for "2022-10-06T19:22:47.0000753Z".
<6> 2022-09-06 19:25:19.077 +00:00 [INF] - No routes set in the edge hub
<6> 2022-09-06 19:25:19.078 +00:00 [INF] - Updated message store TTL to 7200 seconds
<6> 2022-09-06 19:25:19.078 +00:00 [INF] - Updated the edge hub store and forward configuration
<6> 2022-09-06 19:25:19.090 +00:00 [INF] - Started operation Get EdgeHub config
[... 41 lines cut ...]
<6> 2022-09-06 19:25:25.990 +00:00 [INF] - Created cloud proxy for client iotedge/OPCPublisher via AMQP, with client operation timeout 20 seconds.
<6> 2022-09-06 19:25:25.993 +00:00 [INF] - Initialized cloud proxy efe47b57-17b4-49f4-8c6e-5aaf3f288cf8 for iotedge/OPCPublisher
[... cut ...]
<6> 2022-09-06 19:25:28.190 +00:00 [INF] - Processing pending subscriptions for iotedge/OPCPublisher
<6> 2022-09-06 19:25:28.190 +00:00 [INF] - Processing pending subscriptions for iotedge/OPCPublisher
<6> 2022-09-06 19:25:28.251 +00:00 [INF] - Done syncing reported properties for iotedge/OPCPublisher
<6> 2022-09-06 19:25:28.266 +00:00 [INF] - Obtained edge hub config from module twin
<6> 2022-09-06 19:25:33.334 +00:00 [INF] - Updated reported properties for iotedge/$edgeHub
<6> 2022-09-06 19:30:18.913 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
```

hansgschossmann, 2022-09-07:
The startup code looks good. Do you also not see any increase of messages in the Azure Portal IoT Hub overview blade?

spinshin, 2022-09-07:
[image]
[image]

I can see an increase in message count as above. However, when I monitor using the CLI or by trying to receive Telemetry events through Azure IoT Explorer, I can see any messages being received.
To verify if I am monitoring the right way, I simulated the same device using az cli and send some device-to-cloud messages. Then I could see the messages being received in both az cli and IoT explorer.

akkleine1986, 2022-09-22:
I ran into a similar issue recently and the problem was in the "container create options" within IoT Hub.  Ensuring the hostname for container was different than the host server resolved my issue.

spinshin, 2022-09-24:
> I ran into a similar issue recently and the problem was in the "container create options" within IoT Hub. Ensuring the hostname for container was different than the host server resolved my issue.

Can you please elaborate a bit on the solution?

hansgschossmann, 2022-09-27:
@spinshin most likely he meant you should use in the Azure portal not use the hostname of your system you run IoT Edge on as a  name for a module you run in your deployment.

Pls also ensure that you have set your routes correct (use the default) to route all messages upstream.

hansgschossmann, 2022-10-11:
Closing this issue due to inactivity. Please reactivate if you still need help.

closed 2022-10-11 by hansgschossmann