Azure/Industrial-IoT #1770: OPC Publisher log file is growing full, no next log file is created

JohannesM24, 2022-09-05:
Hey,

we have the problem that once the OPC Publisher log file reaches 1 GB, no more logs are written. Also currently we need to stop the service to rename the old logfile, to enable the writing of a new logfile.

I think there should be an automatic change to write to a new log file, maybe adding the current timestamp to end of the filename.
Also there should be a retention policy, based on size or time.

Is there maybe an easy way to do this that I have missed?

hansgschossmann, 2022-09-06:
@JohannesM24 have you checked the section "Option: Adjust log settings for each container module" on [Prepare to deploy your IoT Edge solution in production](https://docs.microsoft.com/en-us/azure/iot-edge/production-checklist?view=iotedge-2018-06)?

JohannesM24, 2022-09-13:
Okay, thanks, just had a look, this looks helpful!

monikavar, 2022-09-27:
glad that it helped!