Azure/Industrial-IoT #1772: Does the OPCUA Publisher accepts NodeIds like this: "Data"."scada"."var"[002] ?

javierdve, 2022-09-13:
I am trying to connect to an OPC UA Server in an Edge device that uses NodeIds strings like above: "Data"."scada"."var"[004]

If I use UAExpert client I am able to connect but if I try to configure in the config file this type of strings there is an error when parsing the config file by the OPCUA Publisher. Is all naming conventions allowed for NodeIds string or should I use some special character that I ignore to use that NodeId string?. If not, could that be taken into consideration for future improvement.

[
  {
    "EndpointUrl": "opc.tcp://1.1.1.1:53530/OPCUA/SimulationServer",
    "UseSecurity": false,
    "OpcNodes": [
      {
        "Id": "ns=5;s="Data"."scada"."var"[002]""
      }
    ]
  }
]

Thanks a lot in advences and best regards,

cristipogacean, 2022-09-13:
Can you try if escaping the `"` characters in the nodeId? 
like: `"Id": "ns=5;s=\"Data\".\"scada\".\"var\"[002]\""`

javierdve, 2022-09-14:
The parsers does not provide error but the opcnode is not connected and give back error:
# Monitored Opc nodes succeeded count:              1
# Monitored Opc nodes failed count   :              1

I added two nodes one with "Id": "ns=2;i=2258" and the one with escaping " but does not work although the parser does not provide error in the file.

Thanks a lot

cristipogacean, 2022-09-14:
it might be the last `\"` mark in your example should be deleted:
`"Id": "ns=5;s=\"Data\".\"scada\".\"var\"[002]"`

javierdve, 2022-09-14:
I have tried several options as above and none seems working. If I use

`"Id": "ns=5; s=\"Data\".\"scada\".\"temperature"`

it warms "Data"."scada"."temperature is not a Nodeid but if I include the \" it does not work.

cristipogacean, 2022-09-14:
can you try to set the loglevel to debug? maybe we see more details on the error

javierdve, 2022-09-14:
Now it is working properly with \". I think was an issue with ns=. So issue is not real. Thanks a lot.

closed 2022-09-14 by javierdve