Azure/Industrial-IoT #1773: BadMonitoredNodeUnkown OPCPublisher

JohannesM24, 2022-09-15:
**Describe the bug**
When subscribing to nodes with using the OPCPublisher some MonitoredItems do not work and do not deliver data.

When connecting with a usual client to the OPC UA server, the data is fine.

**Expected behavior**
The subscription and creation of monitoredItems should work fine.

**Screenshots**

```
2022-09-14 20:15:59.106 +08:00 [WRN] Error monitoring node ns=36;s=DB75.TA013_2678002.GV.toBr.ProductionOrderTrigger due to BadNodeIdUnknown in subscription '1000'/'opc.tcp://SHA-SISD01.hilti.com:4897/Softing/dataFEEDSIS/Server_0689C4CE'
2022-09-14 20:15:59.140 +08:00 [DBG] Item 'ns=36;s=DB75.TA013_2678002.GV.toBr.ProductTrigger' with server id 0 - not created: Changing monitoring mode from "Disabled" to "Reporting"
```

**Desktop (please complete the following information):**
 - Windows Server 2019
 - Publisher version 2.8.3
 - standalone mode

What could be the reasons for the publisher sending the error of BadNodeIdUnkown although the nodeIDs are correct and we are able to subscribe them with another usual OPC UA client?

cristipogacean, 2022-09-22:
@JohannesM24, could it be that the namespace index (36) in this case has been changed in the opc ua server since the publisher was configured?

JohannesM24, 2022-09-26:
@cristipogacean The namespace index has not changed, we double checked things like that. 

Also it used to work sometimes after some restarts of the modules, but we need to find the root cause of this issue.

monikavar, 2022-09-27:
@JohannesM24 could you please provide the support bundle to investigate further?

hansgschossmann, 2022-09-27:
@JohannesM24 the BadNodeIdUnknown is what we get back from the server when we try to monitor the node via the subscription. Do you see any error on the server side?

hansgschossmann, 2022-10-11:
Closing due to inactivity. Please reactivate if you need further assistance.

JohannesM24, 2022-10-18:
Hey, so far the issue has not been resolved. Thanks far for the suggestion to look more into the server side. 

We did not find any error there yet, but when analyzing deeper with wireshark we found that the PublishResponse of the Server holds an empty NotificationData inside the NotificationMessage.

[image]

We are now in contact with the vendor of the OPC UA server and try to get their opinion on it (logs do not show enough right now).

From what I have seen so far there is a difference the publisher builds up a Subscription and MonitoredItems compared to other clients. Other clients usually send a CreateSubscriptionRequest (with PublishingEnabled = True) and afterwards a MonitoredItemsCreateRequest (with MonitoringMode = Enabled) directly, while the Publisher sends both of those with false/disabled and asks for their activiation in an additional request (SetPublishingModeRequest, SendMonitoringModeRequest). Could be this has something to with it. 

We will investigate further and let you know.

JohannesM24, 2022-11-02:
Hi @hansgschossmann

The OPC UA server supplier now told us that it is a bug in how their server handles the buildup of subscription and monitoredItem by the OPCpublisher. We are waiting for a fix.

I would still like to ask if there is a possibility to configure OPCpublisher to send the CreateSubscriptionRequest (with PublishingEnabled = True) and afterwards a MonitoredItemsCreateRequest (with MonitoringMode = Enabled) directly instead of the additional requests SetPublishingModeRequest, SendMonitoringModeRequest?

It seems this is a rather rarely used way of of creating the subscription (although of course perfectly valid inside the OPC UA specification). Is there any specific motivation behind?

marcschier, 2023-07-07:
Use `--eip` or `--immediatepublishing` to create the subscription with publishing enabled.  The monitored items are always in monitoring mode anyway and should fire immediately.

closed 2023-07-07 by marcschier