Azure/Industrial-IoT #1774: Is it possible to send messages from another module to OPC Twin?

felixmondelo, 2022-09-20:
Hi, I have several modules on our Edge devices (OPC Publisher, OPC Twin, ... and other custom modules).

The output of one of our modules, you have to write the value of the result in the OPC.

We think that the best option is to send a message to the OPC Twin through the edgeHub, but we did not find if this is possible.

Which is the best alternative to do this? Invoke an OPC Twin Direct Method?

monikavar, 2022-09-27:
Hi @felixmondelo, to send messages from one module to another, you could use routing in IoT Edge. This doc would be helpful: https://learn.microsoft.com/en-us/azure/iot-edge/module-composition?view=iotedge-1.4#declare-routes

Hope this helps.

felixmondelo, 2022-09-27:
Thank you @monikavar, I know that. My problem is if OPC Twin has any input to receive messages from other modules or not.

felixmondelo, 2022-09-27:
@monikavar then there isn't any way to communicate with OPC Twin from another module without going to the IoT Hub?

hansgschossmann, 2022-09-30:
@felixmondelo OPC Twin has a direct method interface, but this is limited to the use case OPC Twin was built for (being used as the edge side component of the Inudstrial-IoT microservices).

felixmondelo, 2022-10-05:
@hansgschossmann the requirement is (all is in the Edge):

1) I have the OPC Module getting signals from OPC
2) The output of the OPC Module goes through a custom module that in some cases changes the value of a signal.
3) That value changed has to be notified to OPC.

My idea was to use OPC Twin to notify OPC that change inside the Edge, but I suppose for your answer that this is not possible.

hansgschossmann, 2022-10-05:
@felixmondelo thanks for explaining your scenario. Are you using only the IoT Edge modules or do you use the full Industrial IoT platform (with cloud microservices)?

felixmondelo, 2022-10-05:
Only the modules :-(

hansgschossmann, 2022-10-05:
Yes, this is indeed not a supported scenario for OPC Twin. You might reach out to your MSFT contact to get an update on the Industrial product roadmap, there might be another solution better fitting your needs than the Industrial-IoT platform.

hansgschossmann, 2022-10-11:
Closing for now. Please reactivate if you need further assistance.

marcschier, 2023-07-03:
This is now implemented in 2.9.0-preview2. The OPC Twin functionality has been exposed inside OPC Publisher and other modules can use the direct method API which is documented under /docs.

closed 2023-07-03 by marcschier