Azure/Industrial-IoT #1775: OpcPublisher: Server not responding to keep alive requests

endreszilagyi1, 2022-09-20:
Hello,

We started to see that for some of our opc-ua servers we are not able to get the any telemetry using opc publisher (version 2.8.3).
Here is a snippet of the log, every 10 second we can see such an entry.

[2022-09-19 15:11:14.294 INF OpcUa] KEEP ALIVE LATE: 50.0120833s, EndpointUrl=opc.tcp://xx.xx.xx.xx:xxxx/, RequestCount=0/4 
[2022-09-19 15:11:14.294 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session 'opc.tcp://xx.xx.xx.xx:xxxx_A8D55C10' missed 3 Keepalive(s) due to bad BadNoCommunication 'Server not responding to keep alive requests.', waiting to recover...

[2022-09-19 15:11:24.274 INF OpcUa] KEEP ALIVE LATE: 59.9922388s, EndpointUrl=opc.tcp://xx.xx.xx.xx:xxxx/, RequestCount=0/5 
[2022-09-19 15:11:24.275 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session 'opc.tcp://xx.xx.xx.xx:xxxx_A8D55C10' missed 4 Keepalive(s) due to bad BadNoCommunication 'Server not responding to keep alive requests.', waiting to recover...

[2022-09-19 15:11:34.275 INF OpcUa] KEEP ALIVE LATE: 69.9932383s, EndpointUrl=opc.tcp://xx.xx.xx.xx:xxxx/, RequestCount=0/6 
[2022-09-19 15:11:34.275 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session 'opc.tcp://xx.xx.xx.xx:xxxx_A8D55C10' missed 5 Keepalive(s) due to bad BadNoCommunication 'Server not responding to keep alive requests.', waiting to recover...

Connecting to the same opc ua server using a different client (softing) works fine. 
Restarting the publisher didn't helped.
PLC type is Siemens SIMATIC.S7-1500.
We are using an Ubuntu VM to run IotEdge with OpcPublisher.

Do you have any idea?

Please let me know if you need anything from our side.
Thank you in advance.

hansgschossmann, 2022-09-27:
@cristipogacean pls update with your findings.

marcschier, 2023-01-16:
@endreszilagyi1, Apologies that this has gone a bit stale, but please try 2.8.4 and add publisher log file as well as command line used to invoke.  Also, any information re: the PLC and firmware will be helpful.

endreszilagyi1, 2023-01-16:
@marcschier, PLC is s7-1500 with 2.8.1 firmware. All the support bundles/logs are send to @hansgschossmann and @cristipogacean via mail (2022.09.20-09.28)
We have also diagnosed the PLC and checked the load:
•	Cycletime 1,5ms
•	Charging memory 91% free
•	RAM  95% free 
•	Remanent memory 31% free
•	Everything is on green level

We have PLC-s using exactly the same firmware version and running fine with publihser. 
Our goal for the next sprint is to update the version to the suggested one.

Let me know if you need further info from our side.

marcschier, 2023-01-16:
Thank you @endreszilagyi1

endreszilagyi1, 2023-01-16:
@marcschier, I just checked the doc of the new release (2.8.4) and the following MD file seems missing:
To continue deploying the 1.1 LTS modules to your environment follow [these instructions](https://github.com/Azure/Industrial-IoT/blob/2.8.4/deploy/howto-install-iot-edge.md).

hansgschossmann, 2023-01-16:
@endreszilagyi1  i have opened a new issue to track the missing doc: #1935 

I close this issue since we handle it via a support case.

closed 2023-01-16 by hansgschossmann