Azure/Industrial-IoT #1776: OPC monitored items (monitored) does not increase

e-augusto, 2022-09-28:
Hi all,

This week I created/published new items, but when I checked in Kusto explorer the new items did not appear.
I checked the OPC and the source, everything is ok with the published tags, then checking the opcPublisher log I noticed that when I created new items, the value of the configured one increased, but the monitored one did not.
Can anyone direct me on how to solve these problems?

[image]

hansgschossmann, 2022-09-30:
@e-augusto can you provide us the log of OPC Publisher just when it try to start monitoring the item you have configured? Maybe just restart OPC Publisher then we should get it.

hansgschossmann, 2022-10-11:
@e-augusto I just realized that you are using an old version of OPC Publisher (at least the log looks like this). Please ensure you are using the latest version 2.8.3.

closed 2022-10-14 by marcschier