Azure/Industrial-IoT #1778: URGENT: Jobs-Orchestrator requests run in 504 Gateway Time-out

emulic, 2022-09-29:
In our production environment the publisher log of all devices have lot of warnings like this:
```shell
[14:36:11 WRN Microsoft.Azure.IIoT.Http.Default.HttpClient] POST to https://****.westeurope.cloudapp.azure.com/edge/publisher/v2/workers/****_publisher_3 returned GatewayTimeout (took 00:00:15.0457509). 
[14:36:23 WRN Microsoft.Azure.IIoT.Http.Default.HttpClient] POST to https://****.westeurope.cloudapp.azure.com/edge/publisher/v2/heartbeat returned GatewayTimeout (took 00:00:15.0409913). 
[14:36:24 WRN Microsoft.Azure.IIoT.Http.Default.HttpClient] POST to https://****.westeurope.cloudapp.azure.com/edge/publisher/v2/heartbeat returned GatewayTimeout (took 00:00:15.0482120). 
[14:36:24 WRN Microsoft.Azure.IIoT.Http.Default.HttpClient] POST to https://****.westeurope.cloudapp.azure.com/edge/publisher/v2/heartbeat returned GatewayTimeout (took 00:00:15.0475159). 
```
Trying to open edge/publisher microservice swagger site (https://****.westeurope.cloudapp.azure.com/edge/publisher/swagger/index.html) is also running in 504 Gateway Time-out after very long loading time.
According to the logs of the everything looks normal:
```shell
edin@Azure:~$ kubectl logs aiiot-edge-jobs-7f788bbf57-hpv6x -n aiiot
[2022-09-29 14:23:46.710 INF Microsoft.Azure.IIoT.Utils.HostAutoStart] All hosts started. 
[2022-09-29 14:23:47.084 INF Microsoft.Azure.IIoT.AspNetCore.Cors.CorsSetup] CORS is enabled 
[2022-09-29 14:23:47.084 INF Microsoft.Azure.IIoT.AspNetCore.Cors.CorsSetup] Allow all headers, origins and methods 
[2022-09-29 14:23:47.226 INF Root] Edge-Jobs-Service web service started with id 3e180316-4c5e-4824-b5fc-74e813e3d081 
```
Ingress-nginx pods are also full of timeout errors:
```shell
10.244.0.1 - - [29/Sep/2022:14:45:00 +0000] "POST /edge/publisher/v2/heartbeat HTTP/1.1" 504 160 "-" "-" 641 15.000 [aiiot-aiiot-edge-jobs-9051] [] 10.244.0.38:9051, 10.244.0.38:9051, 10.244.0.38:9051 0, 0, 0 5.000, 5.000, 5.000 504, 504, 504 03426f044ba024b9e1295f5228315e44
2022/09/29 14:45:01 [error] 310#310: *118176 upstream timed out (110: Operation timed out) while connecting to upstream, client: 10.244.0.1, server: *****.westeurope.cloudapp.azure.com, request: "POST /edge/publisher/v2/heartbeat HTTP/1.1", upstream: "http://10.244.0.38:9051/edge/publisher/v2/heartbeat", host: "*****.westeurope.cloudapp.azure.com"
2022/09/29 14:45:01 [error] 310#310: *118025 upstream timed out (110: Operation timed out) while connecting to upstream, client: 10.244.0.1, server: *****.westeurope.cloudapp.azure.com, request: "POST /edge/publisher/v2/workers/****_publisher_4 HTTP/1.1", upstream: "http://10.244.0.38:9051/edge/publisher/v2/workers/****_publisher_4", host: "*****.westeurope.cloudapp.azure.com"
10.244.0.1 - - [29/Sep/2022:14:45:01 +0000] "POST /edge/publisher/v2/heartbeat HTTP/1.1" 504 160 "-" "-" 642 15.001 [aiiot-aiiot-edge-jobs-9051] [] 10.244.0.38:9051, 10.244.0.38:9051, 10.244.0.38:9051 0, 0, 0 5.004, 5.000, 5.000 504, 504, 504 9a829a86e5e5bf9720688ccf61144933
2022/09/29 14:45:01 [error] 311#311: *117956 upstream timed out (110: Operation timed out) while connecting to upstream, client: 10.244.0.1, server: *****.westeurope.cloudapp.azure.com, request: "POST /edge/publisher/v2/workers/****_publisher_2 HTTP/1.1", upstream: "http://10.244.0.38:9051/edge/publisher/v2/workers/****_publisher_2", host: "*****.westeurope.cloudapp.azure.com"
2022/09/29 14:45:01 [error] 310#310: *118190 upstream timed out (110: Operation timed out) while connecting to upstream, client: 10.244.0.1, server: *****.westeurope.cloudapp.azure.com, request: "POST /edge/publisher/v2/heartbeat HTTP/1.1", upstream: "http://10.244.0.38:9051/edge/publisher/v2/heartbeat", host: "*****.westeurope.cloudapp.azure.com"
10.244.0.1 - - [29/Sep/2022:14:45:01 +0000] "POST /edge/publisher/v2/heartbeat HTTP/1.1" 504 160 "-" "-" 642 15.001 [aiiot-aiiot-edge-jobs-9051] [] 10.244.0.38:9051, 10.244.0.38:9051, 10.244.0.38:9051 0, 0, 0 5.000, 5.000, 5.000 504, 504, 504 9c80162d954eddc971338eba990a711
```

None of the aks containers is under high load:
[image]

Cosmos DB is also not under high load:
[image]

We use the IIoT version 2.8.3.

We already tried to recreate pods and even to setup a new cluster and have the same issue.
All other microservices are running with no issues (at least swagger endpoints are available).

As already mentioned, we have this issue in the production environment, so we urgently need a fix.

emulic, 2022-09-30:
UPDATE:
Since 07:02 today edge-jobs service is running again. We didn't change anything. We can see bunch of evicted and errored edge-jobs pods in AKS:
```shell
edin@Azure:~$ kubectl get pods -n aiiot | grep ^aiiot-edge-jobs
aiiot-edge-jobs-7f788bbf57-27w58            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-2v6ms            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-452wx            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-4nfdp            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-4q92c            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-5gh9v            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-5s9j9            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-6lc6b            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-6p9bs            0/1     Error     0          6h21m
aiiot-edge-jobs-7f788bbf57-6wrbf            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-74clt            0/1     Evicted   0          6h39m
aiiot-edge-jobs-7f788bbf57-7rrtl            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-7x8gt            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-864rg            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-8m922            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-9285t            0/1     Evicted   0          6h39m
aiiot-edge-jobs-7f788bbf57-9k87k            0/1     Evicted   0          6h39m
aiiot-edge-jobs-7f788bbf57-9s277            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-bmndp            0/1     Evicted   0          6h39m
aiiot-edge-jobs-7f788bbf57-ch4sw            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-ch5z8            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-cnxvm            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-ct4pv            0/1     Error     0          10h
aiiot-edge-jobs-7f788bbf57-d82xl            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-db9bs            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-ddt72            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-f2t6n            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-fc9ll            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-fkwjk            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-gg52r            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-ghpc4            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-gmknh            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-hknb7            0/1     Error     0          10h
aiiot-edge-jobs-7f788bbf57-hnh4r            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-k5tz8            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-mj9mb            0/1     Evicted   0          6h39m
aiiot-edge-jobs-7f788bbf57-mr4nb            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-n8rvs            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-pxl6m            0/1     Error     0          14h
aiiot-edge-jobs-7f788bbf57-q4j6f            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-q8mm2            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-q9pkh            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-qjhtb            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-rjznc            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-rzctt            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-s8d87            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-sdkb6            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-slvxt            0/1     Evicted   0          6h39m
aiiot-edge-jobs-7f788bbf57-st6rs            0/1     Evicted   0          6h39m
aiiot-edge-jobs-7f788bbf57-swm2z            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-tbtft            0/1     Evicted   0          6h39m
aiiot-edge-jobs-7f788bbf57-v7mgs            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-vgdqx            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-w55pk            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-w6s4p            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-wbsgw            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-wnwjz            0/1     Evicted   0          6h39m
aiiot-edge-jobs-7f788bbf57-xc5nx            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-xrzwn            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-xtszf            0/1     Evicted   0          11h
aiiot-edge-jobs-7f788bbf57-z7kxx            0/1     Evicted   0          10h
aiiot-edge-jobs-7f788bbf57-zlb5f            0/1     Evicted   0          170m
aiiot-edge-jobs-7f788bbf57-zrmfk            1/1     Running   0          152m
aiiot-edge-jobs-7f788bbf57-zvf4f            0/1     Evicted   0          11h
```
Logs of all evicted pods have the same error:
```shell
edin@Azure:~$ kubectl logs aiiot-edge-jobs-7f788bbf57-27w58 -n aiiot
Error from server (BadRequest): container "industrial-iot-jobs-orchestrator-service-container" in pod "aiiot-edge-jobs-7f788bbf57-27w58" is not available
```
[... cut ...]
[2022-09-30 05:09:29.161 INF Microsoft.Azure.IIoT.Utils.HostAutoStart] All hosts started. 
[2022-09-30 05:09:29.550 INF Microsoft.Azure.IIoT.AspNetCore.Cors.CorsSetup] CORS is enabled 
[2022-09-30 05:09:29.551 INF Microsoft.Azure.IIoT.AspNetCore.Cors.CorsSetup] Allow all headers, origins and methods 
[2022-09-30 05:09:29.684 INF Root] Edge-Jobs-Service web service started with id 7b00321a-86e0-4718-8da1-aaadfb47a2e2 
```

Please help us to analyze what went wrong here. We must prevent such issues in the production in the future.

karok2m, 2022-09-30:
@emulic, can you please use `kubectl describe pod` for one of the evicted pods to see what events are reported for it and if there is a reason for eviction. The events should be reported in the end of the command output. And please also run `kubectl get event -n aiiot`, that should give us all the events in the namespace.

I am just guessing here, but one of the reasons for eviction could be high resource utilization by a pod. And the limit for this "high" utilization might actually be quite low if no resource utilization `requests` and `limits` are set. So are you setting any values for them in the `values.yaml` file that you use when deploying the chart ?

karok2m, 2022-09-30:
@emulic, yes, you are correct. `MemoryPressure` indicates that the particular node that the pod was running on was running out of memory resources. So setting resource requests and limits would allow Kubernetes to better place the pod on a node that can meet the resource requirements. And you should definitely try that.

One other thing that you can do to increase availability of `edgeJobs` service is to run 2 instances of it. This would make sure that even if one of the pods goes down the other one can still server the requests. Please note that this would mean doubling resource requests. The example bellow shows how to set this as well via `replicas`, which by default has a value of `1`.

And, yes, that is the correct link for an example of setting resource requests and limits, but the indentation there does not look correct. I will fix that. Correctly indented definition would look like this:

```yaml
deployment:
  microServices:
    edgeJobs:
      replicas: 2
      resources:
        requests:
          memory: "128Mi"
          cpu: "50m"
        limits:
          memory: "512Mi"
          cpu: "200m"
```

emulic, 2022-10-04:
Hi @karok2m,
Actually I already tried to increase replicas to 2 but this didn't help. Probably because both of them ran into out of memory.
Thanks a lot for supporting us and please feel free to close the issue.

karok2m, 2022-10-04:
@emulic, please let us know if you see the problem again. For the time being I will close this issue. Feel free to reopen it if needed.

emulic, 2022-10-10:
Hi again @karok2m , 

Unfortunately, we have the same or similar issue again.
In the meantime, we applied following configuration for  edgeJobs service:
```yaml
edgeJobs:
      replicas: 2
      resources:
        requests:
          memory: "512Mi"
          cpu: "200m"
        limits:
          memory: "2Gi"
          cpu: "1000m"
```
By opening ```/edge/publisher/swagger/index.html``` in browser we are getting HTTP 504 Gateway Time-out after approximately 10 seconds.

Here are kubectl describe pod results for both edgeJobs pods:

```shell
edin@Azure:~$ kubectl describe pod aiiot-edge-jobs-786fccf776-hf27w -n aiiot
Name:             aiiot-edge-jobs-786fccf776-hf27w
Namespace:        aiiot
Priority:         0
Service Account:  aiiot-azure-industrial-iot
Node:             aks-iiotpool-20873995-vmss000001/10.224.0.7
Start Time:       Mon, 10 Oct 2022 13:03:17 +0000
Labels:           app.kubernetes.io/component=industrial-iot-jobs-orchestrator-service
                  app.kubernetes.io/instance=aiiot
                  app.kubernetes.io/managed-by=Helm
                  app.kubernetes.io/name=azure-industrial-iot
                  app.kubernetes.io/part-of=azure-industrial-iot
                  app.kubernetes.io/version=2.8.3
                  helm.sh/chart=azure-industrial-iot-0.4.3
                  pod-template-hash=786fccf776
Annotations:      checksum/config: a2224c8116283e8960b2e693452dc4541f7f0b24d96706b71367ff25cdd23bd0
                  prometheus.io/path: /metrics
                  prometheus.io/port: 9051
                  prometheus.io/scrape: true
Status:           Running
IP:               10.244.128.35
IPs:
  IP:           10.244.128.35
Controlled By:  ReplicaSet/aiiot-edge-jobs-786fccf776
Containers:
  industrial-iot-jobs-orchestrator-service-container:
    Container ID:   containerd://4857bad1e845e2d655c6c51d11dc49e93ad130274c0c22e4bffda7826747cf44
    Image:          mcr.microsoft.com/iot/opc-publisher-edge-service:2.8.3
    Image ID:       mcr.microsoft.com/iot/opc-publisher-edge-service@sha256:e3da20d9009f34a67d038731c3c4b1c630653dca7c0957509c024a9cb0ade501
    Port:           9051/TCP
    Host Port:      0/TCP
    State:          Running
      Started:      Mon, 10 Oct 2022 13:03:18 +0000
    Ready:          True
    Restart Count:  0
    Limits:
      cpu:     1
      memory:  2Gi
    Requests:
      cpu:     200m
      memory:  512Mi
    Environment Variables from:
      aiiot-azure-industrial-iot-env  Secret  Optional: false
    Environment:                      <none>
    Mounts:
      /var/run/secrets/kubernetes.io/serviceaccount from kube-api-access-h4psm (ro)
Conditions:
  Type              Status
  Initialized       True 
  Ready             True 
[... 83 lines cut ...]
QoS Class:                   Burstable
Node-Selectors:              <none>
Tolerations:                 node.kubernetes.io/memory-pressure:NoSchedule op=Exists
                             node.kubernetes.io/not-ready:NoExecute op=Exists for 300s
                             node.kubernetes.io/unreachable:NoExecute op=Exists for 300s
Events:
  Type    Reason     Age   From               Message
  ----    ------     ----  ----               -------
  Normal  Scheduled  49m   default-scheduler  Successfully assigned aiiot/aiiot-edge-jobs-786fccf776-n5d6h to aks-iiotpool-20873995-vmss000001
  Normal  Pulled     49m   kubelet            Container image "mcr.microsoft.com/iot/opc-publisher-edge-service:2.8.3" already present on machine
  Normal  Created    49m   kubelet            Created container industrial-iot-jobs-orchestrator-service-container
  Normal  Started    49m   kubelet            Started container industrial-iot-jobs-orchestrator-service-container
```

And the log of the edgeJobs pods are also not showing any warnings or errors:
```shell
edin@Azure:~$ kubectl logs aiiot-edge-jobs-786fccf776-n5d6h -n aiiot
[2022-10-10 13:03:20.318 INF Microsoft.Azure.IIoT.Utils.HostAutoStart] All hosts started. 
[2022-10-10 13:03:20.814 INF Microsoft.Azure.IIoT.AspNetCore.Cors.CorsSetup] CORS is enabled 
[2022-10-10 13:03:20.814 INF Microsoft.Azure.IIoT.AspNetCore.Cors.CorsSetup] Allow all headers, origins and methods 
[2022-10-10 13:03:21.062 INF Root] Edge-Jobs-Service web service started with id f3c80b78-0ba5-4a00-b7c3-419cde93c914 
```

I would appreciate if you re-open the issue (I'm not able to do that, since closed by you) and further investigate.
Please note, we have this issue in our production environment.

emulic, 2022-10-10:
@karok2m Update:
I just tried curl of the edgeJobs swagger site direct from cluster internal network and here we get timeout after 129 seconds:
```shell
edin@Azure:~$` kubectl exec -i --tty mycurlpod -- sh
/ $ curl http://10.244.128.37:9051/edge/publisher/swagger/index.html
curl: (28) Failed to connect to 10.244.128.37 port 9051 after 129196 ms: Operation timed out
```

emulic, 2022-10-11:
@karok2m We have 45 publishers but I suppose not all devices are turned on. Some of them are for future purposes.
We don't have any throttled requests in the last days:
[image]

Also latency also seems to not be an issue:
[image]

Regarding intervals, we already increased HeartbeatInterval for some publishers having large number of subscribed OPC UA endpoints, since this caused some kind of time-outs ("The operation was canceled.!" after some random waiting period for response which was between 20 seconds and 15 minutes) when the heartbeat frequency was high (and it was very high when having lot of OPC UA Endpoints connected to a single publisher instance). But this happened on affected edge devices only. Reaching endpoint via swagger UI was never an issue. I suppose this has been caused by some kind of firewall blocking such frequent calls when they came from the same client. Not sure if this "firewall" was running in Azure or in the customer network, I never further investigated that. 
However, which values would you propose for HeartbeatInterval and JobCheckInterval? And how to get those new settings propagated to the edge devices, when the job orchestrator service is not reachable from publisher modules?

karok2m, 2022-10-11:
To be honest, I cannot comment on the "firewall" side of things.

Regarding the OPC Publishers with many endpoints, the heartbeat is going to be called by each worker that is active on the OPC Publisher. So if you have an OPC Publisher with `10` workers and processing workload from (let's say) `6` endpoints, then `6` of those workers are going to periodically call Job-Orchestrator to report their status (heartbeat) and `4` of them will periodically call it to ask for jobs to process. Those two intervals are configured via `HeartbeatInterval` and `JobCheckInterval`. In this case, I would recommend to set `HeartbeatInterval` to `60` seconds and `JobCheckInterval` to `120` seconds. Please note that those configurations are applied per publisher. So you will potentially have to set the values for all of your publishers. The values are actually set either through Registry microservice or Engineering Tool.

For setting them through Registry microservice please check the documentation [here](https://github.com/Azure/Industrial-IoT/blob/main/docs/manual/readme.md#troubleshooting-the-opc-publisher-module). It walks you through setting the values via `PATCH` call to `/registry/v2/publishers/<PublisherId>` endpoint:

[image]

Alternatively, you can do this from Engineering Tool:

[image]

The values are then going to become part of Module Identity Twin for the specified publisher module in IoT Hub and be part of the deployment templates for them. So availability of Job-Orchestrator service is not required for setting and propagating those values.

emulic, 2022-10-11:
@karok2m 
Please let me know if you need additional information or even if you believe we need to organize a call, where you could make some checks in the affected environment directly.

karok2m, 2022-10-11:
@emulic, any chance you can get RAM and CPU usage numbers for this pod ? I will check how to enable `DEBUG` logging for it, hopefully that can tell us a bit more about what is going on. If you have Application Insights enabled for your deployment, then the Application Map can be quite useful in understanding how the whole system is behaving. It has a neat way of showing erroneous requests happening in the request call/propagation graph.

emulic, 2022-10-12:
@karok2m 
```shell
edin@Azure:~$ kubectl top pod -n aiiot
NAME                                        CPU(cores)   MEMORY(bytes)   
aiiot-edge-jobs-786fccf776-jghzh            8m           1488Mi          
aiiot-edge-jobs-786fccf776-jqq9p            7m           1472Mi          
aiiot-engineering-tool-6c99fdbbdf-jqnxp     3m           160Mi           
aiiot-events-57bd4b5dc4-m4g4b               161m         233Mi           
aiiot-events-processor-74d4f464f4-zn96f     13m          191Mi           
aiiot-history-867c948b95-tblvp              2m           135Mi           
aiiot-onboarding-7dbb66fc6-7vq9n            13m          192Mi           
aiiot-publisher-869bbcf75f-l6kks            3m           204Mi           
aiiot-registry-7f849d88-87jmj               3m           218Mi           
aiiot-sync-68f69487b6-fpwss                 4m           184Mi           
aiiot-telemetry-processor-8bdbf4999-ng89g   76m          86Mi            
aiiot-twin-6498c4fffc-hzq9k                 3m           217Mi   
```

Let me check what I can find in Application Insights.

emulic, 2022-10-12:
It is hard to export Application Map in a meaningful way.
Here are for example details regarding jobs service for 1h time range on 15.09.2022:
[image]
We see 110k requests for this time range and 22 failed requests. Average response time is also fine.
If set filter for the 12 hours, the job service doesn't appear as a map component at all since it doesn't receive any requests. 

According to the Application Map for last 12 hours, the only component showing some issues is Engineering Tool:
[image]

[image]

But  I don't believe this is related to the job orchestrator issue we have.

emulic, 2022-10-27:
Just for documentation purposes:
We had a call with @karok2m where we analyzed the whole environment and tried few things to fix the issue but with no success.
Few days later we had a second call, this time with @cristipogacean and we managed to solve the issue by removing two Jobs configuration documents in the Cosmos DB related to one of the publishers. This, combined with recreating edge-jobs pods immediately solved the issue. We were even able to recreate one of the documents and the job orchestrator was still properly working (even after recreating pods).
But as we tried to recreate the second document too, the issue was there again.
Since the second document contained configuration which was not required any more, we just removed it again and this worked for us as workaround.

@cristipogacean and @karok2m: you already have the full content of the DB required for analyzing and fixing this issue. Just let me know if you need any additional information.

closed 2022-10-27 by cristipogacean

cristipogacean, 2022-10-28:
@emulic, we managed to replicate inhouse the issue with the dataset provided and a fix is available in the main branch already. This will be integrated in the next release of the iiot platform. At this point in time, we cannot give a clear indication on the timeline of the release.

emulic, 2022-10-28:
@cristipogacean Thanks.