Azure/Industrial-IoT #1827: OPCPublisher does not connect to opcua server with username and passwort

NaumannJ, 2022-10-07:
We try to connect to a OPCUA Server with username and password we added to the pn.json : 

{
    "EndpointUrl": "opc.tcp://11.111.111.11:4840",
    "UseSecurity": false,
    "OpcAuthenticationMode": "UsernamePassword",
    "Username": "username",
    "Password": "password",
    "OpcNodes": [
.....

The Connection fails with the following lines in the logfile : 

[2022-09-29 08:28:22.719 INF OpcUa] ChannelId 0: in Connecting state.
[2022-09-29 08:28:22.729 INF OpcUa] ChannelId 0: in Opening state.
[2022-09-29 08:28:22.729 INF OpcUa] ChannelId 0: Token #0 created. CreatedAt=08:28:22.729. Lifetime=3600000.
[2022-09-29 08:28:22.731 INF OpcUa] Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
[2022-09-29 08:28:22.731 INF OpcUa] Sender Certificate: (none)
[2022-09-29 08:28:22.731 INF OpcUa] ChannelId 352: Token #1 activated. CreatedAt=08:28:22.729. Lifetime=3600000.
[2022-09-29 08:28:22.731 INF OpcUa] ChannelId 352: in Open state.
[2022-09-29 08:28:22.732 INF OpcUa] ChannelId 352: Token Expiry 09/29/2022 09:28:22, renewal scheduled in 2699997 ms.
[2022-09-29 08:28:22.735 INF OpcUa] ChannelId 352: in Closing state.
[2022-09-29 08:28:22.737 WRN OpcUa] ChannelId 352: Force reconnect reason=BadConnectionClosed 'Remote side closed connection'
[2022-09-29 08:28:22.737 INF OpcUa] ChannelId 352: in Closed state.
[2022-09-29 08:28:22.737 INF OpcUa] ChannelId 352: CLIENTCHANNEL SOCKET CLOSED: 02C70E5F
[2022-09-29 08:28:22.738 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Azure IIoT opc.tcp://11.111.111.11:4840_8A829D0A took 00:00:00.0201368.
[2022-09-29 08:28:22.739 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session 'opc.tcp://11.111.111.11:4840_8A829D0A' due to One or more errors occurred. (Endpoint does not support the user identity type provided.)

If I try to connect with the softing opcua client a connection with : 

endpoint url : opc.tcp://11.111.111.11:4840
securety mode : none
securety policy : none
message encoding : binary
User Identity : UserName
User Name : username
Password : password

there is no problem.
I tryed the connection with : 

   "UseSecurity": true,

in the pn.json too but the error is the same.
Maybe the opcpublisher does not support connections with username and password instead of anonymous connection
In both cases I got : 
Endpoint does not support the user identity type provided
in the logfile.

I use Ubuntu 20.04 LTS
aziot-edge:amd64/focal 1.4.1-1 upgradeable to 1.4.2-1
moby-engine:amd64/focal 20.10.18+azure-ubuntu20.04u1 upgradeable to 20.10.18+azure-ubuntu20.04u2

there is no blocking firewall between the opcpublisher and opcuaserver 

Hope someone could help me to resolve the problem.
Maybe there are other  types for : 

    "UseSecurity": false,
    "OpcAuthenticationMode": "UsernamePassword",

to connect to a opcua server with no security and only username and password.

cristipogacean, 2022-10-07:
Could you try instead of:
```
"OpcAuthenticationMode": "UsernamePassword",
"Username": "username",
"Password": "password",
```

the following construct:

```
"OpcAuthenticationMode": "UsernamePassword",
"OpcAuthenticationUsername"": "username",
"OpcAuthenticationPassword": "password",
```
Examples of pn.json files can be found here:
https://github.com/Azure/Industrial-IoT/blob/main/docs/modules/publishednodes_2.8.json

cristipogacean, 2022-11-04:
Closing due to inactivity. Please reopen if required.

closed 2022-11-04 by cristipogacean