Azure/Industrial-IoT #1828: PublishTime is newer then the actual time for SubscriptionId - Warning

hvetter-de, 2022-10-07:
Last week we updated from Azure Industrial IoT 2.8.0 to 2.8.3.
Now we get these Warnings in the log from the publisher module:

`[2022-10-07 09:00:22.309 WRN OpcUa] PublishTime 10/07/2022 10:07:07 in publish response is newer than actual time for SubscriptionId XXX.`

In the documentation I couldn't find any information about this.
Does the publisher ignore the message or is this just a Warning that there could be something wrong?

If the publisher ignores, is it possible to configure the publisher to process the messages regardless of the timestamp?

cristipogacean, 2022-10-07:
This warning pops when the clock of the host running the server is out of sync with the clock of the host running the publisher module. With 2.8.3 we enabled the OPC UA Stack's logs in the publisher's log. This is probably the reason these warnings appear only now. Most likely the clocks I mentioned above were out of sync before upgrading as well, just this was not visible.
Anyhow, this warning does not affect the communication nor the subscription cyclic data exchange.

hvetter-de, 2022-10-07:
Thanks for the information. Our client will not change the clock from the host of the Plc server. We will just ignore this warning then.