Azure/Industrial-IoT #1842: Discovery module not connecting

brentcocu-mayker, 2022-10-14:
**Describe the bug**
When deploying the Discovery module to the edge gateway, it starts, but fails to connect and logs the following error message: "Module [deviceId]_discovery disconnected due to Communication_Error - now Disconnected_Retrying...".

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy module with following create options:
```
{
  "Hostname": "discovery",
  "Cmd": [
    "PkiRootPath=/mount/pki"
  ],
  "NetworkingConfig": {
    "EndpointsConfig": {
      "host": {}
    }
  },
  "HostConfig": {
    "NetworkMode": "host",
    "Binds": [
      "/home/iotedge-user/mount:/mount"
    ],
    "CapDrop": [
      "CHOWN",
      "SETUID"
    ]
  }
}
```
2. Check logs.

**Expected behavior**
I expect the module to connect and to allow discovery of an OPC UA server (KepServerEx) running locally on the windows host device.

**Screenshots**
This is what the architecture looks like. What I want is for the discovery module to be able to discover the kepserver.
[image]

**EFLOW host:**
 - OS: Windows 11

**EFLOW vm:**
 - OS: CBL-Mariner/Linux

**Discovery module:**
 - Image URI: mcr.microsoft.com/iotedge/discovery:2.8
 - Version: 2.8

marcschier, 2022-10-14:
I think this will be related to host mode, remove host network mode and try to deploy again.  It should then find the edgehub. 
 Since EEFLOW runs in a vm with switch host mode is not necessary anyway, but you need to define the scan network mask then (like in the windows instructions).

brentcocu-mayker, 2022-10-14:
Okay so I removed the host mode part and deployed it with following create options
```
{
  "Hostname": "discovery",
  "Cmd": [
    "PkiRootPath=/mount/pki"
  ],
  "HostConfig": {
    "Binds": [
      "/home/iotedge-user/mount:/mount"
    ],
    "CapDrop": [
      "CHOWN",
      "SETUID"
    ]
  }
}
```
The module actually connects now and it shows up in the Asset Discovery view in the Engineering Tool. I do notice another problem now. My OPC UA server endpoint is at "opc.tcp://10.1.67.169:49320", so in the Effective Configuration for the discovery I updated the 'Mask Address Range' field to 10.1.67.0/24. I left all other values on default. When I turn on the scanning however some logs get printed, but no endpoints are discovered and after a few seconds the discovery module seems to disconnect.

When I check the logs for the discovery module it says the following:
```
[2022-10-14 13:54:29.103 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] New settings processed. 
[2022-10-14 13:54:29.104 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Discovery operation started. 
[2022-10-14 13:54:29.139 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Start Fast discovery run... 
[2022-10-14 13:54:29.139 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Starting network scan (0 probes active)... 
[2022-10-14 13:54:29.146 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Start scanning ["10.1.67.0/24"]... 
[2022-10-14 13:54:29.174 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.123 (0 scanned)... 
[2022-10-14 13:54:29.175 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.12 (0 scanned)... 
[2022-10-14 13:54:29.177 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.176 (1 scanned)... 
[2022-10-14 13:54:29.177 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.248 (2 scanned)... 
[2022-10-14 13:54:29.177 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.251 (3 scanned)... 
[2022-10-14 13:54:29.177 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.252 (4 scanned)... 
[2022-10-14 13:54:29.177 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.47 (5 scanned)... 
[2022-10-14 13:54:29.177 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.141 (6 scanned)... 
[2022-10-14 13:54:29.178 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.170 (7 scanned)... 
[2022-10-14 13:54:29.178 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.126 (8 scanned)... 
[2022-10-14 13:54:29.178 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.37 (9 scanned)... 
[2022-10-14 13:54:29.179 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.45 (10 scanned)... 
[2022-10-14 13:54:29.179 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.14 (11 scanned)... 
[2022-10-14 13:54:29.179 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.177 (12 scanned)... 
[2022-10-14 13:54:29.181 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.115 (14 scanned)... 
[2022-10-14 13:54:29.181 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.164 (15 scanned)... 
[2022-10-14 13:54:29.182 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.17 (16 scanned)... 
[2022-10-14 13:54:29.208 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.167 (17 scanned)... 
[2022-10-14 13:54:29.211 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.143 (18 scanned)... 
[2022-10-14 13:54:29.232 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.117 (19 scanned)... 
[2022-10-14 13:54:29.238 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.154 (20 scanned)... 
[2022-10-14 13:54:29.248 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.44 (21 scanned)... 
[2022-10-14 13:54:29.249 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.173 (22 scanned)... 
[2022-10-14 13:54:29.305 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.53 (23 scanned)... 
[2022-10-14 13:54:30.005 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.3 (24 scanned)... 
[2022-10-14 13:54:32.177 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.36 (110 scanned)... 
[2022-10-14 13:54:32.182 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.1.67.24 (111 scanned)... 
[... cut ...]
   at Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices.<>c__DisplayClass14_2.<DiscoverServersAsync>b__6(Object _)
   at System.Threading.TimerQueueTimer.<>c.<.cctor>b__23_0(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.TimerQueueTimer.CallCallback(Boolean isThreadPool)
   at System.Threading.TimerQueueTimer.Fire(Boolean isThreadPool)
   at System.Threading.TimerQueue.FireNextTimers()
```

marcschier, 2022-11-08:
#1763

hansgschossmann, 2022-12-05:
@brentcocu-mayker pls verify with the latest release (see #1763 ). Please reactivate if the issue still exists.

closed 2022-12-05 by hansgschossmann