Azure/Industrial-IoT #1890: Subscriptions that become invalid are not detected by OPC Publisher (?)

NoTuxNoBux, 2022-10-27:
**Describe the bug**
Whenever backing OPC UA properties for subscriptions seem to "disappear" (see below), the OPC Publisher does not pick this up, assuming that the subscription is still valid. Changes to these properties are then no longer picked up until the configuration is reloaded or the OPC Publisher is restarted.

**To Reproduce**
This is slightly difficult since it may be a quirk due to the way this specific device works (see also below), but here it goes:

1. Load an application on the target device hosting the OPC UA server that exposes custom user properties, `Foo` and `Bar`.
2. Subscribe to updates for `Foo` and `Bar` in OPC Publisher.
3. On the target device, reload the application (likely causing properties, meaning also the subscriptions, to temporarily disappear and reappear).
4. Changes to these properties now are now no longer propagated to the OPC Publisher. No errors are reported.

**Expected behavior**
The OPC Publisher detects that the subscriptions no longer exist or work and recreates them - like it does if the connection fully drops or if the configuration file is updated.

**Additional context**
We connect to an OPC UA server of a Stäubli CS9 controller, which exposes certain system properties, and can also have "user properties" exposed from applications (VAL 3) written for it. On the controller, the application can be reloaded, e.g. if there is a problem or if changes are made. When this is done, the above problem occurs - if you reboot the device fully, it doesn't since the OPC Publisher detects this as disconnection and goes through the setup phase again (so that is also the workaround).

What I _think_ happens is that by reloading the application, the user properties temporarily disappear, making the subscriptions invalid, and then reappear. This also seems to be confirmed by that other system properties do not exhibit this problem when an application is reloaded.

To make this more difficult: we don't have permanent access to this device, meaning it might be hard for us to test fixes, but I wanted to report this anyway to see if there is something that can be done about this (by us or as a fix) or if this is a limitation of OPC UA, of which I don't know enough myself to say for sure.

cristipogacean, 2022-10-27:
@NoTuxNoBux, would it be possible for you to enable debug level logs on the publisher and replicate the steps above? and of course share the logs with us?
The publisher has a mechanism to retry to reconnect broken monitored items. In this case apparently those monitored items are not reported (by server) respectively detected (by publisher) as broken. Maybe the logs can provide a bit more insights into the reasons.

NoTuxNoBux, 2022-10-27:
We don't currently have access to the physical device any more, but do have an emulator that is designed to behave exactly the same. I will try and test to see if the issue also reproduces there. If it does, it will be easier to debug and test.

It might take some time but I will try to get back with logs.

>  The publisher has a mechanism to retry to reconnect broken monitored items. In this case apparently those monitored items are not reported (by server) respectively detected (by publisher) as broken. Maybe the logs can provide a bit more insights into the reasons.

Does this work using a polling mechanism that works every so often? If so, is this in the magnitude of minutes or seconds? If it is seconds, or not using a polling mechanism, then it is likely indeed not being picked up at all, as we didn't realize what the problem was at first, left it like that for several minutes, and it never ended up recovering by itself.

hansgschossmann, 2022-12-05:
@NoTuxNoBux can you pls provide the requested info, otherwise i will close the issue and you can reactivate.

NoTuxNoBux, 2022-12-05:
@hansgschossmann Feel free to close in that case. As mentioned I no longer have access to said target device, and don't expect to anytime soon. I still believe this is an issue, though I can't aid in debugging it currently, so I understand if you'd rather close.

closed 2022-12-06 by marcschier