Azure/Industrial-IoT #1891: Increase BatchTriggerInterval granularity to milliseconds

NoTuxNoBux, 2022-10-27:
**Is your feature request related to a problem? Please describe.**
We have some OPC UA properties in our server that we'd like to sample and publish more often than once per second (positions of a robot that change as it moves); `BatchTriggerInterval` doesn't allow going below one second.

**Describe the solution you'd like**
Support millisecond granularity in `BatchTriggerInterval`.

**Describe alternatives you've considered**
We first tried to lower the batch size to 1 or a low size, but this results in a very large amount of requests being sent, even if they are virtually at the same time. Ideally we would still like to batch as much as possible to keep the amount of requests low, but lower the interval to a sub-second value so we can get batched updates sooner. 

**Additional context**
We can configure the sample rate and publish interval up to millisecond granularity for the desired properties in the configuration, but it seems the OPC Publisher still batches it because of the batch size and interval.

marcschier, 2022-11-08:
#1893

cristipogacean, 2022-11-09:
@NoTuxNoBux, can you try adding `BatchTriggerInterval` as environment variable for the publisher module having the value in `TimeSpan` format e.g. 00:00:00.100.
[image]

NoTuxNoBux, 2022-11-09:
@cristipogacean From an initial look, this seems to have no effect from my end.

closed 2022-11-10 by marcschier