Azure/Industrial-IoT #1897: Extending OPC Publisher messages with Application Properties

murdockcrc, 2022-11-11:
Hi,

Is there a way for the messages which the Publisher sends upstream (when running inside IoT Edge as a module) to be extended with [user properties](https://learn.microsoft.com/en-us/dotnet/api/microsoft.azure.devices.client.message.properties?view=azure-dotnet#microsoft-azure-devices-client-message-properties)?

This is because we need this metadata from the messages created by the OPC Publisher so that we can make routing decisions upstream Azure Data Explorer.

Curious if the Publisher supports that out of the box, for example, by reading these extension properties from the module's twin and adding them just before sending its messages to whatever module it is configured to route to. I read the operations guide here in GitHub but could not find anything related to this requirement.

Thx

marcschier, 2022-11-14:
For best flexibility it is recommend contextualizing the raw publisher data using a secondary IoT Edge module and route the publisher data through this module.  Documentation on how to build a module in a language of your choice you find [here](https://learn.microsoft.com/en-us/azure/iot-edge/module-development?view=iotedge-1.4). This is simple and reliable and can leverage not just twin properties but also external data sources if required.  Another option is to leverage [ASA](https://learn.microsoft.com/en-us/azure/iot-edge/tutorial-deploy-stream-analytics?view=iotedge-1.4) at the edge, depending on your needs.

murdockcrc, 2022-11-15:
Thx for the reply. What you suggest is a valid option, yes. Our organisation would prefer to not introduce another piece of custom code into the environment, that's why we would like to suggest for the Publisher to support such a scenario. But yes, the second module you suggest should be a pretty simple one to develop.

I noticed later in the docs that you do support adding an application property called `$$Routing` (or something like that). We are looking if this could actually help to solve our requirement. The best fit would be for the module to accept arbitrary properties through the module's twin (or environment variables, for that matter, something you can push from IoT Hub downstream), so we can add these values to the messages by configuration and allow dynamic routing of messages into our corresponding ADX tables with no custom code at all.

Thx for the suggestion.

cristipogacean, 2022-11-15:
@murdockcrc, indeed, `EnableRoutingInfo` is probably what you're looking for:
```
        EnableRoutingInfo=VALUE
                                  Adds the routing info to telemetry messages. The name of the property is 
                                  `$$RoutingInfo` and the value is the `DataSetWriterGroup` for that particular message.
                                  When the `DataSetWriterGroup` is not configured, the `$$RoutingInfo` property will
                                  not be added to the message even if this argument is set. 
                                  Alternative: --ri, --enableroutinginfo
                                  Mode: Standalone only
                                  Type: boolean
                                  Default: false
```
Please note that you can configure multiple `DataSetWriterGroups` even for the very same server endpoint if you need different telemetry streams routed to different destinations from the very same OPC UA server.

hansgschossmann, 2022-12-05:
No recent activity: closing. @murdockcrc pls reactivate if you have further questions.

closed 2022-12-05 by hansgschossmann

marcschier, 2023-07-03:
In 2.9 we now added the ability to add static fields to data set messages using the OPC UA Extension Fields capability. These can be anything and can be configured in publishednodes.json

murdockcrc, 2023-07-03:
@marcschier : that sounds great. Where do I find docs about how these properties work, or a sample of the publishednodes.json file with them configured? Is this already published too? Thx

perkops, 2023-09-14:
Hi @murdockcrc,

Did you manage to find a sample of this?

Otherwise, @marcschier  any update?

Kind Regards

marcschier, 2023-09-19:
Sorry, only just saw this. Here is the documentation: https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/readme.md#key-frames-delta-frames-and-extension-fields

murdockcrc, 2023-09-19:
@marcschier I think this is exactly what we were looking for. Will to a POC to test this but on paper, it looks pretty good. Thx for sharing.