Azure/Industrial-IoT #1902: Register UA Server: "ProcessErrorMessage(BadServiceUnsupported 'Error received from remote host"

mwiot, 2022-11-18:
Hi, 

I am getting an error when trying to add a new UA server. How do I go about troubleshooting this further?
Running version 2.8.1.2. 

[14:48:14 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) Channel 24: ProcessErrorMessage(BadServiceUnsupported 'Error received from remote host: ') 

`[14:48:14 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 6be1cd83-d8b1-4b18-b64f-dbd1b9cc5c55: Discovery operation started. 
[14:48:14 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 6be1cd83-d8b1-4b18-b64f-dbd1b9cc5c55: Searching 1 discovery urls for endpoints... 
[14:48:14 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 6be1cd83-d8b1-4b18-b64f-dbd1b9cc5c55: Trying to find endpoints on opc.tcp://10.0.1.10:4980/Softing_dataFEED_OPC_Suite_Configuration1... 
[14:48:14 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) Channel 24: ProcessErrorMessage(BadServiceUnsupported 'Error received from remote host: ') 
[14:48:14 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 6be1cd83-d8b1-4b18-b64f-dbd1b9cc5c55: Found 1 endpoints on opc.tcp://10.0.1.10:4980/Softing_dataFEED_OPC_Suite_Configuration1. 
[14:48:14 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 6be1cd83-d8b1-4b18-b64f-dbd1b9cc5c55: Found total of 1 servers ... 
[14:48:14 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Uploading 1 results... 
[14:48:14 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] 1 results uploaded. 
[14:48:14 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 6be1cd83-d8b1-4b18-b64f-dbd1b9cc5c55: Discovery operation completed. `

If I am doing a network scan with these settings:

[image]

`[14:52:17 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Discovery operation started. 
[14:52:17 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Start Fast discovery run... 
[14:52:17 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Starting network scan (0 probes active)... 
[14:52:17 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Start scanning ["10.0.1.0/24"]... 
[14:52:17 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.0.1.11 (0 scanned)... 
[14:52:17 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.0.1.10 (1 scanned)... 
[14:52:17 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.0.1.16 (2 scanned)... 
[14:52:17 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] New settings processed. 
[14:52:20 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] GC Mem: 7479 kb, Working set / Private Mem: 103908 kb / 83752 kb, Handles: 195 
[14:52:20 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : 3 addresses scanned - 3 ev.Discovered (100 probes active)... 
[14:52:20 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.0.1.1 (101 scanned)... 
[14:52:20 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.0.1.254 (102 scanned)... 
[14:52:23 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : 105 addresses scanned - 5 ev.Discovered (100 probes active)... 
[14:52:23 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found address 10.0.1.15 (147 scanned)... 
[14:52:26 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : 206 addresses scanned - 6 ev.Discovered (50 probes active)... 
[14:52:26 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found 6 addresses. (256 scanned)... 
[14:52:26 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Including gateway host address 172.18.0.5 
[14:52:26 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Starting port scanning (0 probes active)... 
[14:52:26 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Probe 1 : found OPC UA server at [::ffff:10.0.1.10]:4980 (protocol:0) ... 
[14:52:26 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found server 10.0.1.10:4980 (7 scanned)... 
[14:52:29 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] GC Mem: 10455 kb, Working set / Private Mem: 104260 kb / 98140 kb, Handles: 200 
[14:52:29 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : 7 ports scanned - 1 discovered (1 probes active)... 
[14:52:30 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Found 1 ports on servers (7 scanned)... 
[14:52:30 ERR Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] : Error An item with the same key has already been added. Key: 10.0.1.10:4980 during discovery run. `

Then its complaining that en item with the same key has been added. There is no sign of this registration anywhere in the eng tool or using the api queries in swagger. 

Hope you can clear this up, 
best regards

mwiot, 2022-11-21:
FYI: The endpoint Im trying to add "opc.tcp://10.0.1.10:4980/Softing_dataFEED_OPC_Suite_Configuration1" is working fine in other UA clients.

marcschier, 2022-11-21:
The error you see in the first case is expected.  The discovery operation is sending a FindServersOnNetwork request to see if the endpoint is an LDS-ME and thus can export not just local servers, but servers on the network that have registered with it.  Since an entry was uploaded, I do assume the first run worked and should show this server in the registry (maybe with some delay caused by processing).  

The second case seems to have exposed a bug in an unexpected case where the same discovery url was found because a discovery url and a scan was done in one call.  To work around, remove the discovery url from the discovery request _or_ the network scan configuration (IP address/port range).

mwiot, 2022-11-21:
Hi again. The strange thing is that it didnt work in case 1. It is never registered, even if it says 1 result uploaded.
There is some issue here as i mentioned with "BadServiceUnsupported" being displayed in the log before the result is uploaded. 

Case 2 kind of makes sense because you are finding the server in a scan and also trying to add it again using the discoveryuri, leading to a duplicate result. 

Case 3: What ended up working was doing the scan without the discoveryuri specified. Then it found the ua server + connected endpoints without any issue, and managed to register it. 

In case 1 im using the "/registry/v2/applications" api and discoveryurl parameter specified. This has been working previously for our other deployments, so I was a bit puzzled by this. 

[image]

marcschier, 2022-11-21:
Strange indeed, but the only reason I can think of here is that the onboarding processor did compare it to the existing and did not see a need to change anything (because it is already there), or it did not catch up to processing the event in time and eventually will add it.  If it never happens something in the processor or eventhub side is down or blocked as the "uploaded" log means the event was written to the edgehub and highly likely to iot hub.

Our end to end test are testing the registry/applications POST API and the tests are passing on main, so there must be something edgy going on.

The duplicate key exception I fixed earlier today, and it should be in the next release.

marcschier, 2022-11-24:
@mwiot, since we cannot repro this here, case 3 was a workaround for you, and since case 2 is fixed in 2.8.4, I will close this issue unless you have more information to add (in which case please re-open)