Azure/Industrial-IoT #1909: "Invalid URI: The hostname could not be parsed" error in metricscollector module

emulic, 2022-11-22:
After fresh deployment of new IIoT Edge device the metricscollector module is not properly working.
Following error can be found in the module log file:
```shell 
[2022-11-22 07:52:49 : Starting Module
[2022-11-22 07:52:50.636 INF] Initializing metrics collector
[2022-11-22 07:52:51.430 INF] Version - 1.0.11
[2022-11-22 07:52:51.454 DBG] Waiting on shutdown handler to trigger
[2022-11-22 07:52:52.079 INF] Metrics collector initialized with the following settings:
Settings:
LogAnalyticsWorkspaceId=/subscriptions/********-*****-****-****-***********/resourcegroups/****-test/providers/microsoft.operationalinsights/workspaces/workspace-*****
Endpoints=[
  "http://edgehub:9600/metrics",
  "http://edgeagent:9600/metrics",
  "http://twin:9701/metrics",
  "http://publisher:9702/metrics"
]
ScrapeFrequencySecs=300
UploadTarget=AzureMonitor
CompressForUpload=True
TransformForIoTCentral=False
ExperimentalFeatureAddIdentifyingTags=False
AllowedMetrics=
BlockedMetrics=
ResourceId=/subscriptions/XXX/resourceGroups/XXX/providers/Microsoft.Devices/IotHubs/iothub-*****
IotHubConnectFrequency=1.00:00:00
AzureDomain=azure.com
[2022-11-22 07:52:52.140 INF] Started operation Reconnect to IoT Hub
[2022-11-22 07:52:52.168 INF] Started operation Scrape and Upload Metrics
[2022-11-22 07:53:52.142 INF] Starting periodic operation Reconnect to IoT Hub...
[2022-11-22 07:53:52.170 INF] Trying to initialize module client using transport type [Amqp_Tcp_Only]
[2022-11-22 07:53:53.168 INF] Starting periodic operation Scrape and Upload Metrics...
[2022-11-22 07:53:53.216 INF] Scraping endpoint http://edgehub:9600/metrics
[2022-11-22 07:53:53.583 INF] Scraping endpoint http://edgeagent:9600/metrics
[2022-11-22 07:53:53.587 INF] Scraping endpoint http://twin:9701/metrics
[2022-11-22 07:53:53.589 INF] Scraping endpoint http://publisher:9702/metrics
[2022-11-22 07:53:54.406 INF] Scraping finished, received 18 metrics from endpoint http://publisher:9702/metrics
[2022-11-22 07:53:54.560 INF] Scraping finished, received 145 metrics from endpoint http://edgehub:9600/metrics
[2022-11-22 07:53:54.690 INF] Scraping finished, received 15 metrics from endpoint http://twin:9701/metrics
[2022-11-22 07:53:55.957 INF] Scraping finished, received 178 metrics from endpoint http://edgeagent:9600/metrics
[2022-11-22 07:54:04.822 INF] Successfully initialized module client using transport type [Amqp_Tcp_Only]
[2022-11-22 07:54:04.878 INF] Successfully completed periodic operation Reconnect to IoT Hub
[2022-11-22 07:54:05.802 INF] Successfully created self-signed certificate  for agentGuid : {e0923784-ef54-4c2d-ba0a-b9ae1b513f1d} and workspace: /subscriptions/********-*****-****-****-***********/resourcegroups/*****-test/providers/microsoft.operationalinsights/workspaces/workspace-*****
[2022-11-22 07:54:05.829 INF] OMS endpoint Url : https:///subscriptions/********-*****-****-****-***********/resourcegroups/*****-test/providers/microsoft.operationalinsights/workspaces/workspace-*****.oms.opinsights.azure.com/AgentService.svc/AgentTopologyRequest
[2022-11-22 07:54:05.834 INF] sending registration request
[2022-11-22 07:54:05.959 INF] OMS endpoint Url : https:///subscriptions/********-*****-****-****-***********/resourcegroups/*****-test/providers/microsoft.operationalinsights/workspaces/workspace-09586.oms.opinsights.azure.com/AgentService.svc/AgentTopologyRequest
[2022-11-22 07:54:05.960 INF] sending registration request
[2022-11-22 07:54:05.961 INF] OMS endpoint Url : https:///subscriptions/********-*****-****-****-***********/resourcegroups/*****-test/providers/microsoft.operationalinsights/workspaces/workspace-09586.oms.opinsights.azure.com/AgentService.svc/AgentTopologyRequest
[2022-11-22 07:54:05.961 INF] sending registration request
[2022-11-22 07:54:05.976 INF] OMS endpoint Url : https:///subscriptions/********-*****-****-****-***********/resourcegroups/*****-test/providers/microsoft.operationalinsights/workspaces/workspace-09586.oms.opinsights.azure.com/AgentService.svc/AgentTopologyRequest
[2022-11-22 07:54:05.976 INF] sending registration request
[2022-11-22 07:54:05.978 WRN] exception occurred : Invalid URI: The hostname could not be parsed.
[2022-11-22 07:54:05.981 ERR] Registering agent with OMS failed (are the Log Analytics Workspace ID and Key correct?) : Invalid URI: The hostname could not be parsed.
[2022-11-22 07:54:06.150 FTL] System.UriFormatException: Invalid URI: The hostname could not be parsed.
   at System.Uri.CreateThis(String uri, Boolean dontEscape, UriKind uriKind)
   at System.Uri..ctor(String uriString)
   at Microsoft.Azure.Devices.Edge.Azure.Monitor.Certificategenerator.CertGenerator.RegisterWithOms(X509Certificate2 cert, String AgentGuid, String logAnalyticsWorkspaceId, String logAnalyticsWorkspaceKey, String logAnalyticsWorkspaceDomainPrefixOms) in /mnt/vss/_work/1/s/edge-modules/metrics-collector/src/CertificateGenerator/CertGenerator.cs:line 186
   at Microsoft.Azure.Devices.Edge.Azure.Monitor.Certificategenerator.CertGenerator.RegisterWithOmsWithBasicRetryAsync(X509Certificate2 cert, String AgentGuid, String logAnalyticsWorkspaceId, String logAnalyticsWorkspaceKey, String logAnalyticsWorkspaceDomainPrefixOms) in /mnt/vss/_work/1/s/edge-modules/metrics-collector/src/CertificateGenerator/CertGenerator.cs:line 208
   at Microsoft.Azure.Devices.Edge.Azure.Monitor.Certificategenerator.CertGenerator.RegisterAgentWithOMS(String logAnalyticsWorkspaceId, String logAnalyticsWorkspaceKey, String logAnalyticsWorkspaceDomainPrefixOms) in /mnt/vss/_work/1/s/edge-modules/metrics-collector/src/CertificateGenerator/CertGenerator.cs:line 267
[2022-11-22 07:54:06.156 INF] Termination requested, initiating shutdown.
[2022-11-22 07:54:06.167 INF] Waiting for cleanup to finish
[2022-11-22 07:54:06.507 INF] Done with cleanup. Shutting down.
```
The module shuts down few times and after that it appears with the status "Error" in the list of modules in the Azure Portal.

**To Reproduce**

- Add required IioT tags (__type__: iiotedge and os:Linux) on an Edge Device and wait for deployment to be applied.
- After few minutes check the logs of the metricscollector module.

**Expected behavior**
The metricscollector module should be executed without any errors.

**Additional context**
The issue seems to be related to layered deplyoment definition for metricscollector module, which is wrongly setting the  value `LogAnalyticsWorkspaceId` to value like this `/subscriptions/********-*****-****-****-***********/resourcegroups/******-prod/providers/microsoft.operationalinsights/workspaces/workspace-******` instead of setting it to a guid.

This happens with IIoT version 2.8.3 but we observed the same issue before, with older versions of IIoT like 2.8.0 and  2.8.1.

marcschier, 2022-11-24:
Will be in 2.8.4 release

closed 2022-11-24 by marcschier

emulic, 2022-11-24:
Thanks Marc.