Azure/Industrial-IoT #1912: Get timestamp of OPC Publisher host in pubsub telemetry (as opposed to OPC server timestamp)?

b-enigma-con, 2022-11-22:
Hello all,

We are running OPC Publisher 2.8.3 together with a custom developed IoT Edge module which formats the OPC UA telemetry pubsub JSON to a custom format. The flow is OPC Publisher -> custom developed module -> IoT Hub. We use a polling-like mechanism to gather the data; publishing interval = sampling interval = heartbeat interval.

We use the default telemetry buffering time of the OPC Publisher of 10 seconds. This means that when the heartbeat interval is 5 sec, we would receive 2 messages at a time in our custom developed IoT Edge module.

The issue is that our custom JSON telemetry format (the one going to the IoT Hub) should have a field containing the timestamp of the moment that the data was fetched, based on the datetime of the OPC Publisher host. The reason of this is that the datetime of the connected PLC’s can’t be trusted in this case, they won’t be synced per se. So with the OPC Publisher pub-sub telemetry, we’re not able to use the sourcetimestamp and servertimestamp, because they are both derived  from the PLC’s datetime. We thought that the “Timestamp” field in the pubsub telemetry format would contain the timestamp of when the value was recorded, based on the datetime in the OPC Publisher, but this field also seems to contain a datetime based on the datetime set in the PLC (/OPC server). It boils down to the next question: when messages are buffered in the OPC Publisher, and we want to process them in another upstream IoT Edge module, how would we be able to get the datetime/timestamp of the moment the value was recorded based on the datetime of the OPC Publisher host itself (which we know is always properly synced). We can’t trust the timestamps associated with the PLC’s (/OPC servers) in this case. 

Thank you in advance.

hansgschossmann, 2022-12-05:
@b-enigma-con does this answers your question?

b-enigma-con, 2022-12-05:
Thank you, it surely does. I was not aware of the existence of those headers (I used https://learn.microsoft.com/en-us/azure/industrial-iot/reference-opc-publisher-telemetry-format as reference).