Azure/Industrial-IoT #1917: Official documentation on how to setup the OPCPublisher edge module with X.509 certificates

rmonvfer, 2022-11-29:
**Is your feature request related to a problem? Please describe.**
Yes, the lack of official documentation regarding this topic has been a constant issue for quite some time, forcing me to rely on unofficial / unchecked sources.

**Describe the solution you'd like**
I'd like to be able to find documentation on how to configure the OPC Publisher edge module in order to use X.509 certificates for OPC-UA authentication (NOT `iotedge` authentication)

**Describe alternatives you've considered**
I've been trying to find alternative resources such as blog posts and GitHub issues although my primary reference has been https://kevinsaye.wordpress.com/2021/02/24/using-custom-x509-certificates-in-iot-edges-opc-publisher/

**Additional context**
Most resources recommend terrible security practices (eg: copying a CA cert alongside its private key)

hansgschossmann, 2022-12-05:
@rmonvfer i do not exactly know what you mean by "Most resources recommend terrible security practices (eg: copying a CA cert alongside its private key)". Can you elaborate?

rmonvfer, 2022-12-12:
@hansgschossmann Sure, thank you for your reply.

I was not able to find any **official** documentation regarding X.509 authentication between the OPCPublisher edge module and a OPC Server so I've had to rely on **unofficial** sources which might be incorrect or inappropriate (or so do I believe).

That is just a side issue anyways, but (IMO) highlights the lack of proper docs regarding this specific topic.

Thank you,
Ramon

hansgschossmann, 2022-12-13:
Thanks for your honnest feedback. Appreciated. I will tag it as bug for tracking.

pesetskyps, 2023-07-05:
I agree, official docs would help

marcschier, 2023-10-05:
Added documentation to the opc-publisher readme.md (https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/readme.md#opc-ua-certificates)

closed 2023-10-05 by marcschier