Azure/Industrial-IoT #1923: OPC Publisher Sessions

sckrecords, 2022-12-12:
Hello,

I am trying to pull four (4) different OPC UA endpoints from one (1) OPC UA server using one instance of OPC Publisher. "published_nodes.json" is attached. Documentation states that OPC Publisher is creating one session per endpoint only:

"A single OPC UA session is opened to the asset (OPC UA server) on demand per endpoint, so the OPC UA server is not overburdened with simultaneous requests, which means the implementation of the OPC Twin module is stateless in respect to the OPC UA connections."

The OPC UA Server that we use allows only four (4) simultaneous sessions.

When I start OPC Publisher with four (4) or three (3) endpoints configured, I receive "Too many sessions". When I start OPC Publisher with two (2) or one (1) endpoint configured, everything works fine.

Could there be a condition, when OPC Publisher opens more than one (1) session to receive data for one (1) endpoint?

Thanks + kind regards

Manuel Schulz
Swiss Federal Railways SBB
[published_nodes.json.zip](https://github.com/Azure/Industrial-IoT/files/10206401/published_nodes.json.zip)

marcschier, 2022-12-12:
Hi @sckrecords, a session to a OPC UA server is established with an endpoint url (opens a single socket per endpoint URL). A server can expose more than 1 endpoint with different configurations.  In your case with 4 different endpoint urls, you will end up with 4 sessions with 1 subscription each.  

If the nodes you configured per endpoint url are available through all 4 endpoints, you could pick one of the 4 and use datasetwriterid to create 4 logical subscriptions on this chosen endpoint (and then have a single session) and then even route them to different outputs if you wanted.  If the nodes are only valid in the context of the endpoint url (and thus session) then unfortunately it is not possible.

sckrecords, 2022-12-12:
@marcschier What more information do I need to supply?

marcschier, 2022-12-12:
Oops, sorry, if my answer helps you, we could close this item.  

But I think the issue here is that you see too many sessions even with 3 endpoint urls, which means there must be more than 4 sessions in this case as well.  If that is so, I will leave this open as a bug and we shall investigate.  For the latter it would be great to have a supportbundle from this issue, or at least the logs from publisher so we can see what happened.

marcschier, 2022-12-12:
Super, please send me supportbundle.zip or the publisher log of when you hit the error so we can take a look.

sckrecords, 2022-12-13:
@marcschier Logs will be delivered on 15th (this Thursday).

marcschier, 2022-12-13:
No rush.

sckrecords, 2022-12-20:
Hi @RadouaneGaroiaz, of course you cannot connect to the OPC UA Server that is listed in published_nodes.json, since it is an internal host in our company network. :-)

RadouaneGaroiaz, 2022-12-20:
hello @sckrecords we have the same issue as you in our server, i didn't say that i tried to connect to your company server :) .

marcschier, 2023-05-19:
OPC Publisher 2.9.0-preview2 was just released and has a better implementation of OPC UA Client that s deterministic in the creation of sessions with a server. There is now only 1 session and content of the "ConnectionModel".  The connection model is created from the publishednodes.json information.  There could still be issues in the OPC UA reference stack that have more than a single session around during reconnect or connect, so we will move this to preview 3 to add validation tests for exactly this scenario.  Feedback on 2.9.0 preview 2 though are very welcome.

closed 2023-06-11 by marcschier