Azure/Industrial-IoT #1929: OpcPublisher v2.8.2 is not able to handle large number of tags

anil-k-gopalakrishna, 2022-12-21:
**Describe the bug**
We had deployed older version of opcpublisher (v2.5.4) in multiple terminals long back. Now we are trying to move to  v2.8.2; 
It is observed that the ignition server crashes in case of some terminals which are configured with **configured with large number of tags** (>60K)

**Observed behavior**
1. We are using old JSON schema, with non-encrypted user name and password and previous node-id based node definitions:
        _[{
          "EndpointUrl": "opc.tcp://xx.xx.xx.xx:xxxx/xxxx",            
          "OpcAuthenticationMode": "UsernamePassword",            
          "**OpcAuthenticationUsername**": "xxxxx",            
          "**OpcAuthenticationPassword**": "xxxxx",            
          "OpcNodes": [
              {
                  "DisplayName": "tag1",
                  "Id": "ns=2;s=[default]<path>/tag1",
                  "OpcPublishingInterval": 1000,
                  "OpcSamplingInterval": 1000
              },{
                  "DisplayName": "tag2",
                  "Id": "ns=2;s=[default]<path>/tag2",
                  "OpcPublishingInterval": 1000,
                  "OpcSamplingInterval": 1000
             },
             ....    
      ]}]_
2. When configured with less number of tags (<20K), it is working fine 
3. For some nodeIds, we are getting warnings given below:
   _10:52:26 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] DataChange for subscription 1000_ns=2;s=[default]/ALR/kpi/WasteTime/ContainerMove'/'opc.tcp://xx.xx.xx.xxx:4096/discovery_6429F824' has no monitored items yet_ 
4. After sometime, the ignition server which is being connected to get restart and we are seeing heavy load on its CPU

marcschier, 2022-12-21:
@anil-k-gopalakrishna, could you add full publisher logs with PII redacted?

anil-k-gopalakrishna, 2022-12-22:
please find attached publisher logs
[opcpublisher-test_log.txt](https://github.com/Azure/Industrial-IoT/files/10284837/opcpublisher-test_log.txt)

marcschier, 2022-12-23:
I see that every monitored item has a desired queue size of 24000, which is pretty large and will cause a lot of resource pressure on the server.  Can you send me also the command line you used as well as a snippet of the publishednodes.json file?  (On the latter you can remove all OpcNodes entries except for 2-3, presuming they are all the same configuration.

anil-k-gopalakrishna, 2022-12-29:
Please find container create option used in the deployment. Some of the Cmd options which are not applicable also present here; which are carried forward from 2.5.4:
{
    "Hostname": "opc-publisher",
    "Cmd": [
        "--pf=/appdata/pn.json",
        "--tc=/appdata/telemetryconfiguration.json",
        "--aa",
        "--di=10",
        "--si=10",
        "--ms=225000",
        "--mq=24000",
        "--ap=/pki/own",
        "--sf=True"
    ],
    "HostConfig": {
        "Binds": [
            "/var/iiotedge:/appdata",
            "/var/iiotedge/pki:/pki"
        ],
        "LogConfig": {
            "Type": "json-file",
            "Config": {
                "max-size": "10m",
                 "max-file": 3
            }
        }
    }
}

Please find publishednodes snippet as attached: 
[pn.zip](https://github.com/Azure/Industrial-IoT/files/10317866/pn.zip)

anil-k-gopalakrishna, 2023-01-03:
@marcschier , can you please update on the above. 
Kindly let me know if you are looking for any more inputs.

marcschier, 2023-01-06:
Hi, turns out we made a mistake interpreting the option --mq when we implemented configurable queue size on the server. In 2.5 --mq was used to configure the queue on publisher side, in 2.8 it is the queue inside the server.

We will fix this in 2.9, please try changing the value from 24000 to maybe 3 or less then see if it works. If not please could you send another log with lower queue size setting made?

anil-k-gopalakrishna, 2023-01-10:
Thanks @Marcschier . It worked. Now the module is running promptly.

Only one issue I noticed; we are getting some warnings as "Unexpected Sequence Numbers". Out of 60K tags, with the same configuration, I am getting these warning only for these 4 tags. 
[image]

I understand that this is due to the mismatch in 'message sequence number' used in communication sync. Can you please suggest how we can get rid of these warnings.

marcschier, 2023-01-13:
Several issues in the logging in this area have been addressed in 2.8.4.  I recommend upgrading anyway as 2.8.4 is the latest supported patch. It fixes some CVE and also uses .net 6.  But otherwise the trace is benign as we were erroneously tracking the keep alive seq. Number which does not update.

closed 2023-01-17 by marcschier