Azure/Industrial-IoT #1937: OPC Publisher: Failed to decode the CRL

CONeal, 2023-01-16:
While connection the OPC Publisher in a project I need to add the whole Client Certificate chain including the CRLs for die Client Certificate of the OPC Server.
Reading the CRL of the Root-CA is not possible and I'll get the following error Message:

```
[ERR] Error while trying to read information from trusted issuer store.
System.Security.Cryptography.CryptographicException: Failed to decode the CRL.
---> System.Formats.Asn1.AsnContentException: The provided data is tagged with 'Universal' class value '24', but it should have been 'ContextSpecific' class value '0'.
   at System.Formats.Asn1.AsnDecoder.CheckExpectedTag(Asn1Tag tag, Asn1Tag expectedTag, UniversalTagNumber tagNumber)
   at System.Formats.Asn1.AsnDecoder.ReadSequence(ReadOnlySpan`1 source, AsnEncodingRules ruleSet, Int32& contentOffset, Int32& contentLength, Int32& bytesConsumed, Nullable`1 expectedTag)
   at System.Formats.Asn1.AsnReader.ReadSequence(Nullable`1 expectedTag)
   at Opc.Ua.Security.Certificates.X509CRL.DecodeCrl(Byte[] tbs)
   --- End of inner exception stack trace ---
   at Opc.Ua.Security.Certificates.X509CRL.DecodeCrl(Byte[] tbs)
   at Opc.Ua.Security.Certificates.X509CRL.Decode(Byte[] crl)
   at Microsoft.Azure.IIoT.OpcUa.Protocol.OpcConfigEx.ShowCertificateStoreInformationAsync(ApplicationConfiguration appConfig, ILogger logger)
```
The CRL is in DER-Format with and .crl-Ending
I'm also using the --aa flag in the OPC Publisher Create Container Options, but it seems to be ignored.

The certificates are required as soon as I enable UsernamePassword Authentication (UseSecurity-Flag is set to false)

**Expected behavior**
The CRL should be loaded without any error message and the connection between the Publisher and the OPC Server should be successful.

**Container Create Options**
``` json
{
  "Hostname": "OPCPublisher",
  "Cmd": [
    "--pf=/appdata/telemetrynodes.json",
    "--lf=/appdata/logs/telemetry.log",
    "--PkiRootPath=/appdata/pki",
    "--aa",
    "--tm",
    "--di=60",
    "--MessagingMode=PubSub",
    "--BatchSize=1",
    "--loglevel=verbose"
  ],
  "HostConfig": {
    "Binds": ["/iiotedge:/appdata"]
  }
}
```

**Desktop**
 - OS: Red Hat Enterprise Linux
 - Publisher-Version: latest

CONeal, 2023-01-16:
cc: @vslepakov

vslepakov, 2023-01-16:
+ @mregen

CONeal, 2023-01-16:
I checked it and it seems to be another issue. The errior message in my case doesn't appear in the other one.
But I also checked it on die Machine, set the Publiher Version to 2.8.4 (which currently loads 2.8.4.8) and added the parameter - no change, I've still got the same exception.

mregen, 2023-01-16:
Hi @CONeal , please check if you use a CRL which is encoded as PEM:
https://github.com/OPCFoundation/UA-.NETStandard/issues/1766#issuecomment-1090133780

if not, please provide the CRL for investigation to repro the issue.

CONeal, 2023-01-17:
Hi @mregen,

no, the CRL is in DER-Format.
[Root-CA.zip](https://github.com/Azure/Industrial-IoT/files/10433099/Root-CA.zip)

mregen, 2023-01-17:
Hi @CONeal, its a bug in the X509CRL decoder. 
The issue is the 'NextUpdate' > 2050. It is correctly encoded as GeneralizedTime but the X509CRL decoder doesn't handle it correctly. As a workaround, if you can use a CRL with NextUpdate < 2050 its ok.

mregen, 2023-01-17:
Hi @CONeal, this is a known issue with update times >year 2050 which was addressed here: 
https://github.com/OPCFoundation/UA-.NETStandard/pull/1959
The workaround until the UA .NET stack >= 1.4.371.41 is integrated in the next release of the IIoT platform is to use CRL which do not contain update times > 2050.

CONeal, 2023-01-18:
@mregen Thanks for the check and the workaround. 

@cristipogacean I also got back to your idea. It's a workaround if it's not possible to change the NextUpdate field. When working with the RejectUnknownRevocationStatus it's required to remove the existing CRL file(s) which can't be loaded, because the Publisher always loads the files at startup. First, I assumed that the Flags will be handled first and afterwards the required PKI data get loaded.

marcschier, 2023-05-19:
Preview 2  is now out with latest OPC UA stack.

closed 2023-05-19 by marcschier