Azure/Industrial-IoT #1947: Endpoint Activation returns Bad User Access Denied

JaivyDaam, 2023-02-04:
**Describe the bug**
Using the engineering tool, I have discovered endpoints that uses the UserName AuthenticationMethods. When trying to "Turn on" the endpoint, it is showing an "Unauthorized" Endpoint State. The link to the endpoint becomes clickable, but the return is "Bad User Access Denied".

Looking at the documentation of the "AuthenticationMethodsApiModel" it can receive/pass a "configuration" parameter which I have added to the applications Tags under the found AuthenticationMethods using "Configuration": "user:pass" inside the IoT Hub, manually.

The lack of examples for this specific parameter of "configuration" under https://github.com/Azure/Industrial-IoT/blob/main/docs/api/registry/definitions.md#authenticationmethodapimodel 
makes it impossible to understand what the service is expecting using this credentialtype.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup OPCUA server with Username & Password and Device Certificate trust
2. Discover Endpoints
3. Turn on Discovered Endpoint
4. See error: Bad User Access Denied

**Expected behavior**
I expect that given the "configuration" tag has been added to the AuthenticationMethod object, that I can login into the endpoint and browse the nodes. It would also definitely help if I get a User/Pass field when this occurs.

**Note**
This is my first time posting an issue on Github like this, if I can anywhere explain my situation better and/or edit my issue where I explain it too vaguely, please tell me so I can improve my report writing skills.

JaivyDaam, 2023-02-10:
Hello, I have it working.
The "Asset Login" was also overlooked completely by me. After finding #1473 I noticed I wasn't the only one.

My relevant question would be, how do I pass this information on via the API as the C# models doesn't show me the proper use of such configuration. Any additional information on this would be welcomed.

Thank you

marcschier, 2023-02-14:
This can be accomplished using the Elevation property (CredentialApiModel) in the ServiceRequestHeaderModel model in every request.  It will elevate the call to the credential and execute and then revert the elevation for the next call if it has different credentials.  Note that the Credential model is not typed, i.e., the username and password properties must be set like this: https://github.com/Azure/Industrial-IoT/blob/9750d55b33f35baf69c9705ef297b12bcce7cb60/samples/src/Microsoft.Azure.IIoT.App/src/Services/Publisher.cs#L146

closed 2023-02-14 by marcschier