Azure/Industrial-IoT #1948: Batchsize with IoTHubSendInterval not working as intended

ZachM05, 2023-02-08:
**Describe the bug**
Batchsize and IoTHubSendInterval set in command line arguments but do not seem to be affecting how often messages are sent to the IoTHub

**To Reproduce**
Steps to reproduce the behavior:
1. Go to IoT Hub
2. Set modules for IoT Edge device, with image uri for OPCPublisher being `mcr.microsoft.com/iotedge/opc-publisher:latest`. The command line 
3. Set container create options to the following:
```json
{
    "Hostname": "opcpublisher",
    "Cmd": [
        "--PkiRootPath=/mount/pki",
        "--pf=/mount/published_nodes.json",
        "--lf=/mount/publisher.log",
        "--mm=PubSub",
        "--me=Json",
        "--fm=true",
        "--fd=false",
        "--bs=1028",
        "--di=20",
        "--sc=1",
        "--si=60",
        "--aa"
    ],
    "HostConfig": {
        "Mounts": [
            {
                "Type": "bind",
                "Source": "/home/pi/opcpublisher",
                "Target": "/mount"
            }
        ]
    }
}
```
4. Configure routes to default endpoints
5. Create deployment for device

The `published_nodes.json` looks as follows:

```json
[
    {
        "EndpointUrl": "opc.tcp://{host}:4840",
        "UseSecurity": true,
        "OpcAuthenticationMode": "UsernamePassword",
        "OpcAuthenticationUsername": "{username}",
        "OpcAuthenticationPassword": "{password}",
        "OpcNodes": [
            {
                "Id": "{nodeId}"
            }
            ]
    }
]
```

Filled in with my server's information and ~185 nodes in the array. 

**Expected behavior**
I am following the [documentation here](https://learn.microsoft.com/en-us/azure/industrial-iot/reference-command-line-arguments) for setting command line arguments, and my goal is to be able to monitor specific nodes at different rates, store data changes in the batch/queue, and submit them in a message at the interval specified with --si. However, even with `--si=60` and `--bs=1028`, the messages to the IoT Hub are still happening every few seconds (~10s/msg). 

Am I misunderstanding the documentation above regarding the purpose of batchsize and iothubsendinterval?

**Additional context**
Add any other context about the problem here.
**Edge Device:**
- OS: Ubuntu 20.04.4
- IoT Edge Version: 1.3.0

marcschier, 2023-02-14:
If you have received more than 1028 notifications a batch message is emitted or if the interval has expired and there are not enough notifications yet to be sent (60 seconds).  Can it be that there are more than 1028 notifications every 10 seconds or so?  It is also possible that messages queue up for sending, which are also sent immediately and not held back.  Best would be to use a deterministic OPC UA node as source and try this e.g., you could use the OPC PLC simulation server (which is another project under Azure-Samples organization).

ZachM05, 2023-02-14:
@marcschier Thank you for the response! I have taken your advice and set the `OpcNodes` in `published_nodes.json` to just 1 node. This has allowed me to get to a somewhat desirable state. As of right now, the messages are coming through as follows:
```json
[
    {
      "MessageId": "41",
      "MessageType": "ua-data",
      "PublisherId": "opc.tcp://{ip-address}:{port}_{identifier}",
      "Messages": [
        {
          "DataSetWriterId": "1000",
          "SequenceNumber": 41,
          "MetaDataVersion": {
            "MajorVersion": 1,
            "MinorVersion": 0
          },
          "Timestamp": "2023-02-14T14:52:08.397Z",
          "Payload": {
            "ns=6;s=::AsGlobalPV:OPC.DM2EXT.OEEDataCtr": {
              "Value": 145,
              "SourceTimestamp": "2023-02-14T14:52:07.607Z",
              "ServerTimestamp": "2023-02-14T14:52:07.607Z"
            }
          }
        }
      ]
    },
    {
      "MessageId": "42",
      "MessageType": "ua-data",
      "PublisherId": "opc.tcp://{ip-address}:{port}_{identifier}",
      "Messages": [
        {
          "DataSetWriterId": "1000",
          "SequenceNumber": 42,
          "MetaDataVersion": {
            "MajorVersion": 1,
            "MinorVersion": 0
          },
          "Timestamp": "2023-02-14T14:52:11.397Z",
          "Payload": {
            "ns=6;s=::AsGlobalPV:OPC.DM2EXT.OEEDataCtr": {
              "Value": 146,
              "SourceTimestamp": "2023-02-14T14:52:10.634Z",
              "ServerTimestamp": "2023-02-14T14:52:10.634Z"
            }
          }
        }
      ]
    }
]
```
This is a very close step to where I would like to be, as it is sending these messages every 60 seconds as I defined in `BatchTriggerInterval`. However, it is sending multiple `MessageId`s, whereas I would like it to send one `MessageId` with multiple `Messages` in the array. Since each change has a timestamp associated with it, this would not cause an issue because the post-processing can handle the ingress messages. Is this possible?

marcschier, 2023-02-21:
This will be enabled in 2.9.0 release by supporting strict compliance with part 6 and 14.  A network message will then containore than one dataset in batching mode.  Would that be the desired outcome?

ZachM05, 2023-02-21:
@marcschier Yes! That would be a great feature for my project. Much appreciated and looking forward to the 2.9.0 release!

closed 2023-02-22 by ZachM05