Azure/Industrial-IoT #1949: Dockerfile for OPC Publisher

simith, 2023-02-21:
**Is your feature request related to a problem? Please describe.**
Where can i find the Docerfile for building the OPC Publisher, i want to add the configuration file so that i can run it in standalone mode

**Describe the solution you'd like**
Please provide the Dockerfile

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

marcschier, 2023-04-01:
You can use the official image as base image and add the file to it to create a derivative image.  In 2.9 we will support the new dotnet capability to publish containers using dotnet publish

marcschier, 2023-04-28:
Now supporting dotnet publish to publish the OPC publisher application as docker container.

closed 2023-04-28 by marcschier