Azure/Industrial-IoT #1953: OPC Publisher 2.8.4 : BadMonitoredItemIdInvalid

Ilheu, 2023-03-06:
I'm running OPCPublisher and when I try to get some node's value I get this message: :"StatusCode":{"Symbol":"BadMonitoredItemIdInvalid","Code":2151809024}. 
I never had this problems in other servers.

I use the same nodeId that UaExpert shows:
[image]

My file configurations is the following one:

marcschier, 2023-03-07:
There is a special case BadMonitoredItemIdInvalid is emitted by OPC Publisher and that is during a Heartbeat should the value not exist or the monitored item not yet be created.  Since from the configuration heartbeat is not configured the only possibility is that this value is reported by the server.  Here the possibility is that you used different security or endpoint, or credentials between UA Expert and OPC Publisher (usesecurity = false).  Best to make sure to use a unsecure endpoint in UA expert and repro.  If that does not help, please could you send a) the command line argument for the publisher, and b) the entire publisher log, including how the monitored items are created on startup.

Ilheu, 2023-03-07:
> Best to make sure to use a unsecure endpoint in UA expert and repro. 

[image]

>  please could you send a) the command line argument for the publisher, 

[2023-03-07 15:01:45.617 INF Root] Module exits... 
[2023-03-07 15:01:45.622 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Stopping worker supervisor 
[2023-03-07 15:01:45.624 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Stopping worker... 
[2023-03-07 15:01:45.624 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Disposing processing engine opc.tcp://192.168.18.36:4861_3182BABF 
[2023-03-07 15:01:45.627 INF OpcUa] Subscription PUBLISHING DISABLED, Id=2, LastNotificationTime=15:01:44, GoodPublishRequestCount=1, PublishingInterval=1000, KeepAliveCount=10, PublishingEnabled=False, MonitoredItemCount=1 
[2023-03-07 15:01:45.628 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription '1000' unregistered from session 'opc.tcp://192.168.18.36:4861_3182BABF' in state Running 
[2023-03-07 15:01:45.628 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Closing subscription '1000' in session 'opc.tcp://192.168.18.36:4861_3182BABF'... 
[2023-03-07 15:01:45.628 INF OpcUa] Subscription DELETE, Id=2, LastNotificationTime=15:01:44, GoodPublishRequestCount=1, PublishingInterval=1000, KeepAliveCount=10, PublishingEnabled=False, MonitoredItemCount=1 
[2023-03-07 15:01:45.631 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Subscription '1000' in session 'opc.tcp://192.168.18.36:4861_3182BABF' successfully closed (Remaining: 0). 
[2023-03-07 15:01:45.632 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker StandalonePublisher_0 cancellation received ... 
[2023-03-07 15:01:45.632 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: StandalonePublisher_0, Job: opc.tcp://192.168.18.36:4861_3182BABF processing completed ...  
[2023-03-07 15:01:45.632 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker StandalonePublisher_0 stopping... 
[2023-03-07 15:01:45.632 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker stopped. 
[2023-03-07 15:01:45.632 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Worker supervisor successfully stopped 
[2023-03-07 15:01:45.636 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Stopping all sessions 
[2023-03-07 15:01:45.637 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Successfully stopped all sessions 
[2023-03-07 15:01:45.647 INF Microsoft.Azure.IIoT.Http.HealthChecks.HealthCheckManager] Health checks stopped. 
[2023-03-07 15:01:45.654 INF Root] Stopped prometheus metric server 
[2023-03-07 15:01:45.657 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Stopping Module Host... 
[2023-03-07 15:01:45.663 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 102: Module railesIoTedge_mcm_00_OPCPublisher disconnected due to Client_Close - now Disabled... 
[2023-03-07 15:01:45.668 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host stopped. 
[2023-03-07 15:01:45.671 INF Root] Module stopped. 
[2023-03-07 15:01:45.675 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Stopping all sessions 
[2023-03-07 15:01:45.675 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Successfully stopped all sessions 
[2023-03-07 15:01:46.391 INF OpcUa] ChannelId 27131: in Closing state. 
[2023-03-07 15:01:46.400 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Stopping worker supervisor 
[2023-03-07 15:01:46.400 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Worker supervisor successfully stopped 
[2023-03-07 15:01:46.406 WRN OpcUa] ChannelId 27131: Force reconnect reason=BadConnectionClosed 'Remote side closed connection' 
[2023-03-07 15:01:46.407 INF OpcUa] ChannelId 27131: in Closed state. 
[2023-03-07 15:01:46.407 INF OpcUa] ChannelId 27131: CLIENTCHANNEL SOCKET CLOSED: 02819C2C 
[2023-03-07 15:01:46.407 ERR OpcUa] Publish #2834, Subscription count = 0, Error: Object reference not set to an instance of an object. 
[2023-03-07 15:01:46.407 ERR OpcUa] Publish abandoned after error because session id changed: Old ns=1;i=1678170367 != New null 
[2023-03-07 15:01:47.532 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to railesedge using MqttOverTcp. 
[2023-03-07 15:01:47.610 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.StandaloneJobOrchestrator] File ../appdata/pn.json has changed, last known hash , new hash 46BD7DFE8393D0F9B173F78D3FC4EB5C86FC1E36316E929CE09D031D890BBBF2, reloading... 
[2023-03-07 15:01:47.611 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.StandaloneJobOrchestrator] Validation schema file  does not exist or is disabled, ignoring validation of ../appdata/pn.json file. 
[2023-03-07 15:01:47.753 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Read 1 entry models from published nodes file in 00:00:00.1415005 
[2023-03-07 15:01:47.801 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Total count of OpcNodes after job conversion: 1 
[2023-03-07 15:01:47.801 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Converted published nodes entry models to jobs in 00:00:00.0414449 
[... cut ...]
  # Estimated IoT Chunks (4 KB) per day:          8,640
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :              4 (0.1/s)
  # Connection retries                 :              0
  # Opc endpoint connected?            :           True
  # Monitored Opc nodes succeeded count:              1
  # Monitored Opc nodes failed count   :              0

Ilheu, 2023-03-07:
>  **Best to make sure to use a unsecure endpoint in UA expert and repro.** 

[image]
[image]

>  **please could you send a) the command line argument for the publisher**

{
  "Hostname": "opcpublisher",
  "Cmd": [
    "--pf=../appdata/pn.json",
    "--mm=PubSub",
    "--me=Json",
    "--fm=true",
    "--bs=100",
    "--si=1",
    "--hb=10",
    "--oi=1000",
    "--op=1000",
    "--di=20",
    "--aa"
  ],
  "HostConfig": {
    "Binds": [
      "/iiotedge:/appdata"
    ]
  }
}

>  **and b) the entire publisher log, including how the monitored items are created on startup**

[2023-03-07 15:01:45.617 INF Root] Module exits... 
[2023-03-07 15:01:45.622 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Stopping worker supervisor 
[2023-03-07 15:01:45.624 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Stopping worker... 
[2023-03-07 15:01:45.624 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Disposing processing engine opc.tcp://192.168.18.36:4861_3182BABF 
[2023-03-07 15:01:45.627 INF OpcUa] Subscription PUBLISHING DISABLED, Id=2, LastNotificationTime=15:01:44, GoodPublishRequestCount=1, PublishingInterval=1000, KeepAliveCount=10, PublishingEnabled=False, MonitoredItemCount=1 
[2023-03-07 15:01:45.628 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription '1000' unregistered from session 'opc.tcp://192.168.18.36:4861_3182BABF' in state Running 
[2023-03-07 15:01:45.628 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Closing subscription '1000' in session 'opc.tcp://192.168.18.36:4861_3182BABF'... 
[2023-03-07 15:01:45.628 INF OpcUa] Subscription DELETE, Id=2, LastNotificationTime=15:01:44, GoodPublishRequestCount=1, PublishingInterval=1000, KeepAliveCount=10, PublishingEnabled=False, MonitoredItemCount=1 
[2023-03-07 15:01:45.631 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Subscription '1000' in session 'opc.tcp://192.168.18.36:4861_3182BABF' successfully closed (Remaining: 0). 
[2023-03-07 15:01:45.632 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker StandalonePublisher_0 cancellation received ... 
[2023-03-07 15:01:45.632 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: StandalonePublisher_0, Job: opc.tcp://192.168.18.36:4861_3182BABF processing completed ...  
[2023-03-07 15:01:45.632 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker StandalonePublisher_0 stopping... 
[2023-03-07 15:01:45.632 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker stopped. 
[2023-03-07 15:01:45.632 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Worker supervisor successfully stopped 
[2023-03-07 15:01:45.636 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Stopping all sessions 
[2023-03-07 15:01:45.637 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Successfully stopped all sessions 
[2023-03-07 15:01:45.647 INF Microsoft.Azure.IIoT.Http.HealthChecks.HealthCheckManager] Health checks stopped. 
[2023-03-07 15:01:45.654 INF Root] Stopped prometheus metric server 
[2023-03-07 15:01:45.657 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Stopping Module Host... 
[2023-03-07 15:01:45.663 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 102: Module railesIoTedge_mcm_00_OPCPublisher disconnected due to Client_Close - now Disabled... 
[2023-03-07 15:01:45.668 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host stopped. 
[2023-03-07 15:01:45.671 INF Root] Module stopped. 
[2023-03-07 15:01:45.675 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Stopping all sessions 
[2023-03-07 15:01:45.675 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Successfully stopped all sessions 
[2023-03-07 15:01:46.391 INF OpcUa] ChannelId 27131: in Closing state. 
[2023-03-07 15:01:46.400 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Stopping worker supervisor 
[2023-03-07 15:01:46.400 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Worker supervisor successfully stopped 
[2023-03-07 15:01:46.406 WRN OpcUa] ChannelId 27131: Force reconnect reason=BadConnectionClosed 'Remote side closed connection' 
[2023-03-07 15:01:46.407 INF OpcUa] ChannelId 27131: in Closed state. 
[2023-03-07 15:01:46.407 INF OpcUa] ChannelId 27131: CLIENTCHANNEL SOCKET CLOSED: 02819C2C 
[2023-03-07 15:01:46.407 ERR OpcUa] Publish #2834, Subscription count = 0, Error: Object reference not set to an instance of an object. 
[2023-03-07 15:01:46.407 ERR OpcUa] Publish abandoned after error because session id changed: Old ns=1;i=1678170367 != New null 
[2023-03-07 15:01:47.532 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to railesedge using MqttOverTcp. 
[2023-03-07 15:01:47.610 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.StandaloneJobOrchestrator] File ../appdata/pn.json has changed, last known hash , new hash 46BD7DFE8393D0F9B173F78D3FC4EB5C86FC1E36316E929CE09D031D890BBBF2, reloading... 
[2023-03-07 15:01:47.611 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.StandaloneJobOrchestrator] Validation schema file  does not exist or is disabled, ignoring validation of ../appdata/pn.json file. 
[... cut ...]
   Estimated IoT Chunks (4 KB) per day:          8,640
   Outgress Batch Block buffer size   :              0
   Outgress input buffer count        :              0
   Outgress input buffer dropped      :              0
   Outgress IoT message count         :              4 (0.1/s)
   Connection retries                 :              0
   Opc endpoint connected?            :           True
   Monitored Opc nodes succeeded count:              1
   Monitored Opc nodes failed count   :              0

Ilheu, 2023-03-07:
> in addition, it would be interesting to try and subscribe / create a monitored item for this particular node using UA Expert and check it this works. It could be the case that the server does not support subscribing certain nodes, but only allows read access.

Thank you for the reply! Can you explain me how to do it? How do I check if the server does not support subscribing?

cristipogacean, 2023-03-07:
> Thank you for the reply! Can you explain me how to do it? How do I check if the server does not support subscribing?

You did exactly what I was looking for in the print-screen in your comment above with the `Data Access View`. So the node can be properly monitored. 

On the other hand, looking at the publisher's traces it looks the item is properly monitored now and value-changes are properly generated. Do you still see the `BadMonitoredItemIdInvalid` StatusCode in payload?

Ilheu, 2023-03-07:
> On the other hand, looking at the publisher's traces it looks the item is properly monitored now and value-changes are properly generated. Do you still see the `BadMonitoredItemIdInvalid` StatusCode in payload?

Yes, the payload is still the same...

marcschier, 2023-03-09:
This error is injected when the heartbeat (which is tied to keep alive) is supposed to be sent, but no previous value has ever been received by publisher or the monitored item has not yet successfully been created.  I would suggest removing "--hb" command line argument and see if you still see the status code.  Please let us know if this helped.

Ilheu, 2023-03-13:
> Please let us know if this helped.
It did not help... Now I don't receive any message...

Ilheu, 2023-03-14:
I'm trying troubleshoot the opc server with discovery and opc-twin module, but I can't use opc-twin without other services(microservices, etc) is there any way to do it with just the modules?? I've read the documentation and can't find any solution. 

@marcschier @cristipogacean can you give me any tip or hint?

JohannesM24, 2023-03-15:
Hi @Ilheu ,

from my personal experience regarding those problems, you could try to create a wireshark capture of the (unencrypted) OPC UA communication. You would then be able to compare the working setup of using UA-Expert and the one using opc-publisher. Find a manual on capturing OPC UA packets here https://opcconnect.opcfoundation.org/2017/02/analyzing-opc-ua-communications-with-wireshark/

Ilheu, 2023-03-16:
> Hi @Ilheu ,
>
> from my personal experience regarding those problems, you could try to create a wireshark capture of the (unencrypted) OPC UA communication. You would then be able to compare the [...]

Thank for your hint! I tried and got these logs...

[image]

I get this Error Message in loop, but when I try with UaExpert it works well... 
Can you help me? If it is a server problem, it was not supposed to work in UaExpert... 

@marcschier @cristipogacean What more information do you need?!

marcschier, 2023-03-21:
Many thanks for digging so deep into it. Super helpful.  The server errors on the transport channel layer after server returns the read response.  There might be an issue with us sending a namespace read that the server does not like.  Possible issue could be operation limits.  Could you share the pcap for UA expert and OPC publisher session starting from open 
secure channel including initial reads?  Btw 2.8.5 uses an updated OPC UA stack and should you need to redo the capture it would be awesome to do it with 2.8.5.

Ilheu, 2023-03-21:
> Many thanks for digging so deep into it. Super helpful. The server errors on the transport channel layer after server returns the read response. There might be an issue with us sending a namespace [...]

I will make the pcap as soon as possible! 
I think the server only has the most basic methods, I made a python script to get the data but besides that I just get errors. Unfortunately I have no control over the server.

Ilheu, 2023-03-22:
> Many thanks for digging so deep into it. Super helpful. The server errors on the transport channel layer after server returns the read response. There might be an issue with us sending a namespace [...]

@marcschier

**I tested with the latest version 2.8.5.1.** 

[2023-03-22 10:28:06.130 INF Root] Starting module OpcPublisher version 2.8.5.1. 
[2023-03-22 10:28:06.130 INF Root] Initiating prometheus at port 9702/metrics 
[2023-03-22 10:28:06.149 INF Root] Prometheus metric server started. 
[2023-03-22 10:28:06.158 INF Microsoft.Azure.IIoT.Http.HealthChecks.HealthCheckManager] Health checks started. 
[2023-03-22 10:28:06.162 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client. 
[2023-03-22 10:28:06.168 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context. 
[2023-03-22 10:28:07.131 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module railesIoTedge_mcm_00_OPCPublisher reconnected due to Connection_Ok. 
[2023-03-22 10:28:07.436 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Initialize device twin for railesIoTedge_mcm_00 - OPCPublisher 
[2023-03-22 10:28:07.525 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Applying initial desired state. 
[2023-03-22 10:28:07.530 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Reporting currently initial state. 
[2023-03-22 10:28:07.711 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host started. 
[2023-03-22 10:28:07.719 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker with id StandalonePublisher_0 
[2023-03-22 10:28:07.720 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'StandalonePublisher_0'... 
[2023-03-22 10:28:07.721 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Starting worker StandalonePublisher_0: {} 
[2023-03-22 10:28:07.800 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] No application own certificate found. Creating a self-signed own certificate valid since yesterday for 12 months, with a 2048 bit key and 256 bit hash. 
[2023-03-22 10:28:07.808 INF OpcUa] Checking application instance certificate. 
[2023-03-22 10:28:07.813 INF OpcUa] Creating application instance certificate. 
[2023-03-22 10:28:08.481 INF OpcUa] Adding application certificate to trusted peer store. [CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=railesedge] [AF363EF054C185B23114632DF05AFFFD8EA06B4C] 
[2023-03-22 10:28:08.482 INF OpcUa] Added application certificate to trusted peer store. 
[2023-03-22 10:28:08.573 INF OpcUa] Imported the PFX private key for [AF363EF054C185B23114632DF05AFFFD8EA06B4C]. 
[2023-03-22 10:28:08.584 INF OpcUa] Certificate created for urn:railesedge:Microsoft.Azure.IIoT:microsoft:. [CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=railesedge] [AF363EF054C185B23114632DF05AFFFD8EA06B4C] 
[2023-03-22 10:28:08.585 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Own certificate Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=railesedge' (thumbprint: AF363EF054C185B23114632DF05AFFFD8EA06B4C) created. 
[2023-03-22 10:28:08.595 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Application own certificate store contains 1 certs. 
[2023-03-22 10:28:08.595 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] 01: Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=railesedge' (thumbprint: AF363EF054C185B23114632DF05AFFFD8EA06B4C) 
[2023-03-22 10:28:08.596 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Trusted issuer store contains 0 certs. 
[2023-03-22 10:28:08.597 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Trusted issuer store has 0 CRLs. 
[2023-03-22 10:28:08.598 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Trusted peer store contains 1 certs. 
[2023-03-22 10:28:08.598 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] 01: Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=railesedge' (thumbprint: AF363EF054C185B23114632DF05AFFFD8EA06B4C) 
[2023-03-22 10:28:08.598 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Trusted peer store has 0 CRLs. 
[2023-03-22 10:28:08.598 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Rejected certificate store contains 0 certs. 
[2023-03-22 10:28:08.629 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker StandalonePublisher_0 processing job opc.tcp://192.168.18.36:4861_79E4FF2A, mode: Active 
[2023-03-22 10:28:08.718 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription '500' registered/updated in session 'opc.tcp://192.168.18.36:4861_79E4FF2A' in state Init 
[2023-03-22 10:28:08.726 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription '500' registered/updated in session 'opc.tcp://192.168.18.36:4861_79E4FF2A' in state Init 
[2023-03-22 10:28:08.738 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker+JobProcess] Job opc.tcp://192.168.18.36:4861_79E4FF2A started. 
[2023-03-22 10:28:08.932 INF OpcUa] ChannelId 0: in Connecting state. 
[2023-03-22 10:28:09.761 INF OpcUa] ChannelId 0: in Opening state. 
[2023-03-22 10:28:09.763 INF OpcUa] ChannelId 0: Token #0 created. CreatedAt=10:28:09.763. Lifetime=3600000. 
[... cut ...] scriptionServices.cs:line 289
   at Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices.SubscriptionWrapper.ActivateAsync(Session session) in D:\a\1\s\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Protocol\src\Services\SubscriptionServices.cs:line 259
[2023-03-22 10:35:08.739 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 

 **pcap for UA expert**
[image]
[image]
[image]
[image]
[image]

 **pcap for OPC publisher**

[image]
[image]
[image]
[image]

marcschier, 2023-06-22:
It would be great to give OPC Publisher Preview 4 a try (mcr.microsoft.com/iotedge/opc-publisher:2.9.0-preview4), it contains the latest OPC UA stack which has a lot of fixes around max number of nodes.  This item now covers a way to start OPC Publisher without loading complex type system to get around any misbehavior.

closed 2023-06-23 by marcschier