Azure/Industrial-IoT #1954: Unable to use direct method PublishNodes_V1 with OPCPublisher 2.8.2

TashaRoseLisle, 2023-03-06:
We have recently moved from version 2.5.5 to 2.8.2 of the OPCPublisher and can no longer publish nodes via the PublishNodes or PublishNodes_V1 Direct Method. Below is my payload and response: 
[image]

My understanding based on this documentation is that the publish nodes direct method should work for 2.8.2: https://azure.github.io/Industrial-IoT/modules/publisher-migrationpath.html#opc-publisher-25x-direct-methods-not-supported-in-282 . I have tried adding all optional attributes to the payload as well, but still receive a 500 response. 

I have set up an OPCPublisher module with the default container options and haven't changed anything. The same payload works for version 2.5.5.

marcschier, 2023-03-07:
Could you share the command line the publisher is started with?  Also, a log file for the publisher when this issue occurs would be great.

TashaRoseLisle, 2023-03-09:
Hi @marcschier , 

Thank you for getting back to me so quickly! I have managed to resolve the issue now. The problem was I was not specifying a publishednodes.json file in the container create options:
```

{
  "Hostname": "publisher",
  "Cmd": [
    "--aa",
    "--to",
    "--di=60",
    "publisher",
    "--pf=./publishednodes.json"
  ],
  "HostConfig": {
    "Binds": [
      "/srv/opcpublisher:/appdata"
    ]
  }
}
```
This wasn't an issue in the previous version and due to it being a test environment, it wasn't as important to store the nodes. Maybe this could be added to the documentation somewhere as it wasn't very clear to me.

marcschier, 2023-03-09:
Hi @TashaRoseLisle, happy to hear you could resolve the problem.  In versions prior to 2.6 we created a default published nodes.json file.  With the introduction of orchestrated mode in 2.6 we use the presence of the file name on the command line to understand in which mode the publisher is executing.  We are removing orchestrated mode in 2.9 and revert back to the 2.5 behavior of using and creating a default file, so this will be fixed soon.  I will take a note to update the docs in the meantime if time permits, though.

closed 2023-03-09 by marcschier