Azure/Industrial-IoT #1955: Trying to connect to a KEPServerEx OPC-UA Server crashes at random intervals

JaivyDaam, 2023-03-07:
**Describe the bug**
When we have successfully connected the device and deployed the edge modules, we are able to discover the server and endpoints, browse the endpoints for exposed sensors and publish them via the Engineering Tool. By using the Engineering Tool, a document is created in the CosmosDB with the exposed sensors and a DataWriterSetV2_{endpointId} is created in the IoT Hub. As we need to expose 190 items, we use the `/publisher/swagger/index.html` API page to bulk add these nodes. When the API call has been made, the document in the DB changes but the publisher crashes saying:
<details>
<summary>Error Publisher</summary>
<br>

> Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
>    at Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine.DiagnosticsOutputTimer_Elapsed(Object state) in D:\a\1\s\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Edge.Publisher\src\Engine\DataFlowProcessingEngine.cs:line 213
>    at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
> --- End of stack trace from previous location ---
>    at System.Threading.TimerQueueTimer.Fire(Boolean isThreadPool)
>    at System.Threading.TimerQueue.FireNextTimers()
</details>

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Engineering Tool, discover and activate the endpoint
2. Browse the endpoint and publish a sensor with 10 sec publishing interval and 5 sec sampling interval
3. Go to the https://all-in-one-container.azurewebsites.net/publisher/swagger/index.html and open the "/publisher/v2/publish/{endpointId}/bulk" dropdown
4. Add the list of sensors with `NodesToAdd` and `Headers` parameters as the asset requires UserName AuthenticationMethod.
5. Go to the publisher and see the error.

**Expected behavior**
I am expecting for the publisher to create a new subscription or edit the existing subscription with the newly added sensors. 

**Server (please complete the following information):**
 - OS: Linux Ubuntu 20.04 (for the edge)

**Desktop**
 - MacOS Ventura
 - Browser: Chrome (For the Engineering Tool and Publisher API)
 - Version 109

**Additional context**
- We used the `deploy.ps1` with `app` parameter to deploy the infrastructure
- I have noticed, when the job fails to start and the publisher module crashes, the CosmosDB document for that endpoint loses its `JobConfiguration.Job.writerGroup.dataSetWriters` information with the desired published nodes.
- I have also noticed, that when we use this Bulk API some of the information of the CosmosDB document changes: 
`JobConfiguration.Job.writerGroup.dataSetWriters.dataSetWriterId` changes to a {GUID}_10000 from {endpointId}_10000
- As we are onboarding a customer, the sensitive information has been anonymised from the support-bundle.
  - Support Bundle: [support_bundle_2023_03_02_09_53_54_UTC.zip](https://github.com/Azure/Industrial-IoT/files/10909300/support_bundle_2023_03_02_09_53_54_UTC.zip)

marcschier, 2023-03-07:
It appears the endpoint Url that is set in the data flow for the particular entry is null and the diagnostics output is crashing.  I am tagging this for a 2.8.5 release and will look into reasons that this is so..

Other observations: The --verbose option is an invalid option, it should be removed.  The --DiagnosticsInterval option cannot override the one hard coded in Cosmos DB when running in orchestrated mode.

marcschier, 2023-03-07:
A quick fix is in #1956 to a) handle the 0 subscriptions in the flow correctly and b) catch any errors in the diagnostics output to prevent crashes.

JaivyDaam, 2023-03-07:
> Other observations: The --verbose option is an invalid option, it should be removed. The --DiagnosticsInterval option cannot override the one hard coded in Cosmos DB when running in orchestrated mode.

I've noticed that too, that was my mistake. I've changed that in future iterations. Luckily, for troubleshooting reasons, it happened again with the new configuration (without --verbose), didn't read the documentation correctly (orchestrated vs standalone) 

I saw your release, thank you so much! I will wait until the pipeline has released the new version and deploy the publisher with the new `2.8.5` version specifically. Any pointers other than changing the configuration deployment to the `2.8.5` versioning?

marcschier, 2023-03-07:
Updating to 2.8.5 when ready should be all.  If you are familiar with changing the Cosmos DB document, you can also change the DiagnosticsInterval property from 60 seconds to 0 seconds, which should turn diagnostics logging off and prevent the crash.  If not, then we are not done here yet, then greatly appreciate a heads up.

JaivyDaam, 2023-03-07:
I have changed the `JobConfiguration.Job.engine.diagnosticsInterval` from `00:01:00` to `00:00:00` & `0` and did not give the desired result as I am still seeing the output in publisher module. Just to be sure, I deleted the `--DiagnosticsInterval` command from the publisher module, also to no effect.

Let me know if there is anything else I can try.

marcschier, 2023-03-09:
@JaivyDaam, thank you once more for filing this issue.  We released 2.8.5 today and this release prevents the diagnostics output handler from crashing. If you uncover any additional issues, please file and we will try and help.

closed 2023-03-09 by marcschier