Azure/Industrial-IoT #1962: Deploy.ps1 with type "app" parameter generates errors

JaivyDaam, 2023-03-23:
**Describe the bug**
When deploying the app with the `deploy/scripts/deploy.ps1` script, there is an error generated about "Authenticate" with "7" arguments:
<details>
<summary>Error details</summary>

```
System.Management.Automation.MethodInvocationException: Exception calling "Authenticate" with "7" argument(s): "ManagedIdentityCredential authentication failed: Service request failed.
Status: 400 (Bad Request)
Content:
Headers:
X-Powered-By: REDACTED
ETag: W/"455-HcZcfuI+O3ruUR9VQCocc6UAy9U"
Date: Thu, 23 Mar 2023 13:20:36 GMT
Connection: keep-alive
Keep-Alive: REDACTED
Content-Type: application/json; charset=utf-8
Content-Length: 1109
See the troubleshooting guide for more information. https://aka.ms/azsdk/net/identity/managedidentitycredential/troubleshoot"
 ---> Azure.Identity.AuthenticationFailedException: ManagedIdentityCredential authentication failed: Service request failed.
Status: 400 (Bad Request)
Content:
Headers:
X-Powered-By: REDACTED
ETag: W/"455-HcZcfuI+O3ruUR9VQCocc6UAy9U"
Date: Thu, 23 Mar 2023 13:20:36 GMT
Connection: keep-alive
Keep-Alive: REDACTED
Content-Type: application/json; charset=utf-8
Content-Length: 1109
See the troubleshooting guide for more information. https://aka.ms/azsdk/net/identity/managedidentitycredential/troubleshoot
 ---> Azure.RequestFailedException: Service request failed.
Status: 400 (Bad Request)
Content:
Headers:
X-Powered-By: REDACTED
ETag: W/"455-HcZcfuI+O3ruUR9VQCocc6UAy9U"
Date: Thu, 23 Mar 2023 13:20:36 GMT
Connection: keep-alive
Keep-Alive: REDACTED
Content-Type: application/json; charset=utf-8
Content-Length: 1109
   at Azure.Identity.ManagedIdentitySource.HandleResponseAsync(Boolean async, TokenRequestContext context, Response response, CancellationToken cancellationToken)
   at Azure.Identity.ManagedIdentitySource.AuthenticateAsync(Boolean async, TokenRequestContext context, CancellationToken cancellationToken)
   at Azure.Identity.ManagedIdentityClient.AuthenticateAsync(Boolean async, TokenRequestContext context, CancellationToken cancellationToken)
   at Azure.Identity.ManagedIdentityCredential.GetTokenImplAsync(Boolean async, TokenRequestContext requestContext, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at Azure.Identity.CredentialDiagnosticScope.FailWrapAndThrow(Exception ex, String additionalMessage)
   at Azure.Identity.ManagedIdentityCredential.GetTokenImplAsync(Boolean async, TokenRequestContext requestContext, CancellationToken cancellationToken)
   at Azure.Identity.ManagedIdentityCredential.GetTokenAsync(TokenRequestContext requestContext, CancellationToken cancellationToken)
   at Microsoft.Azure.PowerShell.Authenticators.MsalAccessTokenAcquirer.GetAccessTokenAsync(String callerClassName, String parametersLog, TokenCredential tokenCredential, TokenRequestContext requestContext, CancellationToken cancellationToken, String tenantId, String userId, String homeAccountId)
   at Microsoft.Azure.Commands.Common.Authentication.Factories.AuthenticationFactory.Authenticate(IAzureAccount account, IAzureEnvironment environment, String tenant, SecureString password, String promptBehavior, Action 1 promptAction, IAzureTokenCache tokenCache, String resourceId)
   at CallSite.Target(Closure, CallSite, Object, Object, Object, Object, Object, String, Object, String)
   --- End of inner exception stack trace ---
   at System.Management.Automation.ExceptionHandlingOps.CheckActionPreference(FunctionContext funcContext, Exception exception)
   at System.Management.Automation.Interpreter.ActionCallInstruction`2.Run(InterpretedFrame frame)
   at System.Management.Automation.Interpreter.EnterTryCatchFinallyInstruction.Run(InterpretedFrame frame)
   at System.Management.Automation.Interpreter.EnterTryCatchFinallyInstruction.Run(InterpretedFrame frame) - this must be done manually with appropriate permissions.
iot-app-client' updated with required resource access.
System.Management.Automation.MethodInvocationException: Exception calling "Authenticate" with "7" argument(s): "ManagedIdentityCredential authentication failed: Service request failed.
Status: 400 (Bad Request)
Content:
Headers:
X-Powered-By: REDACTED
ETag: W/"455-HcZcfuI+O3ruUR9VQCocc6UAy9U"
Date: Thu, 23 Mar 2023 13:20:37 GMT
Connection: keep-alive
Keep-Alive: REDACTED
Content-Type: application/json; charset=utf-8
Content-Length: 1109
See the troubleshooting guide for more information. https://aka.ms/azsdk/net/identity/managedidentitycredential/troubleshoot"
 ---> Azure.Identity.AuthenticationFailedException: ManagedIdentityCredential authentication failed: Service request failed.
Status: 400 (Bad Request)
Content:
Headers:
X-Powered-By: REDACTED
ETag: W/"455-HcZcfuI+O3ruUR9VQCocc6UAy9U"
Date: Thu, 23 Mar 2023 13:20:37 GMT
Connection: keep-alive
Keep-Alive: REDACTED
Content-Type: application/json; charset=utf-8
Content-Length: 1109
See the troubleshooting guide for more information. https://aka.ms/azsdk/net/identity/managedidentitycredential/troubleshoot
 ---> Azure.RequestFailedException: Service request failed.
Status: 400 (Bad Request)
Content:
Headers:
X-Powered-By: REDACTED
ETag: W/"455-HcZcfuI+O3ruUR9VQCocc6UAy9U"
Date: Thu, 23 Mar 2023 13:20:37 GMT
Connection: keep-alive
Keep-Alive: REDACTED
Content-Type: application/json; charset=utf-8
Content-Length: 1109
   at Azure.Identity.ManagedIdentitySource.HandleResponseAsync(Boolean async, TokenRequestContext context, Response response, CancellationToken cancellationToken)
   at Azure.Identity.ManagedIdentitySource.AuthenticateAsync(Boolean async, TokenRequestContext context, CancellationToken cancellationToken)
   at Azure.Identity.ManagedIdentityClient.AuthenticateAsync(Boolean async, TokenRequestContext context, CancellationToken cancellationToken)
   at Azure.Identity.ManagedIdentityCredential.GetTokenImplAsync(Boolean async, TokenRequestContext requestContext, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at Azure.Identity.CredentialDiagnosticScope.FailWrapAndThrow(Exception ex, String additionalMessage)
   at Azure.Identity.ManagedIdentityCredential.GetTokenImplAsync(Boolean async, TokenRequestContext requestContext, CancellationToken cancellationToken)
   at Azure.Identity.ManagedIdentityCredential.GetTokenAsync(TokenRequestContext requestContext, CancellationToken cancellationToken)
   at Microsoft.Azure.PowerShell.Authenticators.MsalAccessTokenAcquirer.GetAccessTokenAsync(String callerClassName, String parametersLog, TokenCredential tokenCredential, TokenRequestContext requestContext, CancellationToken cancellationToken, String tenantId, String userId, String homeAccountId)
   at Microsoft.Azure.Commands.Common.Authentication.Factories.AuthenticationFactory.Authenticate(IAzureAccount account, IAzureEnvironment environment, String tenant, SecureString password, String promptBehavior, Action 1 promptAction, IAzureTokenCache tokenCache, String resourceId)
   at CallSite.Target(Closure, CallSite, Object, Object, Object, Object, Object, String, Object, String)
   --- End of inner exception stack trace ---
   at System.Management.Automation.ExceptionHandlingOps.CheckActionPreference(FunctionContext funcContext, Exception exception)
   at System.Management.Automation.Interpreter.ActionCallInstruction 2.Run(InterpretedFrame frame)
   at System.Management.Automation.Interpreter.EnterTryCatchFinallyInstruction.Run(InterpretedFrame frame)
   at System.Management.Automation.Interpreter.EnterTryCatchFinallyInstruction.Run(InterpretedFrame frame) - this must be done manually with appropriate permissions.
```
</details>

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Azure portal and start Azure CLI
2. Download the deploy.ps1 script from the repo
3. run it with the `-type app` parameter
4. See error

**Expected behavior**
To see no error of sorts

**Additional context**
At this point, I am unsure what the implications are. It seems that overall the app is deployed successfully. Wanted to share it anyway to see if I didn't miss something.

marcschier, 2023-04-26:
I think that is an issue with running inside Azure CLI which is running in a VM with managed identity.  If all went well it likely is a non-issue as it seemed to have been able to authenticate nonetheless or else the deployment would have failed.  I will keep this around for when we have time to repro and massage the script accordingly.

marcschier, 2023-05-19:
This is likely fixed now that we moved to the officially supported msgraph powershell module.

closed 2023-05-21 by marcschier