Azure/Industrial-IoT #1967: What would be the recommended way to upload large publishednodes.json files from cloud to OPC Publisher?

b-enigma-con, 2023-03-29:
We are running the OPC Publisher in standalone mode on an IoT Edge device. We have a case in which the node configuration is stored in the cloud and we need to regularly update the published nodes configuration in the OPC Publisher. The total size of the configuration exceeds the maximum size limit of direct methods (64KB), so we're not able to call direct method "PublishNodes" from the cloud. What would be the preferred way to handle this situation?

Implement another Iot Edge module which downloads the configuration file from blob storage, replaces the previous file and then restart the OPC Publisher (if necessary)? Any suggestions?

marcschier, 2023-03-30:
OPC Publisher configuration API has an internal chunk mechanism to overcome the limitation. Unfortunately, this requires using .net on the client-side.  There is a feature request in issues that asks  for a "replace" API, that could be a combination. 

The right approach (in particular for DRI, to upload last state) is to use blob storage as backing store here. This could be built into the storage provider layer in OPC Publisher, but this is work that needs to be done, and I will use this item to track this as a feature request.  Alternatively, yes, one could build this outside of OPC Publisher and then volume mount the file into both containers. OPC Publisher reacts to file system changes and will re-load its configuration.  Such design though is a bit brittle and a simpler solution is probably better here.

marcschier, 2023-05-19:
OPC Publisher 2.9.0-preview2 is now out and we added a SetEndpointConfiguration API (in addition to the AddOrUpdateEndpoint "patching" API in 2.8) that allows uploading entire published nodes.json (when large this requires using chunking layer, or using the new REST or MQTT API endpoints of OPC Publisher module).

closed 2023-05-19 by marcschier