Azure/Industrial-IoT #1969: Device starts sampling and publishing at slower intervals after reconnection in 2.9.0 (preview)

NoTuxNoBux, 2023-04-24:
**Describe the bug**
On 2.9.0 (preview), I'm seeing strange behaviour where I ask a sample _and_ publishing interval of 50 ms for multiple properties, and after a reconnection, this gets somehow changed to a slower interval.

The initial values are always correct.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a few property subscriptions with sampling and publishing intervals of 50 ms.
2. Connect to the server, observe data coming through at the requested rate.
3. Let the server reboot or go offline and the publisher pick this up.
4. Put the server back online and the publisher reconnect.
5. Observe (most times) that updates come in at a slower rate than 50 ms.

**Expected behavior**
Updates come in at 50 ms.

**Additional context**
I'm not sure if this is the OPC Publisher that is at fault - it could also be a weird OPC UA server, but if not, hopefully I'll get an idea as to how this can happen.

Other things I've verified:

* Use 0 as publication interval; the server always negotiates 4, so 50 is certainly not too much for it.
* Publisher still reports "50/50" as requested and expected when the problem occurs.
* The rate at which `MessageTriggerMessageReceived` is triggered is also already visibly slower, so it happens earlier in the publisher (or is caused by the server).
* The amount it becomes slower is random; usually updates suddenly start getting published after 5 seconds, sometimes 300 ms, sometimes 1 second, and so on.
* If just the publishing interval became wrong, I'd expect multiple values to show up in the data as the sampling interval is still low, but **that's not the case**; I still get only one sample per publication, which implies both values got slower somehow.
* There are no other properties in published-nodes.json that have different publishing intervals that may mistakenly get used; they are all 50 ms.

Is there some timing logic somewhere that may get confused in the OPC Publisher? I couldn't find any, but I may have missed it.

I cannot test with a stable version because I'm using the 2.9.0 (preview) with a custom SignalR back end. This custom back end is similar to the new MQTT one and I didn't touch any of the logic related to data subscriptions; as mentioned the problem also seems to happen before the back end is included.

NoTuxNoBux, 2023-04-24:
Doing some more testing for this, I just encountered this problem happening mid-runtime. The publisher reconnected when the server came back and then this happened:

1. Got a bunch of messages with the expected publishing interval; timestamps of messages are mostly on the same second, which is accurate.
2. `... Changing monitoring mode from Disabled to Reporting` and some other logging happening.
3. More messages come in, but now they are all spaced apart at least a second (and I also visibly saw this happening).

These are the logs:

```
[2023-04-24 13:19:16.097 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://THE_SERVER/FOO': server current state: Running 
[2023-04-24 13:19:16.098 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://THE_SERVER/FOO', subscription '50' - sequence# 69-04/24/2023 13:19:16 
[2023-04-24 13:19:16.099 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://THE_SERVER/FOO received message with sequenceNumber 0 
[2023-04-24 13:19:16.118 DBG Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Publishing 318 bytes to IoTDataUpdate 
[2023-04-24 13:19:16.196 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://THE_SERVER/FOO': server current state: Running 
[2023-04-24 13:19:16.197 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://THE_SERVER/FOO', subscription '50' - sequence# 70-04/24/2023 13:19:16 
[2023-04-24 13:19:16.199 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://THE_SERVER/FOO received message with sequenceNumber 0 
[2023-04-24 13:19:16.222 DBG Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Publishing 318 bytes to IoTDataUpdate 
[2023-04-24 13:19:16.300 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://THE_SERVER/FOO': server current state: Running 
[2023-04-24 13:19:16.302 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://THE_SERVER/FOO', subscription '50' - sequence# 71-04/24/2023 13:19:16 
[2023-04-24 13:19:16.303 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://THE_SERVER/FOO received message with sequenceNumber 0 
[2023-04-24 13:19:16.326 DBG Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Publishing 318 bytes to IoTDataUpdate 
[2023-04-24 13:19:16.392 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://THE_SERVER/FOO': server current state: Running 
[2023-04-24 13:19:16.392 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://THE_SERVER/FOO', subscription '50' - sequence# 72-04/24/2023 13:19:16 
[2023-04-24 13:19:16.392 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://THE_SERVER/FOO received message with sequenceNumber 0 
[2023-04-24 13:19:16.429 DBG Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Publishing 318 bytes to IoTDataUpdate 
[2023-04-24 13:19:16.491 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://THE_SERVER/FOO': server current state: Running 
[2023-04-24 13:19:16.492 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://THE_SERVER/FOO', subscription '50' - sequence# 73-04/24/2023 13:19:16 
[2023-04-24 13:19:16.493 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://THE_SERVER/FOO received message with sequenceNumber 0 
[2023-04-24 13:19:16.534 DBG Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Publishing 318 bytes to IoTDataUpdate 
[2023-04-24 13:19:16.553 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://THE_SERVER/FOO': server current state: Running 
[2023-04-24 13:19:16.555 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://THE_SERVER/FOO', subscription '50' - sequence# 74-04/24/2023 13:19:16 
[2023-04-24 13:19:16.555 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://THE_SERVER/FOO received message with sequenceNumber 0 
[2023-04-24 13:19:16.586 DBG Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Publishing 522 bytes to IoTDataUpdate 
[2023-04-24 13:19:16.647 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session 'opc.tcp://THE_SERVER/FOO': server current state: Running 
[2023-04-24 13:19:16.648 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://THE_SERVER/FOO received message with sequenceNumber 0 
[2023-04-24 13:19:16.649 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://THE_SERVER/FOO', subscription '50' - sequence# 75-04/24/2023 13:19:16 
[... cut ...]
[2023-04-24 13:19:28.551 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Notification for session 'opc.tcp://THE_SERVER/FOO', subscription '50' - sequence# 86-04/24/2023 13:19:28 
[2023-04-24 13:19:28.552 DBG Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] Message trigger for opc.tcp://THE_SERVER/FOO received message with sequenceNumber 0 
[2023-04-24 13:19:28.599 DBG Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Publishing 205 bytes to IoTDataUpdate 
```

NoTuxNoBux, 2023-04-24:
I'm now wondering if this is somehow related to #1890: the properties involved are all 'application properties' on the OPC UA server; the server is a robot that has certain system properties that are always available, and application properties that only are available once a specific program is loaded.

I also subscribe to the system time (system property) in the publisher, which samples every second (and also publishes every 50 ms along with the rest whenever it is sampled), but - even when this problem occurs -, it doesn't seem to experience this specific issue.

I'm still not sure who's at fault here - the server or the client - but it may be linked to certain application properties being reloaded on the server and the publisher getting confused somehow because of this.

marcschier, 2023-04-26:
Hi @NoTuxNoBux, this looks like an interesting project :-).  But first off, I cannot really see anything wrong in the logs.  How OPC UA subscriptions work is that when the subscription is created all values are being sent.  After that only value changes are sent.  So even if you sample the changes every 50 ms, if nothing changed from previous sample, then nothing is sent.  The publishing interval decides the timing of when data goes over the wire, and if there is nothing ready to package into a publish response, nothing is returned.  Setting publishing interval to higher means more values are packaged, but latency increases.  So if this does not explain what you are seeing, my suggestion is to take a closer look at the server here (even though I cannot fully exclude publisher, but the area we are talking about has been very well in use over the past years and has not changed in preview 1).  

Out of curiosity, you mention signalr backend, are you looking at pumping data through signalR from MQTT or directly from OPC Publisher?

NoTuxNoBux, 2023-04-26:
>  The publishing interval decides the timing of when data goes over the wire, and if there is nothing ready to package into a publish response

Thanks for the response, you mentioning this helps - the troublesome property is an 'application' property on the OPC UA server, so perhaps it's not the publishing interval timing that's incorrect, perhaps the property (due to a bug on the server application side) is simply not being changed at the same interval any more when this problem occurs.

**I'm going to see if I can verify if this is what is happening and come back to this issue once I do.**

> 1). What server are you connecting to?

It's the OPC UA server on a Stäubli CS9 box for a TX2-60L mechanical robot arm. We're connecting over OPC UA TCP with both authentication (username and password) and encryption active (which works great :+1:).

> Out of curiosity, you mention signalr backend, are you looking at pumping data through signalR from MQTT or directly from OPC Publisher?

The latter - we had problems with the Azure IoT Hub resulting in sporadic random delays of several seconds or more (apparently a known issue because it is shared tenant, so there's not really much Microsoft support can do about it), and we have a minor subset of the properties we are interested in that need to be sent at a more consistent pace and with lower latency than most others.

We were already using SignalR to push updates to client applications, and saw 2.9.0 preview has multiple back ends (IoT Hub and MQTT), so building a SignalR back end based on the new MQTT one seemed like a good approach  - we didn't choose MQTT directly because - If I understood correctly - it would require introducing an additional intermediary (queuing server/implementation) that can in turn introduce make latency more erratic again.

(In case you're curious, the SignalR part seems to be working really well so far and was relatively straightforward to get this in the codebase.)

NoTuxNoBux, 2023-04-27:
> I'm going to see if I can verify if this is what is happening and come back to this issue once I do.

I'm not 100% sure yet, but it's becoming more and more likely that the server is at fault here. I can also reproduce this issue with other tooling such as UA Expert; the publishing interval remains the same, but the update rate of the property itself tanks, so it's not sending any more updates because the property simply wasn't updated, not because the publishing interval is incorrect.

Since it's also reproducible with other tools, I think this can be closed as not a bug in OPC Publisher. I will reopen in case new information appears that implicates the publisher again for some reason.

Thanks for the valuable tip and assistance!

NoTuxNoBux, 2023-05-19:
I just wanted to cycle back and confirm that it was indeed a _server_ issue; turns out the server in question has a quirk where it can only publish at a single interval, so if multiple (yes, that's right) separate clients connect (doesn't matter if it's an OPC Publisher or not), even for separate properties, and use different publication intervals, only the last one is used - the first one then gets updates at a slower rate _without_ it being renegotiated in OPC UA; the property just updates at at a slower pace.

Might be useful to know for others experiencing similar issues.