Azure/Industrial-IoT #1971: How to automate pn.json updation with addition/modification of more than 3000 opc nodes

Priyanka-L-Dalawai, 2023-04-26:
Question Summary
How to add. modify tags in opcnodes array in pn.json? As count of tags increase it is really hard to update it manually
Detailed background

Additional details required to frame the context of your question.
Which is best way to update it
In OPC Publisher we have to configure tags in publishnodes.json, Whenever we have to add.modify new tags we do it manually, instead how we can automate this without any blob storage
https://learn.microsoft.com/en-us/azure/industrial-iot/tutorial-publisher-configure-opc-publisher#configuration-via-configuration-file

marcschier, 2023-04-28:
Hi @Priyanka-L-Dalawai, Is the ask how to update the entries in the array programmatically?  Have you looked at our device method API, we have a PublishNodes and AddOrUpdateEndpoint API for that.  

If not what you were looking for, how would the ideal solution look like for you?

Priyanka-L-Dalawai, 2023-04-29:
Yes, I am looking for the same how to update the entries in the array programmatically
In my case I am looking at
1) If I form json array with modified or new entries how to access that pn.json inside linux vm from any microservice/api 
2) Suppose I got access to that path and I try to update pn.json, Time it will take to update/add new nodes what happens to messgages till opc publisher will be up and running

marcschier, 2023-06-22:
We now have a good set of API documented in openapi.json and generated markdown. How to call API is also documented rudimentarily. This item now will cover building samples for different ways to call the API.  Changing the title.

marcschier, 2023-09-28:
Please see the new http api and the iot hub samples in the samples folder for examples on how to use the API.

closed 2023-09-28 by marcschier