Azure/Industrial-IoT #1976: OPC Twin Module connectivity issue

sa1sen, 2023-05-02:
OPC Twin not connecting to server

1. Deployed OPC Twin as module to EFLOW (mcr.microsoft.com/iotedge/opc-twin)
2. Monitor logs
3. Stuck as 'Rejected certificate store contains 0 certs'
4. Waiting for upstream messages

Logs for OPCTwin Module:

[2023-05-02 08:43:58.286 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to xxxxxxxxxxxx using MqttOverTcp. 
[2023-05-02 08:44:03.852 INF Root] Starting module OpcTwin version 2.8.5.1. 
[2023-05-02 08:44:03.853 INF Root] Initiating prometheus at port 9701/metrics 
[2023-05-02 08:44:03.876 INF Root] Prometheus metric server started. 
[2023-05-02 08:44:03.881 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client. 
[2023-05-02 08:44:03.886 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context. 
[2023-05-02 08:44:05.530 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module xxxxxxxxxxxx_opctwin reconnected due to Connection_Ok. 
[2023-05-02 08:44:07.104 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Initialize device twin for xxxxxxxxxxxx - opctwin 
[2023-05-02 08:44:07.308 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Applying initial desired state. 
[2023-05-02 08:44:07.317 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Reporting currently initial state. 
[2023-05-02 08:44:07.432 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host started. 
[2023-05-02 08:44:07.563 INF OpcUa] Imported the PFX private key for [xxxxxxxxxxxx]. 
[2023-05-02 08:44:07.576 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] Own certificate Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=xxxxxxxxxxxx' (thumbprint: xxxxxxxxxxxx) loaded. 
[2023-05-02 08:44:07.579 INF OpcUa] Checking application instance certificate. 
[2023-05-02 08:44:07.641 INF OpcUa] Imported the PFX private key for [xxxxxxxxxxxx]. 
[2023-05-02 08:44:07.642 INF OpcUa] Check certificate: [CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=xxxxxxxxxxxx] [xxxxxxxxxxxx] 
[2023-05-02 08:44:07.644 INF OpcUa] Check application instance certificate. [CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=xxxxxxxxxxxx] [xxxxxxxxxxxx] 
[2023-05-02 08:44:07.700 INF OpcUa] Using the ApplicationUri: urn:xxxxxxxxxxxx:Microsoft.Azure.IIoT:microsoft: 

[2023-05-02 08:44:07.712 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] Application own certificate store contains 1 certs. 
[2023-05-02 08:44:07.712 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] 01: Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=xxxxxxxxxxxx' (thumbprint: xxxxxxxxxxxx) 
[2023-05-02 08:44:07.712 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] Trusted issuer store contains 0 certs. 
[2023-05-02 08:44:07.713 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] Trusted issuer store has 0 CRLs. 
[2023-05-02 08:44:07.714 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] Trusted peer store contains 1 certs. 
[2023-05-02 08:44:07.714 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] 01: Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=xxxxxxxxxxxx' (thumbprint: xxxxxxxxxxxx) 
[2023-05-02 08:44:07.714 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] Trusted peer store has 0 CRLs. 
[2023-05-02 08:44:07.714 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] Rejected certificate store contains 0 certs. 

Expectation was all available twin nodes would arrive upstream. Final outcome is for me to be able to write back to OPC Server tags.

marcschier, 2023-05-02:
Hi @sa1sen, the OPC Twin in 2.8 must be accessed through the cloud micro services provided in this repository and deployed to AKS.  

In OPC publisher 2.9 preview 2 (coming soon) you will be able to invoke browse and read/write service calls directly on the OPC Publisher through IoT Hub (and later HTTP/MQTT).  If you can wait I recommend leveraging this (simpler) capability.

Q: What are the OPC UA services that you are expecting to use and how will you call these services?  So I can give you more support here.

sa1sen, 2023-05-02:
Thanks @marcschier for your response.

Okay great - is there any ETA on 2.9 preview 2? 

I have requirement where we need to subscribe to OPC Server and publish the telemetry back to Azure IoT Hub. We will also deploy another module (Machine learning inference model) that will read the values from OPC Publisher and output a result that may have to be written back to the OPC Server (edge computing). If you could recommend me any documents around this it would be great!

Thanks

sa1sen, 2023-05-03:
Hi @marcschier ,

For 2.9, OPC Publisher, I read this:

"The same API can also be called via the HTTP Server built into OPC Publisher (Preview). The API supports browse and histrian access streaming, which the other transports do not provide. All calls must be authenticated through an API Key which must be provided as a bearer token. The API key can be read from the OPC Publisher module's module twin."

is there any more documentation on how to trigger a REST call (even if its reading the current connected endpoint) ?

Thnks

marcschier, 2023-06-22:
Added first part of this documentation in the readme.md for opc-publisher. https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/readme.md#using-iot-edge-simulation-environment

closed 2023-06-22 by marcschier