Azure/Industrial-IoT #1979: OPC Twin module - Username authorization not working

VladimirMakarevich, 2023-05-05:
**Describe the bug**
When OpcAuthenticationUsername is set for an Endpoint the authorization not working for OPC Twin module version 2.8.6.

An example of a request that we send via ServiceClient.InvokeDeviceMethodAsync()

`{
    "endpoint":
    {
        "url": "opc.tcp://randomhost:50023",
        "securityMode": "Forward",
        "securityPolicy": "http://opcfoundation.org/UA/SecurityPolicy#None"
    },
    "request":
    {
        "nodeId": "i=85",
        "direction": 0,
        "maxReferencesToReturn": 0,
        "header":
        {
            "elevation":
            {
                "type": "UserName",
                "value": "{\"user\":\"user1\",\"password\":\"password23\"}"
            }
        }
    }
}`

for this request we get the response 500:

`{"Message":"Response 500 ","Details":{"Message":"User/passord token format is not supported.","Exception":"ServiceResultException"}}`

**Expected behavior**
UserName authorization should work and twin browsing should work as well.

**Additional context**
OPC Twin module version 2.8.6.
nuget Microsoft.Azure.Devices version 1.30.0

marcschier, 2023-05-05:
There is no need to quote the Json, it should be part of the regular Json request object. 

```json
...
"elevation": { "type": "UserName", "value": {"user":"user1","password":"password23"} } } } } 
```

marcschier, 2023-05-05:
Please reopen if this did not work ;)

closed 2023-05-05 by marcschier