Azure/Industrial-IoT #1980: .deploy.sh fails

tervoju, 2023-05-09:
**Describe the bug**

https://github.com/Azure/Industrial-IoT/blob/main/docs/web-api/readme.md states that you can download repo and run it with command 

```./deploy.sh.``` 

cannot: it will end with result
```
Register-PackageSource: The specified Uri 'https://www.poshtestgallery.com/api/v2/' for parameter 'Location' is an invalid Web Uri. Please ensure that it meets the Web Uri requirements.
```

**To Reproduce**
Steps to reproduce the behavior:
1. git clone repo
2. run .deploy.sh
3. run fails

**Expected behavior**
deploy success / local run 

**Screenshots**
NA

**Desktop (please complete the following information):**
 - OS: ubuntu 22.04 WSL2

JaivyDaam, 2023-05-09:
Hi @tervoju ,

I would like to recommend to use the 2.8.6 tag release of the repository. If feel that main is the base for the 2.9.0 preview branch and has many changes compared to the 2.8.x releases.

Myself would still need to test out the 2.9.0/main branch and I have not seen this error in any prior releases.

marcschier, 2023-05-09:
It appears that the poshtestgallery server is down.  It should be possible to install the officially released Az and AzureAD modules from PSGallery instead though and then run the /deploy/scripts/deploy.ps1 script directly.

marcschier, 2023-05-19:
Deployment scripts upgraded to msgraph powershell module.

closed 2023-05-19 by marcschier