Azure/Industrial-IoT #1982: OPC Publisher v2.8.4 has less data ingestion rate

anil-k-gopalakrishna, 2023-05-12:
**Describe the bug**
After we upgrade from OPC Publisher v2.5.4 to v2.8.4, we see the data throughput rate has reduced drastically. It took 5 hours to deliver the messages after being queued up at the edge 

**Analysis**
Here is the edge log of a particular tag with value 308.83775, which we received at UTC time 10.24 today. The tag is generated from the source at same time.  When we checked at the messages being received at this point of time in the cloud, it was old messages which was previously queued up at the edge.

[image]

The same tag is actually received at the cloud DB only at UTC time 15.29.  That means, it took approx. 5 hrs to ingest the data into the cloud

[image]

The command line options are below:
                "--di=10",
                "--si=10",
                "--ms=225000",
                "--mq=3"

We have total of 15 VMS out of which we see issues with 4 VMs where there is high number of tags configured. Please find attached support_bundle  

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04

[support_bundle.zip](https://github.com/Azure/Industrial-IoT/files/11466572/support_bundle.zip)

marcschier, 2023-05-13:
Thank you @anil-k-gopalakrishna for reporting the issue.  While we take a look at the bundle, have you followed the steps in the troubleahooting [guide](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/troubleshooting.md#debugging-telemetry-data-issues)?

anil-k-gopalakrishna, 2023-05-15:
Thanks @marcschier . 
Yes, I have followed most of the steps. I have not done simulated device test, because it is our production system. The points to be noted here is-
- It was working without any issues earlier with opcpublisher v2.5.4. After upgrading to v2.8.4, we are facing these issues. 
- Data is flowing, but it is delayed. We observed about 5 hours in this terminal. In some other terminals, we are seeing approx 12 hours delay also.

anil-k-gopalakrishna, 2023-05-15:
I had one more observation today. One of the VM which was sending delayed telemetry data(due to queueing at edge), started sending out data in time, from today morning. We haven't done any changes or restart. One interesting fact is that; if I run 'iotedge check', I am getting below errors in the connection. 

[image]

marcschier, 2023-05-15:
I checked the logs, and I cannot see any issues in OPC Publisher. This appears to be supported by the local message logging module you have built and that as far as I can tell shows that the messages are in IoT Edge correctly.  (please correct).  Therefore, this could be related to EdgeHub to cloud latency/connectivity issues, or IoT Hub to DB (EventHub) latency.  From what I have seen in the past it a missing check point on the EventHub and a restart of the consumer that has to reprocess old messages, but again, this is something that needs to be investigated inside Azure context.  

I recommend opening a support request with IoT Hub to investigate the source of the delays or leverage the Azure monitor/insights to troubleshoot further.

closed 2023-05-15 by marcschier