Azure/Industrial-IoT #1990: Sending commands over OPC Twin takes long the first time

NoTuxNoBux, 2023-05-24:
**Describe the bug**
When sending commands through the OPC Twin, we're noticing delays from 8 to 30 seconds before the first command is delivered after connection was previously lost.

**To Reproduce**
1. Register OPC Twin with a server that supports commands.
2. Send a command the first time.
3. Reboot the OPC UA server.
4. Send a command through the OPC Twin.
5. Notice how the twin starts reconnecting because the previous connection went bad, after which it sends the command, but the authentication flow can take up to half a minute on slower servers.

**Expected behavior**
I'm not sure; perhaps (a flag to) make the OPC Twin reconnect as soon as possible instead of on demand.

**Additional context**
As part of a demo application, we send a command that reboots the device running the OPC UA server. After the device comes back, the OPC Publisher reconnect ASAP to send back data, but it appears the OPC Twin does not follow this behaviour. Instead, it only seems to reconnect 'on demand' when the first command is passed to it afterwards, which makes sending the command incur a rather large delay.

After this happened, sending the next command (when not rebooting) is much faster, likely because the existing connection is intact.

We are sending commands to the OPC Twin through the IoT Hub API, in case it is relevant.

NoTuxNoBux, 2023-05-24:
Wondering if this might be fixed with OPC Twin 2.8.4 (I was on 2.8.3) because of #1910.

NoTuxNoBux, 2023-06-16:
Hmmm, I'm not sure if this is the OPC Twin at fault, but the last few weeks (the only thing that changed is the update to 2.8.6 from 2.8.3) the connection gets closed after about a minute or so after sending the last command.

```
[2023-06-16 13:58:49.521 INF OpcUa] ChannelId 76: in Closing state. 
[2023-06-16 13:58:50.522 WRN OpcUa] ChannelId 76: Force reconnect reason=BadConnectionClosed 'Remote side closed connection' 
[2023-06-16 13:58:50.522 INF OpcUa] ChannelId 76: in Closed state. 
[2023-06-16 13:58:50.523 INF OpcUa] ChannelId 76: CLIENTCHANNEL SOCKET CLOSED: 0300A612 
```

'Remote side' implies the twin isn't causing it, but as far as we know there has been no changes on the network or OPC UA server that could be causing it. Possibly a new regression?

marcschier, 2023-06-18:
Please could you try with OPC Publisher 2.9.0-preview3?  In 2.9 we support calling the API through IOT hub directly (we only support calling through the REST API in 2.8).  The client handling in 2.9 was a complete rewrite and now uses a reference count mechanism to determine connect vs. disconnect.  A configurable linger timeout allows you to keep active connections (sessions) "lingering" around for a configurable time (--cl) after reference count hits 0.  That gives you a quick start with the next call within the timeout period.  Also, should the configured IoT Hub request timeout expire (default 10 seconds) we cancel the call (e.g., in cases where the session cannot be connected and the calling thread has to wait).

NoTuxNoBux, 2023-06-19:
I can confirm it doesn't happen in 2.8.3.

> Please could you try with OPC Publisher 2.9.0-preview3?

I would like to, but I don't immediately find any upgrade notes (possibly because the documentation updates are pending as mentioned in the release notes).

Am I correct in assuming that because `opc-twin:2.9.0-preview3` doesn't exist, the twin been merged into the opc-publisher image? Secondly, we have two publishers running right now with different back-ends, if it is merged into the publisher, will they be conflicting when relaying commands passed on by the IoT Hub since the endpoint URL is only passed in `EndpointApiModel` (and not the IoT Hub module name)?

Assuming a 'yes' to the first question and keeping the second publisher on the old version for now, I immediately hit an exception on startup that seems to be due to configuration I'm not manually passing - but I could be wrong:

```
[08:42:15.9048] info: System.Net.Http.HttpClient.Default.LogicalHandler[100]
      => HTTP POST unix:///var/run/iotedge/workload.sock/modules/OPC-Publisher/genid/537955643018451501/decrypt?api-version=2021-12-07
      Start processing HTTP request POST unix:///var/run/iotedge/workload.sock/modules/OPC-Publisher/genid/537955643018451501/decrypt?api-version=2021-12-07
[08:42:15.9119] info: System.Net.Http.HttpClient.Default.ClientHandler[100]
      => HTTP POST unix:///var/run/iotedge/workload.sock/modules/OPC-Publisher/genid/537955643018451501/decrypt?api-version=2021-12-07
      Sending HTTP request POST unix:///var/run/iotedge/workload.sock/modules/OPC-Publisher/genid/537955643018451501/decrypt?api-version=2021-12-07
[08:42:15.9271] fail: Azure.IIoT.OpcUa.Publisher.Storage.PublishedNodesConverter[0]
      failed to convert the published nodes.
      System.NotSupportedException: The 'unix' scheme is not supported.
         at Microsoft.Extensions.Http.Logging.LoggingHttpMessageHandler.<SendAsync>g__Core|5_0(HttpRequestMessage request, CancellationToken cancellationToken)
         at Microsoft.Extensions.Http.Logging.LoggingScopeHttpMessageHandler.<SendAsync>g__Core|5_0(HttpRequestMessage request, CancellationToken cancellationToken)
         at System.Net.Http.HttpClient.<SendAsync>g__Core|83_0(HttpRequestMessage request, HttpCompletionOption completionOption, CancellationTokenSource cts, Boolean disposeCts, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)
         at Microsoft.Azure.Devices.Edge.Util.Edged.Version20190130.GeneratedCode.HttpWorkloadClient.DecryptAsync(String api_version, String name, String genid, DecryptRequest payload, CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Edge.Util.Edged.WorkloadClientBase.ExecuteAsync[T](Func`1 func, String operation)
         at Microsoft.Azure.Devices.Edge.Util.Edged.Version20190130.WorkloadClient.HandleException(Exception ex, String operation)
         at Microsoft.Azure.Devices.Edge.Util.Edged.WorkloadClientBase.ExecuteAsync[T](Func`1 func, String operation)
         at Microsoft.Azure.Devices.Edge.Util.Edged.Version20190130.WorkloadClient.DecryptAsync(String initializationVector, String encryptedText)
         at Furly.Azure.IoT.Edge.Services.IoTEdgeWorkloadApi.DecryptAsync(String initializationVector, ReadOnlyMemory`1 ciphertext, CancellationToken ct)
         at Azure.IIoT.OpcUa.Publisher.Storage.PublishedNodesConverter.ToUserNamePasswordCredentialAsync(PublishedNodesEntryModel entry) in /home/vsts/work/1/s/src/Azure.IIoT.OpcUa.Publisher/src/Storage/PublishedNodesConverter.cs:line 700
         at Azure.IIoT.OpcUa.Publisher.Storage.PublishedNodesConverter.ToConnectionModel(PublishedNodesEntryModel model) in /home/vsts/work/1/s/src/Azure.IIoT.OpcUa.Publisher/src/Storage/PublishedNodesConverter.cs:line 448
         at System.Linq.Lookup`2.Create[TSource](IEnumerable`1 source, Func`2 keySelector, Func`2 elementSelector, IEqualityComparer`1 comparer)
         at System.Linq.GroupedEnumerable`3.ToList()
         at Azure.IIoT.OpcUa.Publisher.Storage.PublishedNodesConverter.ToWriterGroups(IEnumerable`1 items, PublisherOptions configuration) in /home/vsts/work/1/s/src/Azure.IIoT.OpcUa.Publisher/src/Storage/PublishedNodesConverter.cs:line 262
```

marcschier, 2023-06-19:
Cool, you found a bug :-).  (#1998). Looks like you have a published nodes configuration file with encrypted username and password fields in there.  Is this the case?  

You are correct, there is no opc twin and opc discovery module in 2.9, everything is merged into 2.9 opc publisher.

With device methods in IoT Hub you are directing the method call request at the module you want the method to execute in, so the module name is passed to IOT hub.  This should allow you to distinguish both publishers and then just pass the endpoint url.

marcschier, 2023-06-26:
Please try "2.9.0-preview4" tag.  It is already available on mcr, but not yet a release on GItHub.

bqstony, 2023-06-27:
thx the error is gone

marcschier, 2023-06-27:
@NoTuxNoBux, when you get a chance to retest against 2.9.0-preview4 and let me know if the original issue still exist. Keeping this open.

marcschier, 2023-07-10:
Closing for time being, please re-open if the issue is still in 2.9.

closed 2023-07-10 by marcschier