Azure/Industrial-IoT #1999: standalone OPC-Publisher in Edge Simulator doesn't work

delema98, 2023-06-19:
**Describe the bug**
When starting the opc-publisher in standalone mode in the edge simulator, I get a certificate error (see trace below). The same configuration works on a  regular deployed Azure IoT Edge server. Other modules can be simulated. 

I tried the latest version as well as the preview 3 version. While the latest version is constantly restarting the opc-publisher module the preview 3 version can connect as a client to an existing opc-server. Outgoing messages (from opc-publisher --> edge runtime) fail.
The following traces are from using the preview version:

[image]
[image]
[image]

marcschier, 2023-06-20:
See #1708

marcschier, 2023-06-20:
Could you post the command line you used to start publisher in the iotedgehubdev context?

delema98, 2023-06-20:
I tried it with different command line arguments. The most basic configuration was the following:
"modules": {
          "OPCPublisher": {
            "env": {},
            "restartPolicy": "always",
            "settings": {
              "image": "mcr.microsoft.com/iotedge/opc-publisher:2.9.0-preview3",
              "createOptions": {
                "Hostname": "publisher",
                "Cmd": [
                  "--aa",
                  "--di=60"
                ]
              }
            },
            "status": "running",
            "type": "docker"
          }
        }

The whole EdgeSolution is based on the default deployment.json and opcpublisher is the only module running

marcschier, 2023-06-22:
Fixed in preview 4.

closed 2023-06-22 by marcschier