Azure/Industrial-IoT #2004: Receiving keyvault error when deploying app services

rjdaskevich-DB, 2023-06-24:
Both IIOT web and service apps fail with an error during provisioning.

Several errors are spawning regarding keyvault. 

Here is the output of the default_docker log:

2023-06-24T05:10:20.051185826Z [05:10:19 INF] No managed service identity configured for this service.
2023-06-24T05:10:22.662100535Z [05:10:22 FTL] Host terminated unexpectedly
2023-06-24T05:10:22.662135534Z System.AggregateException: One or more errors occurred. (A keyvault uri was provided could not access keyvault at the address. If you want to read configuration from keyvault, make sure the keyvault is reachable, the required permissions are configured on keyvault and authentication provider information is available. Sign into Visual Studio or Azure CLI on this machine and try again.)
2023-06-24T05:10:22.662145634Z  ---> Microsoft.Azure.IIoT.Exceptions.InvalidConfigurationException: A keyvault uri was provided could not access keyvault at the address. If you want to read configuration from keyvault, make sure the keyvault is reachable, the required permissions are configured on keyvault and authentication provider information is available. Sign into Visual Studio or Azure CLI on this machine and try again.
2023-06-24T05:10:22.662152334Z  ---> Microsoft.Azure.KeyVault.Models.KeyVaultErrorException: Operation returned an invalid status code 'Unauthorized'
2023-06-24T05:10:22.662157634Z    at Microsoft.Azure.KeyVault.KeyVaultClient.GetSecretWithHttpMessagesAsync(String vaultBaseUrl, String secretName, String secretVersion, Dictionary`2 customHeaders, CancellationToken cancellationToken)
2023-06-24T05:10:22.662162934Z    at Microsoft.Azure.KeyVault.KeyVaultClientExtensions.GetSecretAsync(IKeyVaultClient operations, String vaultBaseUrl, String secretName, CancellationToken cancellationToken)
2023-06-24T05:10:22.662168134Z    at Microsoft.Extensions.Configuration.ConfigurationEx.KeyVaultConfigurationProvider.ValidateReadSecretAsync(String secretName) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 232
2023-06-24T05:10:22.662174034Z    at Microsoft.Extensions.Configuration.ConfigurationEx.KeyVaultConfigurationProvider.CreateInstanceAsync(IConfigurationRoot configuration, Boolean allowInteractiveLogon, String keyVaultUrlVarName, Boolean lazyLoad) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 197
2023-06-24T05:10:22.662179834Z    --- End of inner exception stack trace ---

Here is the output of the pull and warm up request that continuously fails:

2023-06-24T05:55:14.034Z INFO  - Initiating warmup request to container rjs935iiotdemo_0_31937273_msiProxy for site rjs935iiotdemo
2023-06-24T05:55:14.168Z INFO  - Container rjs935iiotdemo_0_31937273_msiProxy for site rjs935iiotdemo initialized successfully and is ready to serve requests.
2023-06-24T05:55:14.170Z INFO  - Initiating warmup request to container rjs935iiotdemo_0_31937273 for site rjs935iiotdemo
2023-06-24T05:55:19.023Z ERROR - Container rjs935iiotdemo_0_31937273 for site rjs935iiotdemo has exited, failing site start
2023-06-24T05:55:19.109Z ERROR - Container rjs935iiotdemo_0_31937273 didn't respond to HTTP pings on port: 80, failing site start. See container logs for debugging.
2023-06-24T05:55:20.158Z INFO  - Stopping site rjs935iiotdemo because it failed during startup.

Tried in multiple subscriptions with same issue.  Also tried an older release: 2.8.4.

Please advise

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

marcschier, 2023-06-25:
Hi, could you please provide repro steps, including the exact command you used when you deployed.?

Meanwhile 2 notes:
* The deployment script in main line deploys latest I believe, please try setting the version to "2.9.0-preview4" to see if this is the cause.  
* The -type app deployment currently does not work, issue is filed (engineering tool not working)

marcschier, 2023-06-27:
Tested with 2.9.0-preview4 and scripts in main and it worked fine. Closing until more info is available.

closed 2023-06-27 by marcschier