Azure/Industrial-IoT #2009: OPCPublisher:latest API Server not working

sa1sen, 2023-07-04:
Hello,

I am trying to follow this example below:

> **Calling the API over HTTP**
> _(This feature is in preview)_
> 
> You can now send HTTP requests to the publisher module http server at https://localhost:8081. The unsecure endpoint is mounted at https://localhost:8080 for testing purpuses. E.g. to get the swagger definition run:
> curl http://localhost:8080/swagger/v2/openapi.json"
> 
> However, I get 'Failed to connect to localhost port 8080' error message. I have tried using the 'netstat -an' command to list out all ports available on localhost and I cannot find 8080 / 8081

.

What I have done:

1. Made sure all local ports connection are enabled for UDP + TCP in Windows Firewall for both inbound and outbound rules
2. Tried Adding '--DisableHttpServer=disabled' , "--httpserverport", "--unsecurehttp" to the OPC Publisher Module as part of Container Create Options

But still no luck

marcschier, 2023-07-04:
If you want to access the server on the localhost you need to create a port binding from 80 to 8080 and 443 to 8081 like in the [deployment.json](docs/opc-publisher/deployment.json?raw=1) in the same folder.  No other command line arguments should be needed.  You do not need to export the ports if you use a module in the same Azure IoT Edge docker network to access the server.  (But then you use just https://<publisher-module-name>) since 443 port is secured using a IoT Edge module SSL certificate which should be in the module trust list.

marcschier, 2023-07-04:
You can follow the instructions to run the module in iotedgehubdev using the deployment.json file in the repo.  If the http api is then not accessible, please reopen and add repro steps, and the deployment manifest you used.

marcschier, 2023-07-04:
I noticed that swagger is off in release builds. Will add a command line option to turn it on or off for both release and debug.

marcschier, 2023-07-04:
Also, the title says ":latest", the tag for 2.9.0 is "2.9.0-preview4" if that makes a difference.  It is possible that the deployment.json refers to latest, but latest is still 2.8.6 at this point.

marcschier, 2023-07-07:
Swagger UI will be always on in 2.9, you can disable using `--doa`, `--disableopenapi`.

closed 2023-07-07 by marcschier