Azure/Industrial-IoT #2011: Documentation - Direct calls in opc-twin

pesetskyps, 2023-07-05:
**I would like to be able to invoke direct calls from my microservice to opc-publisher triggering browsing processing and getting the value on the demand from the opc node.**
Is there ability to guide to the docs and the code I can dig into to understand how and with what parameters I should invoke direct call from my microservice to get value on demand from opc-ua node and also trigger browsing process.

marcschier, 2023-07-06:
Please find docs here: https://github.com/Azure/Industrial-IoT/tree/main/docs/opc-publisher#programming-against-opc-publisher-using-the-opc-publisher-api.  A sample setup is in deploy/docker folder.

marcschier, 2023-07-06:
Duplicate of #1971 - building samples for OPC Publisher APIs

closed 2023-07-06 by marcschier